iiab/iiab #2276: ./iiab-network failed to set a Fixed IP, due to double quotes around eth0 in /etc/dhcpcd.conf

holta, 2020-03-05:
Running `cd /opt/iiab/iiab` then `./iiab-network` on RPi 3 B+ (the IIAB microSD was originally built in a RPi 4) appears to have put the following into /etc/dhcpcd.conf &mdash; after a Fixed IP Address was specified in local_vars.yml:

```
interface "eth0"
```

The specified Fixed IP Address never appeared (even after reboot) as a result of the double quotes above.  So I manually changed it to:

```
interface eth0
```

And then rebooted...at that point eth0's Fixed IP Address began to work.  @jvonau may have ideas why.

Aside: Being extremely offline, thankfully I had an offline copy of FAQ.IIAB.IO -> "22 How do I set a static IP address?" on the IIAB itself at http://box/info (telling me what to do manually) to rescue the situation in front of 20 teachers & 5 Chiapas govt inspectors!

Related:

- http://FAQ.IIAB.IO -> "22 How do I set a static IP address?"
- #1965 "eth0" appears (with quotes, sometimes alone on its own line) in /etc/iiab/iiab.ini

jvonau, 2020-03-05:
Without a gateway present the recorded IIAB_WAN_DEVICE is used as a seed value via prior_gw -> device_gw -> gui_wan_iface for the device to be configured. See 2278

holta, 2020-03-05:
@jvonau's PR #2278 should hopefully fix this, please take a look those who can!

holta, 2020-03-07:
This appears fixed by @jvonau's PR #2278, now merged.

closed 2020-03-07 by holta