iiab/iiab #2277: http://box/usb is becoming cluttered with dead dirs USB0, USB1, USB2, USB3, USB4 (etc)

holta, 2020-03-05:
Many teachers absolutely adore IIAB's zero-hassle content-sharing from USB sticks (their content appears nearly instantly at http://box/usb without the bureaucracy of an LMS) but recently it's become unbearable, as prior sticks' directories keep accumulating without being deleted.

Specifically, /library/www/html/local_content quickly ends up containing a flood of subdirectories, arising from "every" insertion of USB memory stick(s) &mdash; somehow creating yet more directories, even when the very same stick(s) are later re-inserted into IIAB.  This ever-growing flood of stale/bogus directories[*] needs to be deleted (somehow) as soon as is reasonably possible anyway:

USB0
USB1
USB2
USB3
USB4
...

[*] Does anybody idea know why these stale/artifact directories above (USB0 etc) are sticking around & cluttering up /library/www/html/local_content _much more than in the past?_

@tim-moody writes:
> I also see symlink on the usb itself. maybe some bugs?

@holta writes:
> There's always been a stale link or two, which was annoying to teachers...but they'd eventually find their stuff in the end.
>
> Seems something recently/likely (with NGINX or something else recent in Raspbian??) brings 2 softlinks for each stick...at least one of which is never deleted...so this growing flood...confuses everyday teachers.

@tim-moody writes:
> look at the stick itself and I think it has a link, which is new

Related:

http://FAQ.IIAB.IO -> "4 Can teachers display their own content?"

jvonau, 2020-03-05:
>Process '/usr/share/usbmount/usbmount remove' failed with exit code 1 

Noted in #2040

jvonau, 2020-03-05:
Are the usb devices being unmounted before being removed or just yanked out?

holta, 2020-03-05:
> Are the usb devices being unmounted before being removed or just yanked out?

In all 3 cases, stale dirs are sticking around in /library/www/html/local_content, i.e. when...

1) IIAB is powered off then USB stick(s) are removed.
2) USB stick(s) are unmounted then removed, on a very rare day when a Linux person exists in the IIAB field community (not realistic in developing world communities, but I tried it to confirm the result is indeed much the same).
3) USB sticks are pulled out by the teacher at the end of her/his hour-long lesson, while the IIAB remains on.

holta, 2020-03-06:
If it helps towards understanding and later automating this, here's one way to manually clear out the clutter after a USB memory stick was inserted &mdash; using these 3 commands on Raspbian Desktop 2020-02-13 on RPi 3 B+:

```
umount /dev/sda1
umount /dev/sda1
/library/www/html/local_content/USB0
```

Details showing what happens:

```
root@box:~# lsblk
NAME        MAJ:MIN RM   SIZE RO TYPE MOUNTPOINT
sda           8:0    1  14.5G  0 disk
└─sda1        8:1    1  14.5G  0 part /media/usb0
...

root@box:~# umount /dev/sda1

root@box:~# lsblk
NAME        MAJ:MIN RM   SIZE RO TYPE MOUNTPOINT
sda           8:0    1  14.5G  0 disk
└─sda1        8:1    1  14.5G  0 part /media/pi/Ubuntu 16.04.2 LTS amd64
...

root@box:~# umount /dev/sda1

root@box:~# lsblk
NAME        MAJ:MIN RM   SIZE RO TYPE MOUNTPOINT
sda           8:0    1  14.5G  0 disk
└─sda1        8:1    1  14.5G  0 part

root@box:~# ls -l /library/www/html/local_content/
total 4
drwxr-xr-x 2 pi   pi   4096 mar  2 23:11 software
lrwxrwxrwx 1 root root   11 feb 25 23:12 USB0 -> /media/usb0

root@box:~# rm /library/www/html/local_content/USB0
```

holta, 2020-03-08:
@jvonau's PR #2282 looks incredibly promising!

As I hope he has time to explain during tmrw's 10AM EDT community call (http://minutes.iiab.io).

holta, 2020-03-09:
Let's close this in a few days after confirming PR #2282 works well enough!

holta, 2020-03-25:
@tim-moody found a related bug (or at least lack of documentation) when Kolibri also tries to automount devices:

"Question: workflow for content export"
https://github.com/learningequality/kolibri/issues/6685

jvonau, 2020-03-25:
Guess I should add use `systemctl stop usbmount@dev-sda*` to unmount the drive manually.

holta, 2020-03-25:
> systemctl stop usbmount@dev-sda*

@jvonau confirms this should work across all OS's (Linux distributions).

holta, 2020-03-25:
Things like `systemctl stop usbmount@dev-sda*` tend to scare teachers to death, but still I've posted that to http://FAQ.IIAB.IO > "4 Can teachers display their own content?" as an FYI.

(The good news is that teachers insert USB memory sticks [into IIAB] as read-only devices, and to date there are no known problems with teachers spontaneously pulling out their personal USB sticks.)

holta, 2020-03-25:
Re-opening this ticket (#2277) so we can get a better understanding of @tim-moody's question here:

> on latest master neither poweroff or umount remove links in
/library/www/html/local_content/ on raspbian

https://github.com/iiab/iiab/pull/2282#issuecomment-603976039

jvonau, 2020-03-25:
> Things like `systemctl stop usbmount@dev-sda*` tend to scare teachers to death, but still I've posted that to http://FAQ.IIAB.IO > "4 Can teachers display their own content?" as an FYI.
> 

Put a wrapper around it like everything else you wanted to make easy...  iiab-remove-usbkey maybe?

holta, 2020-03-25:
> Put a wrapper around it like everything else you wanted to make easy... iiab-remove-usbkey maybe?

(1) Teachers pull out their USB sticks regardless
(2) Sticks are used in a read-only way

So in the real world e.g. rural / low-literacy schools, I personally don't see how a wrapper would make any material difference, when the human ergonomics issue appears fundamentally insurmountable.

holta, 2020-03-25:
@tim-moody also writes:

> and usb1 -> /media/usb1 can still appear in the mounted device

https://github.com/iiab/iiab/pull/2282#issuecomment-603999619

holta, 2020-03-26:
@avni may not have time in the coming week, but improving the usability of http://box/usb for teachers was discussed during today's IIAB call (http://minutes.iiab.io) &mdash; and would require things (like!) :

- understanding why a stray link keeps appearing on the USB stick itself
- testing all 4 sticks inserted into an RPi (repeatedly) until usability of the http://box/usb site is improved (confirming that dead dirs are no longer remain)
- increase the font size so that NGINX renders http://box/usb in a readable/clickable way on modern smartphones
- when could/should mounting of the stick be made read-only, to protect the teacher's own personal USB memory stick?

holta, 2020-03-27:
Thank you @shanti-bhardwa for your assistance testing all these above ( https://github.com/iiab/iiab/issues/2277#issuecomment-604551930 ) USB stick/drive usability issues, very greatly appreciated!

holta, 2020-03-30:
Appears fixed thanks to @jvonau's patches and @shanti-bhardwa's testing, even across more unusual scenarios like stick insertation/removal when IIAB's is _off._

Thanks to all others who helped as well!

closed 2020-03-30 by holta

holta, 2020-04-06:
Somewhat Related:

- iiab/iiab-admin-console#288 "remove mount symlinks on umount"

- #2040 "http://box/usb behaves differently on IIAB 7.0 built on A. [Raspberry Pi Desktop (for PC and Mac)] compared to B. Raspbian Buster Lite"