iiab/iiab #2284: ./iiab-network fails repeatedly on Debian 10.3 (and other OS's?)

holta, 2020-03-09:
@jvonau take a look when you can:

> TASK [network : Copy the bridge script] ****************************************
fatal: [127.0.0.1]: FAILED! => {"msg": "The conditional check 'not is_raspbian (iiab_lan_iface == \"br0\" or wan_ip != \"dhcp\" or gui_static_wan_ip == \"undefined\")' failed. The error was: Unexpected templating type error occurred on ({% if not is_raspbian (iiab_lan_iface == \"br0\" or wan_ip != \"dhcp\" or gui_static_wan_ip == \"undefined\") %} True {% else %} False {% endif %}): 'bool' object is not callable\n\nThe error appears to be in '/opt/iiab/iiab/roles/network/tasks/debian.yml': line 40, column 3, but may\nbe elsewhere in the file depending on the exact syntax problem.\n\nThe offending line appears to be:\n\n\n- name: Copy the bridge script\n  ^ here\n"}

holta, 2020-03-09:
The syntax error in PR #2275 is visible on Line 11 of roles/network/tasks/debian.yml here:

https://github.com/iiab/iiab/pull/2275/files#diff-e7bca1fe0faa3e6810aa7f06d00a63fc

holta, 2020-03-09:
PR #2286 seems to fix this...smoke-test now completes.

closed 2020-03-09 by holta

jvonau, 2020-03-09:
Odd that debian.yml is being invoked, I would of expected the sysd-netd-debian.yml path to be taken. Can you post a diagnostic link once the install completes