iiab/iiab #2287: Please post results of any/all testing installs

jvonau, 2020-03-10:
Perhaps posting a link after running iiab-diagnostics on all testing installs or tests of PR's could be a quasi tracker for the current state of successful installs on various distros/PRs? In #2284 I spotted something that is of interest but without any feedback I can't investigate, had there been a link to the diagnostic information I could of sought out what I was looking for without bothering to ask for more information.

kmw, 2020-03-13:
Yesterday on a p4x4g, I installed iiab on a 64g mico-sd w/ raspian
I used the "install content" to select and install, what appeared to be, less than 60g 
Naturally, the selected install packages was way more than 60g. I was lucky to be monitoring the drive and it did fill up, to where nothing else could be downloaded.
I tried to use a couple different usb drives to copy off one or two .mp4 files to get back ~1g of space. But I was not successful. I was able to uses the box.lan install and go through khan academy and the other packages, fine.
I then imaged the micro-sd chip into a 62g .img and created a 27g tar.gz 
I used Disks program to write the 64g.img to a micro-sd which DID NOT complete.
I've ordered a couple 128g micro-sd chips which I figure would give me enough space to work with. 
thank you so much for all you do.

holta, 2020-03-13:
Thanks @kmw !

Can you post the URL that is output after you run `iiab-diagnostics` ?

kmw, 2020-03-13:
pastebinit < /etc/iiab/diag/7.0_raspbian-10_2020-03-13_09:48:52_PDT_p4-4g-64g
pastebinit < /etc/iiab/diag/7.0_raspbian-10_2020-03-13_09:53:26_PDT_test

kmw, 2020-03-13:
[7.0_raspbian-10_2020-03-13_09:53:26_PDT_test.txt](https://github.com/iiab/iiab/files/4330764/7.0_raspbian-10_2020-03-13_09.53.26_PDT_test.txt)

kmw, 2020-03-13:
oh the above (file) is from my big install that i'm updating, 
The new p4x4g iiab install, of yesterday, is not running, right now, but i'll bring it up and send you that, too. thanks.

kmw, 2020-03-13:
hopefully I can get a little help with paths on the packages so that they resolve from the main menu. in other words there are several packages, listed on the menu that do not resolve. i'm wondering if they can be corrected based on the diagnostic. thanks.

tim-moody, 2020-03-13:
Please tell me which content items you selected that under reported their
size as 60G. You said you had a 64G sdcard. This will probably be 64
billion bytes, which is less than 64G. Also, the OS and other software and
the boot partition would probably leave less than 60G free. The Admin
Console allows you to over select, but the total should be colored
depending on how much of available space is required, including more than
100% should be red.

Please tell me which menu items do not resolve. If they are wikis please
run iiab-make-kiwix-lib, which would not have been run if a download of
Kiwix content failed.

kmw, 2020-03-13:
[7.1_raspbian-10_2020-03-13_11_13_13_PDT_test1.txt](https://github.com/iiab/iiab/files/4331168/7.1_raspbian-10_2020-03-13_11_13_13_PDT_test1.txt)

holta, 2020-04-08:
See Also:

#1737 "Who can help test IIAB 7.1 on Ubuntu 19.10+, Debian 10.3 & Raspbian Lite/Desktop?"
#1799 "Who can test IIAB 7.x on Raspberry Pi Zero W?"

holta, 2020-04-08:
See Also:

PR #2322 "pastebinit bugs: try dpaste.com instead (for iiab-diagnostics command)"

tim-moody, 2020-04-08:
medium install ran to completion on rpi 4

holta, 2020-04-08:
> medium install ran to completion on rpi 4

Can you boot it in RPi Zero W?

tim-moody, 2020-04-08:
in the process (forgot to install adm cons, so had to go back)

tim-moody, 2020-04-08:
Not sure how far to go with this. It boots; home page comes up; adm cons comes up.
From menu can reach kiwix (test), test module, kalite, osm test, calibre-web (nothing has content.)
Took WP and did install and got hello world
Took nextcloud, logged in, got gateway timeout 504 (twice)
Took sugarizer and got 502 bad gateway quickly (possible config error?)
htop doesn't show a lot of stress on cpu or memory

holta, 2020-04-08:
> Took nextcloud, logged in, got gateway timeout 504 (twice)
Took sugarizer and got 502 bad gateway quickly (possible config error?)

To be 100% clear, what RPi / CPU were you running?

holta, 2020-04-08:
And was this on Raspbian Lite?

tim-moody, 2020-04-08:
W raspbian lite

holta, 2020-04-08:
> Took nextcloud, logged in, got gateway timeout 504 (twice)

That's expected: Nextcloud fails with such little RAM &mdash; Nextcloud has for years recommended 512MB _minimum_ for PHP, as we enact (sometimes, depending on vaeiable nginx_high_php_limits) at https://github.com/iiab/iiab/blob/master/roles/www_options/tasks/main.yml#L66

> Took sugarizer and got 502 bad gateway quickly (possible config error?)

Makes sense.  Due to the underlying Node.js issue we're discussing at https://github.com/iiab/iiab/issues/1799#issuecomment-611123006

jvonau, 2020-04-08:
> 
> > Took sugarizer and got 502 bad gateway quickly (possible config error?)
> 
> Makes sense. Due to the underlying Node.js issue we're discussing at [#1799 (comment)](https://github.com/iiab/iiab/issues/1799#issuecomment-611123006)

#2082

holta, 2020-04-27:
@jvonau @kmw @tim-moody can we resolve this yet now?

Or move it to IIAB 7.2's milestone?

closed 2020-06-16 by jvonau