iiab/iiab #2288: RTC routine 2 issues

jvonau, 2020-03-13:
The current code always installs support for an add-on RTC even if there is not one present, this is a result rtc_id being defined to a value other than 'none' in 0-init/defaults/main.yml. Feels like an unwise thing to do but so far has not caused issues sans for this warning in dmesg. 
> pi@box:~ $ dmesg | grep rtc
[    4.732719] rtc-ds1307: probe of 1-0068 failed with error -121

This is part of the 'dead wood' referred to in #2270. Might cause some issues should somebody make use of the gpio pins for a different purpose. Think rtc_id should really be defined in local_vars so the exact type of RTC added could be populated by the person who added the RTC so the model of the RTC is an exact match to what hardware was installed. 

The other issue would be the file to be modified is in a different location on Ubuntu, would be /boot/firmware/config.txt. Perhaps this RTC support can go away or should it get fixed?

holta, 2020-03-13:
> Perhaps this RTC support can go away or should it get fixed?

I'm assuming this should be fixed.  But I might be wrong.

Let's get to the bottom of this with Suyash in Bhopal, India in coming days or during our next call (http://minutes.iiab.io)

holta, 2020-03-13:
One possible way around tbe RTC issue @georgejhunt ran into on #2270 — might be @jvonau's suggestion here:

- Set variable `rtc_id: none` in 0-init/defaults/main.yml

- AND: remove variable 'rtc_id' from all vars/raspbian-10.yml and other <OS>.yml files...putting it into /etc/iiab/local_vars.yml instead?

tim-moody, 2020-03-13:
When this code was written most installs had an rtc and we didn't want the average user to have to figure out the rtc model, especially since it has been harmless for probably three years.

Suyashdwivedi, 2020-03-13:
I have tested RTC with DS3231module, DS1307 is not good external temperature can affect the frequency of the oscillator circuit which drives the DS1307’s internal counter.The DS3231 is much more accurate, as it has an internal oscillator which isn’t affected by external factors – and thus is accurate down to a few minutes per year at the most.

jvonau, 2020-03-14:
> One possible way around tbe RTC issue @georgejhunt ran into on #2270 — might be @jvonau's suggestion here:
> 
>     * Set variable `rtc_id: none` in 0-init/defaults/main.yml

This should be the only change required for ubuntu to run pass this part of the code.
 
>     * AND: remove variable 'rtc_id' from all vars/raspbian-10.yml and other .yml files...putting it into /etc/iiab/local_vars.yml instead?

This would be the complete solution, providing the variable is left commented out in local_vars, to correct the behavior of always installing hardware support for non-existent hardware on raspbian.

jvonau, 2020-03-14:
> When this code was written most installs had an rtc and we didn't want the average user to have to figure out the rtc model, especially since it has been harmless for probably three years.

Using the ostrich approach to the issue is fine until somebody is actually using the gpio pins for some other purpose and the add-on hat is now non-functional with this 'harmless' entry.

holta, 2020-03-17:
@m-anish mentions:

Adding a DS3231 Real Time Clock To The Raspberry Pi
https://www.raspberrypi-spy.co.uk/2015/05/adding-a-ds3231-real-time-clock-to-the-raspberry-pi/

holta, 2020-03-19:
@jvonau should this ticket now be closed?

Now that PR #2289 is merged?

jvonau, 2020-03-19:
"rtc_id" needs to migrate from OS.yml to *_local_vars.yml for the full solution.

PR #2302 (2020-03-26): Move 'rtc_id: ds3231' from vars/*.yml to vars/local_vars_*.yml
And comment it out.

On @jvonau's request.  @georgejhunt agrees.

To resolve #2288
closed 2020-03-26 by holta