iiab/iiab #2296: gitea does not install on 64 bit ubuntu on rpi

tim-moody, 2020-03-20:
rpi 4 2G with ubuntu 19.10.1 64 bit

TASK [gitea : Fail if we detect unknown architecture] ****************************************************************************************************************************************
fatal: [127.0.0.1]: FAILED! => {"changed": false, "msg": "Could not find a binary for the CPU architecture \"aarch64\""}

holta, 2020-03-20:
@aidan-fitz can you look into this Gitea bug/issue, please if possible?  FYI:

1) Many people are now installing IIAB on Ubuntu 19.10.1 (and the daily pre-release builds of Ubuntu 20.04 LTS) on Raspberry Pi 4, as an alternative to the Raspbian OS &mdash; as Ubuntu on Raspberry Pi has proven to allow 32 simultaneous WiFi users &mdash; instead of just 18 as offered by the Raspbian OS: #823

2) Direct OS links here, including links to Ubuntu 20.04 LTS pre-releases: https://github.com/iiab/iiab/wiki/IIAB-Platforms

3) Perhaps even better: https://ubuntu.com/blog/how-to-install-ubuntu-with-the-new-raspberry-pi-imager

4) Or, if you need a direct VPN-link connection to one of the above OS's, please contact me directly!

tim-moody, 2020-03-20:
@aidan-fitz to be clear this is 64 bit and there is no aarch64 package.

tim-moody, 2020-03-20:
with kiwix we were able to run the 32 bit exec on 64 bit os

holta, 2020-03-20:
Aside: [Lokole](https://github.com/iiab/iiab/tree/master/roles/lokole) also does not install/run on Ubuntu on Raspberry Pi.

Hopefully @tim-moody (or someone!) can post the Lokole error message to a separate ticket in coming days.

holta, 2020-03-26:
Should be solved by @jvonau's PR #2301 -- thanks to everyone who can help with ongoing testing.

closed 2020-03-26 by holta