iiab/iiab #2297: lokole does not install on 64bit ubuntu on rpi

tim-moody, 2020-03-20:
ubuntu 19.10.1 64 bit on rpi 4 2G

TASK [lokole : DEFAULT: pip install opwen_email_client (Lokole, latest available version) from PyPI to /library/lokole/venv, if above vars both UNdefined] ***********************************
fatal: [127.0.0.1]: FAILED! => {"changed": false, "cmd": ["/library/lokole/venv/bin/pip3", "install", "--no-cache-dir", "opwen_email_client"], "msg": "stdout: Collecting opwen_email_client\n  Downloading https://files.pythonhosted.org/packages/2e/f9/c13ee81848ad475981fb26b2df208c9115668b5d19fff02a40c1612af315/opwen_email_client-0.5.11.tar.gz (605kB)\nRequirement already satisfied: Babel==2.8.0 in /library/lokole/venv/lib/python3.7/site-packages (from opwen_email_client) (2.8.0)\nRequirement already satisfied: Flask-BabelEx==0.9.3 in /library/lokole/venv/lib/python3.7/site-packages (from opwen_email_client) (0.9.3)\nRequirement already satisfied: Flask-Caching==1.8.0 in /library/lokole/venv/lib/python3.7/site-packages (from opwen_email_client) (1.8.0)\nRequirement already satisfied: Flask-Cors==3.0.8 in /library/lokole/venv/lib/python3.7/site-packages (from opwen_email_client) (3.0.8)\nRequirement already satisfied: Flask-Migrate==2.5.2 in /library/lokole/venv/lib/python3.7/site-packages (from opwen_email_client) (2.5.2)\nRequirement already satisfied: Flask-SQLAlchemy==2.4.1 in /library/lokole/venv/lib/python3.7/site-packages (from opwen_email_client) (2.4.1)\nRequirement already satisfied: Flask-Script==2.0.6 in /library/lokole/venv/lib/python3.7/site-packages (from opwen_email_client) (2.0.6)\nRequirement already satisfied: Flask-Security==3.0.0 in /library/lokole/venv/lib/python3.7/site-packages (from opwen_email_client) (3.0.0)\nRequirement already satisfied: Flask-Testing==0.7.1 in /library/lokole/venv/lib/python3.7/site-packages (from opwen_email_client) (0.7.1)\nRequirement already satisfied: Flask-WTF==0.14.2 in /library/lokole/venv/lib/python3.7/site-packages (from opwen_email_client) (0.14.2)\nRequirement already satisfied: Flask==1.1.1 in /library/lokole/venv/lib/python3.7/site-packages (from opwen_email_client) (1.1.1)\nCollecting Pillow==7.0.0 (from opwen_email_client)\n  Downloading https://files.pythonhosted.org/packages/39/47/f28067b187dd664d205f75b07dcc6e0e95703e134008a14814827eebcaab/Pillow-7.0.0.tar.gz (38.2MB)\nRequirement already satisfied: SQLAlchemy==1.3.13 in /library/lokole/venv/lib/python3.7/site-packages (from opwen_email_client) (1.3.13)\nRequirement already satisfied: WTForms==2.2.1 in /library/lokole/venv/lib/python3.7/site-packages (from opwen_email_client) (2.2.1)\nRequirement already satisfied: Werkzeug==0.16.1 in /library/lokole/venv/lib/python3.7/site-packages (from opwen_email_client) (0.16.1)\nCollecting apache-libcloud==3.0.0rc1 (from opwen_email_client)\n  Downloading https://files.pythonhosted.org/packages/1a/67/4672f3c50b9ac7a67addffa2224c3beab9057ad4b3db26435c57af50920d/apache_libcloud-3.0.0rc1-py2.py3-none-any.whl (3.3MB)\nCollecting bcrypt==3.1.7 (from opwen_email_client)\n  Downloading https://files.pythonhosted.org/packages/fa/aa/025a3ab62469b5167bc397837c9ffc486c42a97ef12ceaa6699d8f5a5416/bcrypt-3.1.7.tar.gz (42kB)\n  Installing build dependencies: started\n  Installing build dependencies: finished with status 'done'\nCollecting beautifulsoup4==4.8.2 (from opwen_email_client)\n  Downloading https://files.pythonhosted.org/packages/cb/a1/c698cf319e9cfed6b17376281bd0efc6bfc8465698f54170ef60a485ab5d/beautifulsoup4-4.8.2-py3-none-any.whl (106kB)\nCollecting cached-property==1.5.1 (from opwen_email_client)\n  Downloading https://files.pythonhosted.org/packages/3b/86/85c1be2e8db9e13ef9a350aecd6dea292bd612fa288c2f40d035bb750ded/cached_property-1.5.1-py2.py3-none-any.whl\nCollecting celery[sqlalchemy]==4.4.0 (from opwen_email_client)\n  Downloading https://files.pythonhosted.org/packages/47/53/6e2ccc87b18ddc582d2de31b3ed9144c72a68062659e9e4a68e19312d254/celery-4.4.0-py2.py3-none-any.whl (421kB)\nCollecting environs==7.1.0 (from opwen_email_client)\n  Downloading https://files.pythonhosted.org/packages/8c/e0/0f97feb7728de5acd6cb1c42fe4bbc9eb779c1c1e8a641e08b069a4aa441/environs-7.1.0-py2.py3-none-any.whl\nCollecting gunicorn==20.0.4 (from opwen_email_client)\n  Downloading https://files.pythonhosted.org/packages/69/ca/926f7cd3a2014b16870086b2d0fdc84a9e49473c68a8dff8b57f7c156f43/gunicorn-20.0.4-py2.py3-none-any.whl (77kB)\nCollecting mkwvconf==0.1.1 (from opwen_email_client)\n  Downloading https://files.pythonhosted.org/packages/de/d2/03c75327ce5295a98f0d5e205aaae9dfef651bacb76ca95fc7f93bd117dd/mkwvconf-0.1.1.tar.gz\nRequirement already satisfied: passlib==1.7.2 in /library/lokole/venv/lib/python3.7/site-packages (from opwen_email_client) (1.7.2)\nCollecting python-crontab==2.4.0 (from opwen_email_client)\n  Downloading https://files.pythonhosted.org/packages/bc/03/852f9c5f8da7f58abce06fe8ae769cbcf6502f1c60684b16bad72ffcced3/python-crontab-2.4.0.tar.gz (50kB)\nCollecting requests==2.22.0 (from opwen_email_client)\n  Downloading https://files.pythonhosted.org/packages/51/bd/23c926cd341ea6b7dd0b2a00aba99ae0f828be89d72b2190f27c11d4b7fb/requests-2.22.0-py2.py3-none-any.whl (57kB)\nCollecting typing==3.7.4.1 (from opwen_email_client)\n  Downloading https://files.pythonhosted.org/packages/fe/2e/b480ee1b75e6d17d2993738670e75c1feeb9ff7f64452153cf018051cc92/typing-3.7.4.1-py3-none-any.whl\nCollecting tzlocal==2.0.0 (from opwen_email_client)\n  Downloading https://files.pythonhosted.org/packages/ef/99/53bd1ac9349262f59c1c421d8fcc2559ae8a5eeffed9202684756b648d33/tzlocal-2.0.0-py2.py3-none-any.whl\nCollecting watchdog==0.10.1 (from opwen_email_client)\n  Downloading https://files.pythonhosted.org/packages/34/65/dca6d9653656630169e9f71fe3fcf2a487b671a096cf11dc3011c822a408/watchdog-0.10.1.tar.gz (94kB)\nCollecting xtarfile[zstd]==0.0.2 (from opwen_email_client)\n  Downloading https://files.pythonhosted.org/packages/3f/cb/b8bc79b4dafea85b754e59d6c337e2611a6f4e80fba1eea6874fe3716039/xtarfile-0.0.2.tar.gz\nRequirement already satisfied: pytz>=2015.7 in /library/lokole/venv/lib/python3.7/site-packages (from Babel==2.8.0->opwen_email_client) (2019.3)\nRequirement already satisfied: Jinja2>=2.5 in /library/lokole/venv/lib/python3.7/site-packages (from Flask-BabelEx==0.9.3->opwen_email_client) (2.11.1)\nRequirement already satisfied: speaklater>=1.2 in /library/lokole/venv/lib/python3.7/site-packages (from Flask-BabelEx==0.9.3->opwen_email_client) (1.3)\nRequirement already satisfied: Six in /library/lokole/venv/lib/python3.7/site-packages (from Flask-Cors==3.0.8->opwen_email_client) (1.14.0)\nRequirement already satisfied: alembic>=0.7 in /library/lokole/venv/lib/python3.7/site-packages (from Flask-Migrate==2.5.2->opwen_email_client) (1.4.2)\nRequirement already satisfied: itsdangerous>=0.21 in /library/lokole/venv/lib/python3.7/site-packages (from Flask-Security==3.0.0->opwen_email_client) (1.1.0)\nRequirement already satisfied: Flask-Login>=0.3.0 in /library/lokole/venv/lib/python3.7/site-packages (from Flask-Security==3.0.0->opwen_email_client) (0.5.0)\nRequirement already satisfied: Flask-Principal>=0.3.3 in /library/lokole/venv/lib/python3.7/site-packages (from Flask-Security==3.0.0->opwen_email_client) (0.4.0)\nRequirement already satisfied: Flask-Mail>=0.7.3 in /library/lokole/venv/lib/python3.7/site-packages (from Flask-Security==3.0.0->opwen_email_client) (0.9.1)\nRequirement already satisfied: click>=5.1 in /library/lokole/venv/lib/python3.7/site-packages (from Flask==1.1.1->opwen_email_client) (7.1.1)\nCollecting cffi>=1.1 (from bcrypt==3.1.7->opwen_email_client)\n  Downloading https://files.pythonhosted.org/packages/05/54/3324b0c46340c31b909fcec598696aaec7ddc8c18a63f2db352562d3354c/cffi-1.14.0.tar.gz (463kB)\nCollecting soupsieve>=1.2 (from beautifulsoup4==4.8.2->opwen_email_client)\n  Downloading https://files.pythonhosted.org/packages/05/cf/ea245e52f55823f19992447b008bcbb7f78efc5960d77f6c34b5b45b36dd/soupsieve-2.0-py2.py3-none-any.whl\nCollecting vine==1.3.0 (from celery[sqlalchemy]==4.4.0->opwen_email_client)\n  Downloading https://files.pythonhosted.org/packages/7f/60/82c03047396126c8331ceb64da1dc52d4f1317209f32e8fe286d0c07365a/vine-1.3.0-py2.py3-none-any.whl\nCollecting kombu<4.7,>=4.6.7 (from celery[sqlalchemy]==4.4.0->opwen_email_client)\n  Downloading https://files.pythonhosted.org/packages/34/7e/44445f7d9031e82b4a1160141de35ca810f44041bf5065b45c68ab0ec9e8/kombu-4.6.8-py2.py3-none-any.whl (183kB)\nCollecting billiard<4.0,>=3.6.1 (from celery[sqlalchemy]==4.4.0->opwen_email_client)\n [... cut ...] \"/library/lokole/venv/bin/python3 -u -c \"import setuptools, tokenize;__file__='/tmp/pip-install-rv3sln_0/Pillow/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\\r\\n', '\\n');f.close();exec(compile(code, __file__, 'exec'))\" install --record /tmp/pip-record-2n26rx7e/install-record.txt --single-version-externally-managed --compile --install-headers /library/lokole/venv/include/site/python3.7/Pillow\" failed with error code 1 in /tmp/pip-install-rv3sln_0/Pillow/\n"}

holta, 2020-03-20:
fyi @c-w & @aidan-fitz &mdash; with much more context at #2296

(i.e. schools around the world ask for and very much need 32 WiFi connections not just 18, hence the ongoing push from the Raspbian OS to Ubuntu on Raspberry Pi.)

jvonau, 2020-03-25:
> RequiredDependencyException:\n \n 
The headers or library files could not be found for jpeg,\n 
a required dependency when compiling Pillow from source.

missing libjpeg-dev for the headers if compiling?

jvonau, 2020-03-25:
Adding libjpeg-dev to the list of packages lets lokole complete the installation on U-20

holta, 2020-03-26:
Should be solved by @jvonau's PR #2301 -- thanks to everyone who can help with ongoing testing.

closed 2020-03-26 by holta