iiab/iiab #2300: trying to set up offline modus

sh1va73, 2020-03-25:
### Expected behavior and actual behavior.

I have set up a portal on my Pi4 and it works fine on my phone. When I open internet then the portal appears. But when I open internet on my laptop then it does not work.

### Steps to reproduce the problem.
 
I have added address=/#/192.168.158.10 pointing to what I have set up as router

### Specifications like the version of the IIAB, operating system version, or hardware details.

latest version on latest Raspbian on a Raspberry Pi 4.

holta, 2020-03-25:
Are you talking about Captive Portal?

Can you please explain more details & context?

- what URL's are you using/seeing?
- what laptop OS & browser are you using?
- what phone OS & browser are you using?
- can you explain more "address=/#/192.168.158.10" ?

sh1va73, 2020-03-25:
Yes I mean captive portal.
I am just opening Google on my iPhone and the captive portal opens. Thats good.
I want same on my laptop. My laptop is macOS Catalina. But on my laptop captive portal does not open. My browsers are Safari, Chrome and Firefox.
I have read in this forum I need to add address=/#/192.168.158.10 to my dnsmasq.conf to make it work offline. I tried that but it didnt help This IP is the IP I set as router IP in dnsmasq.conf.

sh1va73, 2020-03-25:
I have posted this on the wrong page. I am sorry for this, you can close this.

closed 2020-03-25 by sh1va73