iiab/iiab #2304: IIAB fails to install on Ubuntu 18.04 during roles/kalite: "No package matching 'python2' is available"

holta, 2020-03-26:
@jvonau can you help?

> TASK [kalite : Install python2] ************************************************
fatal: [127.0.0.1]: FAILED! => {"changed": false, "msg": "No package matching 'python2' is available"}

If nec, see the VM itself: 10.8.0.18

holta, 2020-03-26:
Code in question is https://github.com/iiab/iiab/blob/master/roles/kalite/tasks/install.yml#L14-L17:
``` - name: Install python2
  package: 
    name: python2
    state: present
```

Arising from:

PR #2294 "U20test / Ubuntu 20.04 to offer PHP 7.3 and 7.4 [Nextcloud PHP deps, php-stem on U20, ap0 prep, KA Lite on Ubermix, etc]"

jvonau, 2020-03-26:
#2305

holta, 2020-03-27:
Resolved by PR #2307.

closed 2020-03-27 by holta