iiab/iiab #2306: These PR's need to be in master at this point of release cycle

georgejhunt, 2020-03-26:
I encountered the python2 problem on Ubuntu, but the fix is in #2290, and as Jerry pointed out this morning, he has put ap0 fixes on top of what for me was basically a ubuntu PR. I don't think I have the patience to sort it all out.
I vote we merge #2290, #2266,#2293, and #2305 before next Thursday, and then see what falls out.
I'd be willing to smoke test a raspbian lite with these 4 PR's merged into a test branch in the next day, if everyone agrees, before we do the actual merge into master.

georgejhunt, 2020-03-26:
On second thought, I don't think ap0 is very tested. I think what I suggested is a bad idea.

We need some way to keep master from setting up ap0 by default.

holta, 2020-03-27:
FYI PR #2305 was replaced by PR #2307 now merged.

georgejhunt, 2020-03-27:
Without response, or agreement, I think it's too late for 7.1.

closed 2020-03-27 by georgejhunt