iiab/iiab #2310: Can IIAB favor 7.4 over 7.3 on Ubuntu 20.04 LTS? (To be released 2020-04-23)

holta, 2020-03-30:
@jvonau observes that Ubuntu 20.04 LTS will apparently ship with both PHP 7.3 and 7.4, upon the OS's release "2020-04-23".

It would be great to transition to PHP 7.4 &mdash; if that turns out to be practical on *both* amd64 (e.g. PC's, laptops, VM's) and arm64 (e.g. Raspberry Pi 4) possibly!?

holta, 2020-03-31:
Resolved by @jvonau's PR #2313

holta, 2020-04-01:
@shanti-bhardwa reports:

> I have started testing with a fresh build of  Ubuntu 20.04 and 7.1 on PI 4. This error with php 7.4 , TASK [www_base : Unarchive http://download.iiab.io/packages/php7.4-stem.aarch64.tar to / (rpi)] ******************************************************************************************************
fatal: [127.0.0.1]: FAILED! => {"changed": false, "msg": "Failure downloading http://download.iiab.io/packages/php7.4-stem.aarch64.tar, HTTP Error 404: Not Found"}
>
> PLAY RECAP *******************************************************************************************************************************************************************************************
127.0.0.1                  : ok=161  changed=92   unreachable=0    failed=1    skipped=42   rescued=0    ignored=0   
>
> looks like the package for 7.4 has not been installed yet and exits for 7.3

jvonau, 2020-04-01:
#2315

holta, 2020-04-01:
PR #2315 appears to fix the above bug.

If anybody can find a doc explaining Ansible's syntax here (disallowing both single-quotes and double-quotes...even in string comparisons?) that'd be tremendously helpful to all !

closed 2020-04-01 by holta