iiab/iiab #2311: Unable to install Calibre-Web on Ubuntu 20.04 pre-release: Python 3.8 issue?

holta, 2020-03-30:
@OzzieIsaacs & @jvonau: do you know why @shanti-bhardwa was unable to install Calibre-Web on Ubuntu 20.04 LTS (`arm64` pre-release) from http://cdimage.ubuntu.com/ubuntu-server/daily-preinstalled/current/ ?

> TASK [calibre-web : Download Calibre-Web dependencies (using pip) into python3 virtual environment /usr/local/calibre-web-py3] ***********************************************************************
fatal: [127.0.0.1]: FAILED! => {"changed": false, "cmd": ["/usr/local/calibre-web-py3/bin/pip3", "install", "-r", "/usr/local/calibre-web-py3/requirements.txt"], "msg": "stdout: Collecting Babel<2.9,>=1.3\n Downloading Babel-2.8.0-py2.py3-none-any.whl (8.6 MB)\nCollecting Flask-Babel<1.1.0,>=0.11.1\n Downloading Flask_Babel-1.0.0-py3-none-any.whl (9.5 kB)\nCollecting Flask-Login<0.5.1,>=0.3.2\n Downloading Flask_Login-0.5.0-py2.py3-none-any.whl (16 kB)\nCollecting Flask-Principal<0.5.0,>=0.3.2\n Downloading Flask-Principal-0.4.0.tar.gz (5.5 kB)\nCollecting singledispatch<3.5.0.0,>=3.4.0.0\n Downloading singledispatch-3.4.0.3-py2.py3-none-any.whl (12 kB)\nCollecting backports_abc>=0.4\n Downloading backports_abc-0.5-py2.py3-none-any.whl (5.2 kB)\nCollecting Flask<1.2.0,>=1.0.2\n Downloading Flask-1.1.1-py2.py3-none-any.whl (94 kB)\nCollecting iso-639<0.5.0,>=0.4.5\n Downloading iso-639-0.4.5.tar.gz (167 kB)\nCollecting PyPDF2<1.27.0,==1.26.0\n Downloading PyPDF2-1.26.0.tar.gz (77 kB)\nCollecting pytz>=2016.10\n Downloading pytz-2019.3-py2.py3-none-any.whl (509 kB)\nCollecting requests<2.23.0,>=2.11.1\n Downloading requests-2.22.0-py2.py3-none-any.whl (57 kB)\nCollecting SQLAlchemy<1.4.0,>=1.1.0\n Downloading SQLAlchemy-1.3.15.tar.gz (6.1 MB)\n Installing build dependencies: started\n Installing build dependencies: finished with status 'done'\n Getting requirements to build wheel: started\n Getting requirements to build wheel: finished with status 'error'\n\n:stderr: ERROR: Command errored out with exit status 1:\n command: /usr/local/calibre-web-py3/bin/python3 /usr/local/calibre-web-py3/share/python-wheels/pep517-0.7.0-py2.py3-none-any.whl/pep517/_in_process.py get_requires_for_build_wheel /tmp/tmp9jwptm9q\n cwd: /tmp/pip-install-kdtvzqit/SQLAlchemy\n Complete output (1 lines):\n /usr/local/calibre-web-py3/bin/python3: can't find '__main__' module in '/usr/local/calibre-web-py3/share/python-wheels/pep517-0.7.0-py2.py3-none-any.whl/pep517/_in_process.py'\n ----------------------------------------\nERROR: Command errored out with exit status 1: /usr/local/calibre-web-py3/bin/python3 /usr/local/calibre-web-py3/share/python-wheels/pep517-0.7.0-py2.py3-none-any.whl/pep517/_in_process.py get_requires_for_build_wheel /tmp/tmp9jwptm9q Check the logs for full command output.\n"}

holta, 2020-03-30:
@jvonau suspects this might be caused by https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=954256 ?

And as such this might be a transient issue (possibly even a mirroring issue) that will go away on its own in coming days?

jvonau, 2020-03-31:
[pep-517 error](https://github.com/sqlalchemy/sqlalchemy/issues/5207) should be fixed but may not be released yet.

jvonau, 2020-03-31:
temporary workaround:
add extra_args: "--no-use-pep517" to the pip call in roles/calibre-web/tasks/install.yml 
>+++ b/roles/calibre-web/tasks/install.yml
@@ -50,6 +50,7 @@
     virtualenv: "{{ calibreweb_venv_path }}"    # /usr/local/calibre-web-py3
     virtualenv_site_packages: no
     virtualenv_command:  python3 -m venv {{ calibreweb_venv_path }}
     extra_args: "--no-use-pep517"    
   when: internet_available | bool

holta, 2020-03-31:
> temporary workaround:
> add extra_args: "--no-use-pep517" to the pip call in roles/calibre-web/tasks/install.yml

@OzzieIsaacs & @jvonau:
Should this be put in a PR?  Conditional on which OS's if so?

OzzieIsaacs, 2020-04-01:
I'm currently not planing actions for Calibre-Web. 
I'll wait at least for the final version of Ubuntu.

holta, 2020-04-01:
I agree with @jvonau who writes:

> [let's] wait 48 hours and see what pypi/Ubuntu offers then??

jvonau, 2020-04-01:
There are two issues at play here, first is pep517 via pip3/setuptools seems to be influx ATM, second sqlalchemy is where this pep517 issue starts with but I don't think the fix is released yet.

holta, 2020-04-01:
_Huge thanks to @jvonau for having analyzed & explained this:_

> when the sqlalchemy fix is released this should go away

> moving python forward entails deploying a new dependency resolver (pep517) in pip with setuptools being neutered and looks to be going away.

> I'd just toggle the install to False in medium_vars, till [Ubuntu 20.04] is released.. Bug catchers will use [BIG-sized IIAB] anyway and the average joe won't hit this issue with [MEDIUM-sized IIAB]

jvonau, 2020-04-06:
With this update to pip3 in place I'm unable to reproduce
Start-Date: 2020-04-06  06:39:55
Commandline: /usr/bin/unattended-upgrade
Upgrade: python3-pip:arm64 (20.0.2-2, 20.0.2-4), python-pip-whl:arm64 (20.0.2-2, 20.0.2-4)
End-Date: 2020-04-06  06:39:59

jvonau, 2020-04-06:
python3 packages installed in the the virtualenv
> source /usr/local/calibre-web-py3/bin/activate
(calibre-web-py3) ubuntu@box:/usr/local/calibre-web-py3$ pip3 list
Package         Version   
--------------- ----------
Babel           2.8.0     
backports-abc   0.5       
blinker         1.4       
certifi         2020.4.5.1
chardet         3.0.4     
click           7.1.1     
Flask           1.1.2     
Flask-Babel     1.0.0     
Flask-Login     0.5.0     
Flask-Principal 0.4.0     
idna            2.8       
iso-639         0.4.5     
itsdangerous    1.1.0     
Jinja2          2.11.1    
MarkupSafe      1.1.1     
pip             20.0.2    
pkg-resources   0.0.0     
PyPDF2          1.26.0    
pytz            2019.3    
requests        2.22.0    
setuptools      45.2.0    
singledispatch  3.4.0.3   
six             1.14.0    
SQLAlchemy      1.3.15    
tornado         6.0.4     
Unidecode       1.1.1     
urllib3         1.25.8    
Wand            0.5.9     
Werkzeug        1.0.1

holta, 2020-04-06:
@shanti-bhardwa can you reconfirm this is no longer an issue?

holta, 2020-04-06:
Tangentially related:

#2322 "Should Calibre-Web playbook & vars be renamed with 'calibre_web' internally?"

holta, 2020-04-08:
@shanti-bhardwa confirms this is no longer an issue.

closed 2020-04-08 by holta