iiab/iiab #2312: KA Lite fails to install on Ubuntu 20.04 pre-release (arm64): "You are running Setuptools on Python 2, which is no longer supported" "ensure you are installing Setuptools using pip 9.x or later or pin to `setuptools<45`"

holta, 2020-03-30:
Thanks to @shanti-bhardwa & @jvonau for reporting this.

Issue appears to be: https://github.com/pypa/setuptools/issues/1458

Install was on Ubuntu 20.04 LTS (arm64 pre-release from http://cdimage.ubuntu.com/ubuntu-server/daily-preinstalled/current/) onto Raspberry Pi 4:

> TASK [kalite : Use pip to install KA Lite static to /usr/local/kalite/venv] **************************************************************************************************************************
fatal: [127.0.0.1]: FAILED! => {"changed": false, "cmd": ["/usr/local/kalite/venv/bin/pip", "install", "--no-cache-dir", "ka-lite-static==0.17.5"], "msg": "stdout: created virtual environment CPython2.7.17.final.0-64 in 1589ms\n  creator CPython2Posix(dest=/usr/local/kalite/venv, clear=False, global=False)\n  seeder FromAppData(download=False, requests=latest, six=latest, colorama=latest, webencodings=latest, certifi=latest, pip=latest, distro=latest, urllib3=latest, chardet=latest, pyparsing=latest, setuptools=latest, retrying=latest, appdirs=latest, idna=latest, pep517=latest, lockfile=latest, msgpack=latest, packaging=latest, wheel=latest, pytoml=latest, progress=latest, distlib=latest, html5lib=latest, pkg_resources=latest, CacheControl=latest, contextlib2=latest, via=copy, app_data_dir=/root/.local/share/virtualenv/seed-app-data/v1.0.1)\n  activators BashActivator,CShellActivator,FishActivator,PowerShellActivator,PythonActivator\n\n:stderr: /usr/local/kalite/venv/lib/python2.7/site-packages/pkg_resources/py2_warn.py:22: UserWarning: Setuptools will stop working on Python 2\n************************************************************\nYou are running Setuptools on Python 2, which is no longer\nsupported and\n>>> SETUPTOOLS WILL STOP WORKING <<<\nin a subsequent release (no sooner than 2020-04-20).\nPlease ensure you are installing\nSetuptools using pip 9.x or later or pin to `setuptools<45`\nin your environment.\nIf you have done those things and are still encountering\nthis message, please comment in\nhttps://github.com/pypa/setuptools/issues/1458\nabout the steps that led to this unsupported combination.\n************************************************************\n  sys.version_info < (3,) and warnings.warn(pre + \"*\"  60 + msg + \"*\"  60)\nTraceback (most recent call last):\n  File \"/usr/local/kalite/venv/bin/pip\", line 5, in <module>\n    from pip._internal.cli.main import main\n  File \"/usr/local/kalite/venv/lib/python2.7/site-packages/pip/_internal/cli/main.py\", line 10, in <module>\n    from pip._internal.cli.autocompletion import autocomplete\n  File \"/usr/local/kalite/venv/lib/python2.7/site-packages/pip/_internal/cli/autocompletion.py\", line 9, in <module>\n    from pip._internal.cli.main_parser import create_main_parser\n  File \"/usr/local/kalite/venv/lib/python2.7/site-packages/pip/_internal/cli/main_parser.py\", line 7, in <module>\n    from pip._internal.cli import cmdoptions\n  File \"/usr/local/kalite/venv/lib/python2.7/site-packages/pip/_internal/cli/cmdoptions.py\", line 25, in <module>\n    from pip._internal.locations import USER_CACHE_DIR, get_src_prefix\n  File \"/usr/local/kalite/venv/lib/python2.7/site-packages/pip/_internal/locations.py\", line 20, in <module>\n    from pip._internal.utils.compat import WINDOWS\n  File \"/usr/local/kalite/venv/lib/python2.7/site-packages/pip/_internal/utils/compat.py\", line 29, in <module>\n    import ipaddr as ipaddress  # type: ignore\nImportError: No module named ipaddr\n"}

jvonau, 2020-03-31:
https://bugs.launchpad.net/ubuntu/+source/python-pip/+bug/1869247

holta, 2020-03-31:
Thanks @jvonau:

This means that the latest python2 on ubuntu-20 has a broken pip/setuptools that can't install virtual environments.

Unclear if and when this will be fixed :/

Very regrettably, this means KA Lite is a No Go on Ubuntu 20.04 for now.

holta, 2020-03-31:
@shanti-bhardwa & @jvonau: can we definitely close this now that PR #2314 is merged?

shanti-bhardwa, 2020-04-01:
got this error with Ubuntu 20.04 on PI 4 and 7.1
TASK [kalite : Use pip to pin older working versions in /usr/local/kalite/venv if Raspbian/Debian > 10 or Ubuntu > 19] *******************************************************************************
fatal: [127.0.0.1]: FAILED! => {"changed": false, "cmd": ["/usr/local/kalite/venv/bin/pip", "install", "--no-use-pep517", "--no-cache-dir", "--no-python-version-warning", "ipaddr", "pip<20", "setuptools<45", "wheel<34"], "msg": "stdout: created virtual environment CPython2.7.17.final.0-64 in 1584ms\n  creator CPython2Posix(dest=/usr/local/kalite/venv, clear=False, global=False)\n  seeder FromAppData(download=False, requests=latest, six=latest, colorama=latest, webencodings=latest, certifi=latest, pip=latest, distro=latest, urllib3=latest, chardet=latest, pyparsing=latest, setuptools=latest, retrying=latest, appdirs=latest, idna=latest, pep517=latest, lockfile=latest, msgpack=latest, packaging=latest, wheel=latest, pytoml=latest, progress=latest, distlib=latest, html5lib=latest, pkg_resources=latest, CacheControl=latest, contextlib2=latest, via=copy, app_data_dir=/root/.local/share/virtualenv/seed-app-data/v1.0.1)\n  activators BashActivator,CShellActivator,FishActivator,PowerShellActivator,PythonActivator\n\n:stderr: /usr/local/kalite/venv/lib/python2.7/site-packages/pkg_resources/py2_warn.py:22: UserWarning: Setuptools will stop working on Python 2\n************************************************************\nYou are running Setuptools on Python 2, which is no longer\nsupported and\n>>> SETUPTOOLS WILL STOP WORKING <<<\nin a subsequent release (no sooner than 2020-04-20).\nPlease ensure you are installing\nSetuptools using pip 9.x or later or pin to `setuptools<45`\nin your environment.\nIf you have done those things and are still encountering\nthis message, please comment in\nhttps://github.com/pypa/setuptools/issues/1458\nabout the steps that led to this unsupported combination.\n************************************************************\n  sys.version_info < (3,) and warnings.warn(pre + \"*\" * 60 + msg + \"*\" * 60)\nTraceback (most recent call last):\n  File \"/usr/local/kalite/venv/bin/pip\", line 5, in <module>\n    from pip._internal.cli.main import main\n  File \"/usr/local/kalite/venv/lib/python2.7/site-packages/pip/_internal/cli/main.py\", line 10, in <module>\n    from pip._internal.cli.autocompletion import autocomplete\n  File \"/usr/local/kalite/venv/lib/python2.7/site-packages/pip/_internal/cli/autocompletion.py\", line 9, in <module>\n    from pip._internal.cli.main_parser import create_main_parser\n  File \"/usr/local/kalite/venv/lib/python2.7/site-packages/pip/_internal/cli/main_parser.py\", line 7, in <module>\n    from pip._internal.cli import cmdoptions\n  File \"/usr/local/kalite/venv/lib/python2.7/site-packages/pip/_internal/cli/cmdoptions.py\", line 25, in <module>\n    from pip._internal.locations import USER_CACHE_DIR, get_src_prefix\n  File \"/usr/local/kalite/venv/lib/python2.7/site-packages/pip/_internal/locations.py\", line 20, in <module>\n    from pip._internal.utils.compat import WINDOWS\n  File \"/usr/local/kalite/venv/lib/python2.7/site-packages/pip/_internal/utils/compat.py\", line 29, in <module>\n    import ipaddr as ipaddress  # type: ignore\nImportError: No module named ipaddr\n"}

PLAY RECAP *******************************************************************************************************************************************************************************************
127.0.0.1                  : ok=166  changed=72   unreachable=0    failed=1    skipped=54   rescued=0    ignored=0

jvonau, 2020-04-01:
Short story, python-pip-whl is not providing ipaddress-0.0.0-py2.py3-none-any.whl is the root cause of this issue. I can manually copy ipaddress-0.0.0-py2.py3-none-any.whl into place from an older install and this error goes away as per the upstream bug report.

holta, 2020-04-01:
Crazy madness!

Is the manual patch likely still nec after Ubuntu 20.04's final release in 3 weeks?

jvonau, 2020-04-06:
> Start-Date: 2020-04-06  06:39:55
Commandline: /usr/bin/unattended-upgrade
Upgrade: python3-pip:arm64 (20.0.2-2, 20.0.2-4), python-pip-whl:arm64 (20.0.2-2, 20.0.2-4)
End-Date: 2020-04-06  06:39:59

holta, 2020-04-08:
@shanti-bhardwa confirms this is fixed, thanks to @jvonau PR #2321.

closed 2020-04-08 by holta

jvonau, 2020-04-10:
evolving story on pip breakage: https://bugs.launchpad.net/ubuntu/+source/python-pip/+bug/1870878