iiab/iiab #2318: Is Kolibri shutting down cleanly when IIAB powers off?

holta, 2020-04-01:
Or is Kolibri possibly/painfully delaying IIAB shutdown?

Thanks @tim-moody for noticing this issue...being discussed in both places here:
- http://community.rachelfriends.org/t/new-image-shutdown-period/1687
- https://community.learningequality.org/t/stop-job-on-raspberry-pi-when-shutting-down/1482

jvonau, 2020-04-01:
I saw the discussion also, we went with the upstream timeouts which are under review. I suspect TimeoutStopSec should be removed/altered, the stock timeout is 90 seconds, if anything it should be increased to allow a clean shutdown of the db backend.

jvonau, 2020-04-15:
Might be fixed on new installs now given the change in upstream change, if the PPA was used then it would be a apt upgrade fix but with the deb location fixed in the install.yml file ./runrole --reinstall kolibri is required.

holta, 2020-04-16:
Related:

#1832 "Tracking the best systemd unit file(s) for Kolibri?"

holta, 2020-04-16:
Recommendation from our http://minutes.iiab.io call today &mdash; thanks to @jvonau:

- Revert (or comment out) IIAB's customized timeouts for Kolibri
- Go with systemd's default of 90 seconds
- In future we can also consider more than 90 seconds if large Kolibri Channels require that

holta, 2020-04-21:
Can this be closed now, now that PR #2343 is merged?

Whereas the other major Kolibri issue remaining, as IIAB 7.1's release is finalized in coming days (this last issue is ongoing for 15 months) might have to be deferred, to the IIAB 7.2 release cycle:

#1532 "http://box/kolibri causes browser to infinitely loop, preventing login (works initially...server reboot does not help...happens with all client OS's & all server OS's)"

jvonau, 2020-04-30:
think we are good, haven't seen any long delays (re)booting.

closed 2020-04-30 by holta