iiab/iiab #2324: Validate Nextcloud 18.0.4 on "2020-04-23"

holta, 2020-04-07:
Per https://github.com/nextcloud/server/wiki/Maintenance-and-Release-Schedule

https://nextcloud.com/changelog/#latest18

jvonau, 2020-04-08:
7.2

holta, 2020-04-08:
Each additional Nextcloud 18.0.x point release is auto-included in IIAB (approx monthly) so this is just one additional spot check we'll do later April to make sure Nextcloud 18.0.4 is working tolerably and decently!

holta, 2020-04-23:
Nextcloud 18.0.4 is now downloadable from places like:

- http://d.iiab.io/packages/latest.tar.bz2
- http://d.iiab.io/packages/nextcloud-18.0.4.tar.bz2
- https://download.nextcloud.com/server/releases/

closed 2020-04-23 by holta

holta, 2020-04-24:
Nextcloud 18.0.4 basic functionality tested as part of IIAB...on multiple different versions of Ubuntu 20.04 LTS:
- amd64 desktop
- amd64 server
- arm64 server