iiab/iiab #2325: DB warnings recently began appearing during IIAB install: "did not set no_log for no_password_changes"

holta, 2020-04-07:
Here is an example, arising while installing `curl d.iiab.io/install-test.txt | sudo bash` on Raspbian Lite on RPi 4:

> TASK [moodle : Create PostgreSQL db user Admin/changeme] ***********************
[WARNING]: Module did not set no_log for no_password_changes
changed: [127.0.0.1]

There are others, that appear very similar to the above.

Ansible discussion:
- https://github.com/ansible/ansible/issues/68106
- https://github.com/ansible/ansible/pull/68116 (closed)
- https://github.com/ansible/ansible/pull/68118 (closed)
- https://github.com/ansible-collections/community.general/pull/17 (merged)

holta, 2020-04-07:
Another example waraning...that we never used to see: (until Ansible 2.9.6?)

> TASK [mediawiki : Create MySQL database user iiab_mediawiki_user with password, and permissions to above db] ***
[WARNING]: Module did not set no_log for update_password
changed: [127.0.0.1]

jvonau, 2020-04-07:
OS and ansible version would be useful

holta, 2020-04-07:
> OS and ansible version would be useful

- Raspbian Lite 2020-02-13 (with all apt updates applied)
- Ansible 2.9.6

jvonau, 2020-04-08:
Also noted on U-20 with ansible 2.9.6 updated on 2020-04-06 from 2.9.4 as distributed by Ubuntu. 
>apt list ansible
Listing... Done
ansible/focal,now 2.9.6+dfsg-1 all [installed]

The original install on 2020-04-01 used 2.9.4 did not have this warning present.

holta, 2020-04-18:
Does anyone know...if these DB warnings went away...now that IIAB's install uses Ansible 2.9.7?

Ref: #2340 Recommend Ansible 2.9.7+, Mandate Ansible 2.8.11+

holta, 2020-04-18:
> Does anyone know...if these DB warnings went away...now that IIAB's install uses Ansible 2.9.7?

No such luck!  FYI these pink warnings continue ("Module did not set no_log for no_password_changes") when using Ansible 2.9.7 like 2.9.6

holta, 2020-05-08:
Not the same issue...but this is reminiscent of a "wontfix" tkt here: (originally from 2019-06-01)

#1714 "MySQL database password-changing failure during fresh install of IIAB [using "sudo iiab"] on Ubuntu 19.04 [and now 18.04 too...whereas 19.10 & Debian 10.x work!] -- Ansible Python 3 needs python3-distutils"

holta, 2020-09-03:
Is this still an issue?  Or can it be closed?

holta, 2020-09-10:
@jvonau summarizes on our weekly IIAB call: a boolean flag can be passed to Ansible to suppress this warning.

holta, 2023-01-01:
No longer an issue.

closed 2023-01-01 by holta