iiab/iiab #2327: MediaWiki: login broken link + broken image

holta, 2020-04-07:
0) I installed BIG-sized IIAB onto Raspbian Lite on RPi 4, using `curl d.iiab.io/install-test.txt | sudo bash`

1) When trying to log in to MediaWiki (http://10.8.0.62/wiki -> "Personal tools" -> "Log in") one is sent to broken link:

   http://10.8.0.62/mwlink/index.php?title=Special:UserLogin&returnto=Main+Page

   (Browser shows "File not found." and nothing else.)

2) This PNG image does not appear (it shows a broken image) on the main MediaWiki page after installing IIAB onto Raspbian Lite on RPi 4:

   http://10.8.0.62/mwlink/resources/assets/poweredby_mediawiki_88x31.png

tim-moody, 2020-04-07:
not sure it works over vpn. can you confirm locally

I was able to login on my local network (not sure what the Personal Tools part is)

tim-moody, 2020-04-07:
are you talking about http://192.168.3.96/wiki/Special:SpecialPages. it has a login link that took me to the login page

holta, 2020-04-07:
> not sure it works over vpn. can you confirm locally

The failures are identical failures over VPN, 192.168.0.x and 172.18.y.z
 
> I was able to login on my local network (not sure what the Personal Tools part is)

The only obvious place to login on the HTML page (http://box/wiki) appears as follows:

> Navigation menu
>
> Personal tools
> - Not logged in
> - Talk
> - Contributions
> - Create account
> - Log in

When clicking on "Log in" I'm sent to http://box/mwlink/index.php?title=Special:UserLogin&returnto=Main+Page which fails (as explained above) with message "File not found." and nothing else on the browser page.

holta, 2020-04-07:
This bug report was based on the erroneous belief that `curl d.iiab.io/install-test.txt | sudo bash` was equivalent to master + PR #2266 as we agreed to during last Thursday's IIAB call.

It turns out (I'm not entirely sure why) that the underlying branch (https://github.com/jvonau/iiab/tree/jv-test) does not correspond precisely to master + #2266 as it was supposed to.  It has an already-resolved months-old MediaWiki bug @tim-moody & I tracked down this afternoon...in short:

Master appears as:
--scriptpath=/{{ mediawiki_symlink }}
https://github.com/iiab/iiab/blob/master/roles/mediawiki/tasks/install.yml#L58

Whereas jv-test appears to contain (at least one?) months-old bug:
--scriptpath=/mwlink
https://github.com/jvonau/iiab/blob/jv-test/roles/mediawiki/tasks/setup.yml#L27

Conclusion: let's not cry over spilt milk, as accidents happens, but also let's all do a better job in future so that testing volunteers' time is not wasted as such in future.

closed 2020-04-07 by holta