iiab/iiab #2332: Cannot update MySQL root password for localhost root accounts.

siriuswapnil, 2020-04-09:
### Hi everyone. I encountered the following error( screenshot) while installing IIAB. Cannot update MySQL root passwords for localhost root accounts. Moreover, on giving command 
`mysql`
It says, :  `access denied : for user`

### Running 
`sudo iiab`
 
for installing IIAB onto Virtualmachine.

### OS: Ubuntu 18.04 (Virtualbox v 5.xx)
### IIAB version : 7.1
[image]

holta, 2020-04-09:
@tim-moody @jvonau do you know what's happening here?  Does this relate to #1714 in any way?

@siriuswapnil can you please try the same thing with the latest daily build of Ubuntu 20.04 LTS?

tim-moody, 2020-04-09:
Swapni, you need to run the install as root. For some reason sudo doesn't
work with setting the mysql password.

I assume the vm is ubuntu server with nothing installed except sshd.

holta, 2020-04-09:
@jvonau believes this is indeed a duplicate ticket, the same as #1714.

But let's leave it open for @siriuswapnil to confirm, Thanks!

siriuswapnil, 2020-04-09:
Yes @holta . I went throught what @tim-moody said. I was also having some issues with my VM. Had to reinstall everything. Thanks for the help.

closed 2020-04-09 by siriuswapnil

holta, 2020-05-07:
Just fyi @siriuswapnil IIAB no longer supports Ubuntu 18.04 as a mainline OS, now that Ubuntu 20.04 has been released.

Apologies!  For more details, please see: https://github.com/iiab/iiab/wiki/IIAB-Platforms