iiab/iiab #2338: Rpi Fast-lane to IIAB Required

georgejhunt, 2020-04-14:
Over the last years, we've created a lot of documentation. As our focus has shifted from big iron towards RPi, we have not pruned our documentation as much as I would like. The Rpi-Imager is a game changer, in that it greatly simplifies the task of presenting a new user with a working prototype.

I'm afraid that the path forward is not clear enough for that newcomer, who might just have clicked on a link on the rpi-imager  from the Raspberry Pi Foundation.

I think our installation instructions should assume the Rpi hardware, without hiding the fact that our software has an upper end and growth path to other hardware and OS's. Within that context, perhaps only a small amount of pruning is required.

I'd like to reorganize our IIAB wiki, renaming previous release notes, so that they fall to the bottom of the list on the right side of the wiki page.

I think we need a video, or more than one, which highlights installation of rpi-imager, and a headless installation of a desktop raspbian.

I don't think the fast lane I'm advocating is essential for release 7.1. But I do think we need to improve our documentation, before  we ask the Raspberry Pi Foundation to link to our released images.

holta, 2020-04-17:
Background context... "Raspberry Pi Imager" (introduced 2020-03-05) promises to get rid of much/most of the installation pain:
https://www.raspberrypi.org/blog/raspberry-pi-imager-imaging-utility/

tim-moody, 2020-04-17:
One of the problems with the github wiki is that it lists pages in alpha order rather than some other more logical order, so numbers (of releases) come before other instructions. Not sure what can be done about that other than putting numbers on the start of titles or moving to a blog type environment.  I agree with the issue.

georgejhunt, 2020-05-03:
I've created a 12 minute video about using the RPi imager:
https://www.youtube.com/watch?v=JWliUi2e8w8&feature=youtu.be

holta, 2020-07-11:
Progress is ongoing as @georgejhunt & I revise these docs...in recent days and in upcoming weeks/months:

- https://github.com/iiab/iiab/wiki/Raspberry-Pi-Images:-Summary
- https://github.com/iiab/iiab/wiki/Raspberry-Pi-Images:-Features
- https://github.com/iiab/iiab/wiki/Raspberry-Pi-Images:-Installation

Marazm1980, 2020-07-11:
Wspaniałe.

Marazm1980, 2020-07-11:
Pl.wspaniale

holta, 2020-10-25:
Related: #2082 "Up-to-date raspberry pi images [Node.js 12.x (similar to 10.x) has unofficial builds for ARMv6 e.g. RPi Zero W]"