iiab/iiab #2347: php7.4-stem support for Ubuntu 20.04 LTS?  On which architectures if so?

holta, 2020-04-19:
With PHP 7.4 becoming mainline as part of Ubuntu 20.04 starting this coming Thursday, should we consider adding support for some subset of {aarch64, rpi, x64} ?

( e.g. to enable WikiHow search in Spanish, via http://d.iiab.io/packages/ )

Background:
- https://github.com/iiab/iiab/blob/master/roles/www_base/tasks/main.yml#L9-L10
- https://github.com/iiab/iiab/blob/master/roles/www_base/tasks/php-stem.yml

Most recently @tim-moody built these at http://d.iiab.io/packages/ :
- php7.3-stem.aarch64.tar
- php7.3-stem.rpi.tar
- php7.3-stem.x64.tar

Refs: #2123, PR #2198, PR #2269, PR #2294, #2345

tim-moody, 2020-04-19:
After #2345 this is a lower priority, but I think to start with php7.4-stem.aarch64.tar and then php7.4-stem.x64.tar once 20.04 is released.

tim-moody, 2020-04-25:
php7.4-stem.aarch64.tar  and php7.4-stem.x64.tar uploaded to d.iiab.io

needs testing

holta, 2020-04-28:
Good enough for now, with PR #2349 now merged.