iiab/iiab #2348: KA-Lite serving to XO laptops

llovemixo, 2020-04-19:
IIAB version 7.0 with KA-Lite on Pi Zero W
Client :  XO laptop  with free RpmFusion for MP4 videos; Browse web browser. 

Recent versions of software on OLPC XO laptops fail to start MP4 video content files when KA-Lite>Learn is reached with http://box.lan:8008 .
OLPC versions 13.2.10 and 13.2.11 fail.

The purpose of this post is to point out that older versions 13.2.8 and 13.2.9 still work and provide a way for XO laptops to access the Khan Academy MP4 videos.  The Browse web browser is not compatible with the Dynamic Menuing.

Browse versions are 157.3, 157.4, and 157.5, but do not seem to be factors.  Non-working 157.5 of 13.2.11 was replaced with working 157.4 of 13.2.9  but did not change non-working status.
Tested on XO-1.5   Works with XO-1, but is too slow to be useful there.

The failure appears to be in the handshaking between Browse and KA-Lite's js video viewer.

holta, 2020-04-19:
> The purpose of this post is to point out that older versions 13.2.8 and 13.2.9 still work and provide a way for XO laptops to access the Khan Academy MP4 videos. The Browse web browser is not compatible with the Dynamic Menuing.

Do the latest IIAB 7.1 pre-releases presumably work to deliver KA Lite in the same way here, just like your IIAB 7.0?

llovemixo, 2020-04-21:
Regrets.  Downloading KA-lite leaves me short in bytes until end of month.
Made pitch to local school for KA-lite server sneaker net with multiplexing by time.  Shoebox IIAB for 2 hours in yard of each student and then teacher to evaluate. However, school got lots of money for extra laptops and home internet service.  Was planning 20 units.

closed 2020-04-21 by llovemixo