iiab/iiab #2361: install wipes my openvpn_handle

tim-moody, 2020-04-25:
during install my openvpn_handle was renamed to openvpn_handle.6756.2020-04-25@16\:58\:40~ (by backup I assume) and the original was wiped (no value in local vars). 

the template directive should be force: no

holta, 2020-04-25:
FWIW Ansible variable openvpn_handle has always been the "master location" to store this value.

It sounds like you're proposing to change the master location to be /etc/iiab/openvpn_handle instead (if it exists!)

FYI things like /usr/bin/iiab-support (sourced from https://github.com/iiab/iiab/blob/master/iiab-support) would have to be re-architected if indeed this design change moves forward.

tim-moody, 2020-04-25:
I guess I never hit this before, because I add a handle to images after they are built, but replacing a non-empty openvpn_handle file with an empty one doesn't make much sense to me.

jvonau, 2020-04-30:
> during install my openvpn_handle was renamed to openvpn_handle.6756.2020-04-25@16:58:40~ (by backup I assume) and the original was wiped (no value in local vars).
> 
> the template directive should be force: no

Any reason to force a freshly created image back though 1-prep? Or to use runrole --reinstall openvpn? install.yml is where the template is copied into place. You didn't by chance remove or alter /etc/iiab/iiab_state.yml? Provisioning a new image should not really need to reinstall anything, fiddle a few files/services maybe but no real need to start at stage 1. Can you go into more detail about your provisioning routine, might be able to understand better what steps you are taking.

tim-moody, 2020-04-30:
I have started creating pre-install images and probably put a handle in the prep code. You could also need to do a software upgrade. What would be the reason to overwrite a user's file with an empty file under any circumstances? What would break if the one line change I proposed were applied?

holta, 2020-05-15:
If as part of the IIAB 7.2 release cycle we rethink and redesign how-and-where OpenVPN's handle should be stored (thruout the complete life cycle of diverse IIAB machines' installation->implementation->ops->renaming/moving to another school, etc) we should keep in mind not only the future[*] (but also the past/evolving mechanics) of this de facto IIAB naming scheme.  Here is one example:

```
# openvpn_handle variable can no longer be left completely undefined [as] of August 2018 (EMPTY STRING "" IS TOLERATED, in which case OpenVPN server should use /etc/iiab/uuid in lieu of the handle)
```
From: https://github.com/iiab/iiab/blob/master/roles/openvpn/tasks/main.yml#L45

[*] Important future questions will apply just as well -- around what various volunteers truly want to support and invest in here as we navigate this -- as there are very real personal/financial/privacy/security costs to bear here.  Even more so when VPN is not a broadly public support forum (by definition it's almost the opposite, a non-public means of private collaboration for small teams, who often don't understand the security implications at all...!)

tim-moody, 2020-05-16:
I agree completely with the requirement that /etc/iiab/openvpn_handle must exist and have a value. I also agree that if it does not exist (or is empty) that the value of the openvpn_handle variable should be written to it. But I think in principle systems should not overwrite user data without permission.

Also note that there is nothing at present the prevents duplicate handles. In fact image duplication will generally produce duplicates. FWIW I have tried to write the sd card serial number into /etc/iiab on cmdsrv startup which will mean there is a unique identifier. However the algorithm to find the serial number is fraught and can take minutes to run on some devices because the path can not be relied upon. The relationship to /etc/iiab/uuid also needs to be considered.

jvonau, 2020-05-17:
> I agree completely with the requirement that /etc/iiab/openvpn_handle must exist and have a value. I also agree that if it does not exist (or is empty) that the value of the openvpn_handle variable [...]

Would that not be part of the cleanup script's job to correct?
> FWIW I have tried to write the sd card serial number into /etc/iiab on cmdsrv startup which will mean there is a unique identifier. However the algorithm to find the serial number is fraught and can [...]

The UUID is re-created on boot [if absent](https://github.com/iiab/iiab/blob/master/roles/2-common/templates/iiab-startup.sh)

closed 2021-08-23 by holta