iiab/iiab #2366: When logged into Admin Console Nextcloud tries to use admin credentials to login.

floydianslips, 2020-04-26:
### Expected behavior and actual behavior.
Nextcloud(NC) should load upon menu click and show login screen. When logged into Admin Console as admin user (iiab-admin) clicking on NC menu button loading takes ~1 minute. This is caused by NC attempting to login using admin credentials: iiab-admin a minimum of 9 times. This also creates a message on the NC login page:

> We have detected multiple invalid login attempts from your IP. Therefore your next login is throttled up to 30 seconds.

### Steps to reproduce the problem.
1. Prior to logging into Admin Console, click NC menu link and notice the time it takes to load NC.
2. Open an new tab and log into Admin Console using admin login credentials. 
3. Go back to IIAB menu and open NC again and notice considerable delay. 

### Specifications like the version of the IIAB, operating system version, or hardware details.
- Raspberry Pi 4 2GB RAM
- Ubuntu 20 LTS
- Entry from NC log: 
`{"reqId":"BT6wmWayI2C3Q85Yk7KS","level":2,"time":"2020-04-26T23:00:04+00:00","remoteAddr":"192.168.1.250","user":"--","app":"core","method":"GET","url":"/nextcloud/apps/theming/favicon?v=0","message":"Login failed: 'iiab-admin' (Remote IP: '192.168.1.250')","userAgent":"Mozilla/5.0 (X11; Linux x86_64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/81.0.4044.113 Safari/537.36","version":"18.0.4.2"}`

holta, 2020-04-26:
Seems...very much related to: #1532 "http://box/kolibri causes browser to infinitely loop, preventing login (works initially...server reboot does not help...happens with all client OS's & all server OS's)"

tim-moody, 2020-04-28:
I think this happens in incognito mode:

Joshua, 4:14 PM
I can make the loop happen in Chrome when not in incognito mode. Firefox doesn't seem to be taking the bait however. 

Tim, 4:17 PM
then you need to give me the exact steps as I am not able to reproduce. btw do you save admin pw in chrome?

Joshua, 4:33 PM
I don't save it

Joshua, 4:41 PM
I don't know. I did a chrome://restart and now it is not doing the loop. I am at a loss.

tim-moody, 2020-04-28:
https://stackoverflow.com/questions/5957822/how-to-clear-basic-authentication-details-in-chrome suggests that in incognito mode the same credentials are sent to every url on a site and that is my experience as well. Once the browser has http basic auth credentials it sends them everywhere.

tim-moody, 2020-04-28:
For php sites, such as NextCloud, the phenomenon of the browser sending a single set of credentials to all sites may be true for regular browser mode as well:

I created /test1 and /test2 both with the same php file:

```
<?php
phpinfo(INFO_VARIABLES);
?>

```

I declared both secure:

```
location /test1 {
   auth_pam                "Secure zone";
   auth_pam_service_name   "pwauth";

}
location /test2 {
   auth_pam                "Secure zone";
   auth_pam_service_name   "pwauth";

}
```

As soon as I log into test1 I am automatically logged into test2

tim-moody, 2020-04-28:
BUT they take separate usernames and passwords

if I change /etc/nginx/conf.d/test.conf as follows:

```
location /test1 {
   auth_pam                "Secure zone";
   auth_pam_service_name   "pwauth";

}
location /test2 {
   auth_pam                "Another zone";
   auth_pam_service_name   "pwauth";

}

```

holta, 2020-04-28:
Does PR https://github.com/iiab/iiab-admin-console/pull/309 (now merged) partially/fully solve #2366 here and/or #1532 ?!

tim-moody, 2020-04-28:
>Does PR iiab/iiab-admin-console#309 (now merged) partially/fully solve #2366 here and/or #1532 ?!

Don't know!! Probably not?!

tim-moody, 2020-04-29:
@floydianslips can you retest with  iiab/iiab-admin-console#309 applied

tim-moody, 2020-04-30:
> These comments where towards #1532. I can still reproduce this Nextcloud login error in Chrome and Firefox without using Incognito or Private browsing.
> 
> To reproduce just log in to Admin Console first and then try to navigate to Nextcloud.

I have now seen the same behavior.

holta, 2020-05-05:
Huge thanks @floydianslips & @tim-moody for working on this!

I'm moving this to IIAB 7.2's milestone so we can hopefully release [IIAB 7.1](https://github.com/iiab/iiab/wiki/IIAB-7.1-Release-Notes) very soon now...?

holta, 2020-09-10:
I believe @tim-moody solved this months ago.

closed 2020-09-10 by holta