iiab/iiab #2367: vnStats playbook cannot be rerun on a VM w/o LAN?

holta, 2020-04-27:
Thanks @jvonau who is looking into this:

On VM 191-u20-srv-BIG-0423 I clicked ICO and 2.5min later http://10.8.0.26/admin -> Utilities -> Display Job Status shows:

> TASK [vnstat : Create database for LAN to collect vnStat data if not appliance config] ***
fatal: [127.0.0.1]: FAILED! => {"changed": true, "cmd": "/usr/bin/vnstat -i none", "delta": "0:00:00.005928", "end": "2020-04-27 17:48:08.988125", "msg": "non-zero return code", "rc": 1, "start": "2020-04-27 17:48:08.982197", "stderr": "", "stderr_lines": [], "stdout": "Error: Interface \"none\" not found in database.", "stdout_lines": ["Error: Interface \"none\" not found in database."]}
>
> PLAY RECAP *********************************************************************
127.0.0.1 : ok=239 changed=57 unreachable=0 failed=1 skipped=106 rescued=0 ignored=0

(Both /opt/iiab/iiab and /opt/iiab/iiab-admin-console have the very latest code.)

iiab-diagnostics output: http://dpaste.com//35AQEAJ

Here was the original install 4 days ago, according to /opt/iiab/iiab/iiab-install.log :

> 2020-04-23 21:38:04,084 p=8538 u=root n=ansible | TASK [vnstat : Create database for LAN to collect vnStat data if not appliance config] ***
> 2020-04-23 21:38:04,115 p=8538 u=root n=ansible | skipping: [127.0.0.1]

This line https://github.com/iiab/iiab/blob/master/roles/vnstat/tasks/main.yml#L19 :

```
when: iiab_lan_iface is defined
```

...could be revised to:

```
when: iiab_lan_iface is defined and iiab_lan_iface != "none"
```

...but that might be papering over a deeper problem?

_Is it possible [iiab-from-console.yml](https://github.com/iiab/iiab/blob/master/iiab-from-console.yml) is setting vars differently from the original [iiab-stages.yml](https://github.com/iiab/iiab/blob/master/iiab-stages.yml) ??_

jvonau, 2020-04-27:
[image]

tim-moody, 2020-04-27:
#2362 was not applied

PR #2369 (2020-04-27): vnStat workaround for #2367
Resolves #2367 "vnStat playbook cannot be rerun on a VM w/o LAN? Is iiab-from-console.yml part of the issue?"

A more thorough solution might be in order later?
closed 2020-04-27 by holta