iiab/iiab #2370: runrole --reinstall doesn't

tim-moody, 2020-04-28:
With osm-vector-maps installed and enabled, run runrole osm-vector-maps --reinstall. It should refresh the install with any newly pulled code but it doesn't. It is necessary to remove  osm_vector_maps_installed: True from iiab_state.yml in order for it to work.

jvonau, 2020-04-28:
Can't reproduce works here.. 
buntu@box:/opt/iiab/iiab$ sudo ./runrole --reinstall osm-vector-maps
        "ansible_python": {
            "executable": "/usr/bin/python3",
            "type": "cpython",
        "ansible_python_version": "3.8.2",
        "ansible_selinux_python_present": true,

PLAY [all] ********************************************************************************************

TASK [Gathering Facts] ********************************************************************************
ok: [127.0.0.1]

TASK [0-init : ...IS BEGINNING ============================================] **************************
ok: [127.0.0.1]

TASK [0-init : Set first_run flag] ********************************************************************
skipping: [127.0.0.1]

TASK [0-init : Set top-level variables from local_facts for convenience] ******************************
ok: [127.0.0.1]

TASK [0-init : Create IIAB tools and /etc/iiab/iiab.ini, if first_run] ********************************
skipping: [127.0.0.1]

TASK [0-init : Copy /opt/iiab/iiab/scripts/iiab-diagnostics to /usr/bin/iiab-diagnostics] *************
ok: [127.0.0.1]

TASK [0-init : Create globally-writable directory /etc/iiab/diag so non-root users can run iiab-diagnostics] ***
ok: [127.0.0.1]

TASK [0-init : Re-read local_facts.facts from /etc/ansible/facts.d] ***********************************
ok: [127.0.0.1]

TASK [0-init : Pre-check that IIAB's "XYZ_install" + "XYZ_enabled" vars (1) are defined, (2) are boolean-not-string variables, and (3) contain plausible values.  Also checks that "XYZ_install" is True when "XYZ_installed" is defined.] ***
included: /opt/iiab/iiab/roles/0-init/tasks/validate_vars.yml for 127.0.0.1

TASK [0-init : Set vars_checklist for 53 + 53 + up-to-53 vars ("XYZ_install" + "XYZ_enabled" + "XYZ_installed") to be checked] ***
ok: [127.0.0.1]

TASK [0-init : Assert that 53 "XYZ_install" vars are all... defined] **********************************
snip
ok: [127.0.0.1] => (item=calibreweb)

TASK [0-init : Assert that 53 "XYZ_enabled" vars are all... defined] **********************************
snip

TASK [0-init : Assert that 53 "XYZ_install" vars are all... type boolean (NOT type string, which can invert logic!)] ***
snip

TASK [0-init : Assert that 53 "XYZ_enabled" vars are all... type boolean (NOT type string, which can invert logic!)] ***
snip
TASK [0-init : DISALLOW "XYZ_install: False" WITH "XYZ_enabled: True" ...for all 53 var pairs] ********
snip
TASK [0-init : DISALLOW "XYZ_install: False" WHEN "XYZ_installed is defined" IN /etc/iiab/iiab_state.yml ...for up-to-53 var pairs] ***
snip

TASK [0-init : apache_install is currently 'True' and apache_enabled is currently 'False'.  Now let's set 'apache_install: True' and 'apache_enabled: True' if NGINX is set not to be enabled OR if any of {Elgg, Lokole, Moodle, Node-RED} are set to install.] ***
skipping: [127.0.0.1]

TASK [0-init : Set exFAT_enabled if xo_model != "none"] ***********************************************
skipping: [127.0.0.1]

TASK [0-init : Do we have a gateway? If so set discovered_wan_iface: wlan0] ***************************
ok: [127.0.0.1]

TASK [0-init : Verify gateway active: ping -c4 192.168.0.1] *******************************************
changed: [127.0.0.1]

TASK [0-init : If so, set gw_active, iiab_wan_iface to wlan0] *****************************************
[... 124 lines cut ...]
ok: [127.0.0.1]

TASK [osm-vector-maps : Add 'osm_vector_maps_installed: True' to /etc/iiab/iiab_state.yml] ************
changed: [127.0.0.1]

TASK [osm-vector-maps : Enable/Disable/Reload NGINX for OSM, if nginx_enabled] ************************
included: /opt/iiab/iiab/roles/osm-vector-maps/tasks/nginx.yml for 127.0.0.1

TASK [osm-vector-maps : Enable http://box/maps & http://box/osm-vector-maps via NGINX, by installing /etc/nginx/conf.d/osm-vector-maps-nginx.conf from template] ***
ok: [127.0.0.1]

TASK [osm-vector-maps : Disable http://box/maps & http://box/osm-vector-maps via NGINX, by removing /etc/nginx/conf.d/osm-vector-maps-nginx.conf] ***
skipping: [127.0.0.1]

TASK [osm-vector-maps : Reload 'nginx' systemd service] ***********************************************
changed: [127.0.0.1]

TASK [osm-vector-maps : Add 'osm-vector-maps' variable values to /etc/iiab/iiab.ini] ******************
ok: [127.0.0.1] => (item={'option': 'name', 'value': 'OSM Vector Maps'})
ok: [127.0.0.1] => (item={'option': 'description', 'value': '"OpenStreetMap is like Google Maps but better, for schools especially, as it works offline and avoids all the advertising.  Download detailed \'vector maps\' for an entire continent, or the entire world!  Also includes 10+ zoom levels of satellite imagery!"'})
ok: [127.0.0.1] => (item={'option': 'install', 'value': True})
ok: [127.0.0.1] => (item={'option': 'enabled', 'value': True})

PLAY RECAP ********************************************************************************************
127.0.0.1                  : ok=47   changed=7    unreachable=0    failed=0    skipped=15   rescued=0    ignored=0

jvonau, 2020-04-28:
ubuntu@box:/opt/iiab/iiab$ sudo ./runrole
>Usage: ./runrole name of role
Usage: ./runrole --reinstall name of role
>
>Optional 2nd parameter is full PATH/FILENAME for logging.
If omitted, <current directory>/iiab-debug.log is used.

--reinstall is in the wrong place.

holta, 2020-04-28:
@jvonau since this is a common mistake (@tim-moody is not the 1st one to put the `--reinstall` flag in the wrong place) can we get `./runrole` to issue a loud warning when such incorrect syntax is attempted?

tim-moody, 2020-04-28:
retesting:
on a vm with osm_vector_maps not installed I issued
./runrole osm_vector_maps [I accidentally typed the name with underscores as I was confused between the role name and the variable name]
```
[WARNING]: While constructing a mapping from /etc/iiab/local_vars.yml, line 1, column 1, found a duplicate dict key (osm_vector_maps_install). Using last defined value only.
[WARNING]: While constructing a mapping from /etc/iiab/local_vars.yml, line 1, column 1, found a duplicate dict key (osm_vector_maps_enabled). Using last defined value only.
ERROR! the role 'osm_vector_maps' was not found in /opt/iiab/iiab/roles:/root/.ansible/roles:/usr/share/ansible/roles:/etc/ansible/roles:/opt/iiab/iiab

The error appears to be in '/opt/iiab/iiab/run-one-role.yml': line 13, column 7, but may
be elsewhere in the file depending on the exact syntax problem.

The offending line appears to be:

    - { role: 0-init, tags: ['0-init'] }
    - { role: "{{ role_to_run }}", tags: ['run'] }
      ^ here
We could be wrong, but this one looks like it might be an issue with
missing quotes. Always quote template expression brackets when they
start a value. For instance:

    with_items:
      - {{ foo }}

Should be written as:

    with_items:
      - "{{ foo }}"

```

tim-moody, 2020-04-28:
works with the right name. now ./runrole osm-vector-maps  --reinstall
TASK [osm-vector-maps : Install OSM Vector Maps if 'osm_vector_maps_installed' not defined, e.g. in /etc/iiab/iiab_state.yml] ****************************************************************
skipping: [127.0.0.1]

but ./runrole  --reinstall osm-vector-maps 

TASK [osm-vector-maps : Install OSM Vector Maps if 'osm_vector_maps_installed' not defined, e.g. in /etc/iiab/iiab_state.yml] ****************************************************************
included: /opt/iiab/iiab/roles/osm-vector-maps/tasks/install.yml for 127.0.0.1

So my issue was placement of --reinstall

holta, 2020-04-28:
I recommend we introduce a much louder warning during the IIAB 7.2 release cycle if possible, to catch this very common syntax error.

Or honor it as correct syntax?  (If the "customer is always right" ;)

Meantime, @jvonau suggests that ./runrole could at least broadcast to all users:

> "if you are using --reinstall it must appear immediately after runrole"

holta, 2020-04-29:
Further doc improvements are needed to make the [./runrole](https://github.com/iiab/iiab/blob/master/runrole) command and its purpose much more clear to the new implementers (#2096).

But on this narrow issue of warning operators/implementers when they place the flag `--reinstall` in the wrong place on the command-line, PR #2374 (now merged) is good enough for now.

closed 2020-04-29 by holta