iiab/iiab #2371: install stage arithmetic could be more flexible

tim-moody, 2020-04-28:
When an aggregate role finishes it sets last stage to its stage. It could only do this if its stage is greater than the current last stage. This would allow an earlier stage to be rerun without requiring all subsequent stages to be rerun.

For example:

```
- name: Recording STAGE 2 HAS COMPLETED ==========================
  lineinfile:
    path: "{{ iiab_env_file }}"
    regexp: '^STAGE=*'
    line: 'STAGE=2'
  when: {{ iiab_stage }} < 2

```

jvonau, 2020-04-29:
What is the use case for (re)running the numbered meta roles via runrole or runroles? Just because runrole allows such nonsense doesn't make that a design flaw, when there were checks in place to prevent such usage you argued against the check being in place.

tim-moody, 2020-04-29:
never mind. I can use runroles.

closed 2020-04-29 by tim-moody