iiab/iiab #2375: RFQ: Adaptive Network Service

tim-moody, 2020-04-29:
Here are some ideas I've been thinking about for awhile. I notice that jvonau:ap0+wifi_up_down2 is starting in this direction so I'd like to suggest going even further:

To configure networking what about a different approach? Instead of tweaking a bunch of configuration files and trying to make third party software do the right thing using commands at install time, just let the network come up in its base configuration at boot, which includes at least one way of reaching the server over the network. When its done, the adaptive-network service runs, examines the environment, looks at the variables that define the desired network, and reconfigures it with cli on the fly. It does each step in the order needed and only for hardware present. 

If there is a problem, nothing is carved in stone. If adaptive-network fails it could turn itself off either with disable or a flag somewhere. Failure means there is no way to reach the server. To change a running server, you just change the variables and restart the service. There are probably different versions of this service for different platforms and the appropriate one is installed with ansible.

jvonau, 2020-04-29:
Thank for noticing ap0+wifi_up_down2, it's a working prototype on U-20. The issue at hand due to firmware limitations is one of keeping the channels in sync between wlan0 for upstream use and ap0 for AP use. Currently on raspbian the dhcpcd hook does the editing of hostapd.conf on the fly but doesn't restart the service as a restart deletes ap0. We must have the device cloned before starting wpa_ supplicant or nothing works, a reboot aligns the channels up. There is no hook like that currently for ubuntu because I was thinking with the cloning split apart from starting hostapd we might be able to (re)start hostapd once the channel was adjusted in hostapd.conf on the fly in dhcpcd's hooks or one written for ubuntu just haven't explored that far yet. The test-wifi routine and related service is intended for boot time staging so the services come up in a sane order. Come to think about it that would be the place to slip the netplan country= workaround in. This all works ok with the upstream channel not changing, needs testing for when the wifi channel does change.

tim-moody, 2020-04-30:
no meaningful comments. closing

closed 2020-04-30 by tim-moody