iiab/iiab #2377: WiFi hotspot disappeared after rebooting a laptop containing a fresh install of IIAB on Ubuntu 20.04 Desktop (NetworkManager)

holta, 2020-04-29:
193-u20-desk-PR2337-MIN-0429 = 10.8.0.66 is a real laptop where PR #2337 was just installed via a USB-to-Ethernet dongle...and 'wifi_up_down: False' in /etc/iiab/local_vars.yml : http://dpaste.com//3Y0B5T9

Initially the WiFi AP (access point, hotspot) worked, allowing nearby phones to browse to http://box

After reboot (as requested at the end of IIAB's install) this WiFi AP was no longer on the air.

@jvonau explored the situation which appears to involve a https://netplan.io bug (affecting NetworkManager here, on Ubuntu 20.04 LTS Desktop) ?

> handle_probe_req: send failed in 'systemctl status hostapd' ?

Running `systemctl restart hostapd` was not enough...as @jvonau writes:

> iiab-admin@box:~$ sudo netplan try
br0: reverting custom parameters for bridges and bonds is not supported
>
> Please carefully review the configuration and use 'netplan apply' directly.

```
iiab-admin@box:~$ sudo cat /etc/netplan/01-iiab-config.yaml 
network:
  version: 2
  renderer: NetworkManager
  ethernets:
    enx001c49010c71:
      dhcp4: yes
      optional: true

  bridges:
      # the key name is the name for virtual (created) interfaces
   br0:
#        # IDs of the components; switchports expands into multiple interfaces
     interfaces: wlan0
     dhcp4: no
     dhcp6: no
#            # should really compute the netmask like the wan side
     addresses: [172.18.96.1/19]
#            # there should really not be a gateway present testing to prove
#            gateway4: 172.18.96.1
#            nameservers:
#                addresses: [172.18.96.1]
     parameters:
       forward-delay: 0
       stp: false
#
```

> interfaces: wlan0   -> interfaces: [ ]  probably  worked with netplan apply but the vpn was cut off..

[VPN restored...]

```
iiab-admin@box:~$ nmcli c
NAME                     UUID                                  TYPE      DEVICE          
netplan-enx001c49010c71  4e99d19e-b271-33f5-9354-09e1a54d946b  ethernet  enx001c49010c71 
tun0                     123aedfb-a02f-4f0d-afa8-f79ec9e67f89  tun       tun0            
bridge-br0               9a9cb673-d04e-44b4-9246-936532fcad21  bridge    --              
netplan-br0              00679506-5c05-3c3d-bdfe-474849762078  bridge    --              
netplan-wlan0            8cc8469d-6527-3c8a-915d-e147f79deae0  ethernet  --
```

> so netplan did not activate either /run/NetworkManager/system-connections/ or /etc/NetworkManager/system-connections/

> sudo nmcli c up bridge-br0  kicked it up...

jvonau, 2020-05-01:
Major points - minor changes
020668e forces the use of systemd-networkd to always handle br0 allowing hostapd to start and bind with br0 before allowing NetworkManager to start.

242b9cf and 3958c65 deals with wifi_up_down False, NetworkManager aggressively wanting total control over the network devices interfering with hostapd

863bd52 deals with wifi_up_down True, NetworkManager aggressively wanting total control over the network devices interfering with hostapd

fa19d46 same routines used on RPi hardware are needed on amd64 based hardware with a wifi chipset present.

holta, 2020-05-02:
@jvonau should we close this now, or keep it open for further resolution/cleanup ?

jvonau, 2020-05-03:
think this can be closed

closed 2020-05-03 by holta