iiab/iiab #2379: Explain how to connect IIAB's internal Wi-Fi to the Internet: how to enter SSID & password

holta, 2020-04-30:
1) Can the language be improved within this screenshot (yellow areas) &mdash; so that non-technical people understand this is not about IIAB's knowledge hotspot &mdash; but rather about their own pre-existing Wi-Fi connection to the Internet?  (If they have one!)

   As IIAB's knowledge hotspot is configured in an entirely different tab, e.g. within Admin Console's "Configure" tab, or using commands like `iiab-hotspot-on` and `iiab-hotspot-off`.

   During our call this morning everyone suggested we strive for language that's consistent across all of IIAB, to eliminate much confusion.  Language like "Upstream Wi-Fi" could be considered, however that is unfamiliar language to many rural/offline people.  Language like "Your building's Wi-Fi connection to the Internet" might be much more clear.  Or if it's possible to be more both clear AND concise (ideally, but this might not be possible?) then perhaps language similar to the following might work...

   - "Wi-Fi to Internet" on the blue button on the left?
   - "Wi-Fi to Internet Name (SSID)" for the 1st textfield at bottom?
   - "Wi-Fi to Internet Password" for the 2nd textfield at bottom?
   - Is there a better word than "Credentials" for less literate folk, on the blue button at bottom?

2. Mentioning "Hotspot in ON" just above there leads to further confusion.  If it needs to be there, perhaps explain better?  If so saying something like: "IIAB's internal hotspot is ON.  Use the 'Configure' tab to rename it etc."

[image]

3. Sometimes the "Wifi control" blue button on the left does not appear at all, e.g. when IIAB is installed on Ubuntu 20.04 LTS Desktop on an actual PC, in cases like #2377. 

   What would it take to fix that, allowing the operator/implementer to enter the SSID & its password for Ubuntu as well &mdash; at least for Ubuntu Server at the minimum  &mdash; e.g. if it's detemined that Ubuntu Desktop users should instead use the Wi-Fi GUI built-in to the top-right of their Ubuntu Desktop screen?

tim-moody, 2020-05-01:
>so that non-technical people understand this is not about IIAB's knowledge hotspot
As IIAB's knowledge hotspot is configured in an entirely different tab, e.g. within Admin Console's "Configure" tab, or using commands like iiab-hotspot-on and iiab-hotspot-off.

It is about the internal hotspot. it switches the internal wifi between hotspot and upstream. This is pre ap0. the hotspot is named elsewhere but this button does the switch, which is temporary.

>Is there a better word than "Credentials"

There's probably room to make it "Login/Password"

>Mentioning "Hotspot is ON" just above there leads to further confusion

This refers to the internal hotspot. I tried to avoid upstream/downstream as I doubt many will understand that. Internal/External might work. I tried to avoid the words lan and wan because lan can mean up or down.

>sometimes the "Wifi control" blue button on the left does not appear at all

It only appears for a rpi as I'm not sure it will work elsewhere.

holta, 2020-05-02:
"Dual WiFi" is implemented experimentally by PR #2380 (now merged) on Raspberry Pi and many regular PC laptops too!

DRAFT instructions for newcomer IIAB implementers just getting started...are here:

https://github.com/iiab/iiab/wiki/IIAB-7.1-Release-Notes#whats-new

Related:

#796 "Rapid networking & other IIAB config changes that can't wait ~20min for Ansible"
#2096 "NGINX & ./runrole & ./iiab-network [Dual WiFi] & IIAB Maps need approachable docs for newbie/legacy implementers"

jvonau, 2020-05-03:
> > sometimes the "Wifi control" blue button on the left does not appear at all
> 
> It only appears for a rpi as I'm not sure it will work elsewhere. The only thing new is support for u20 on rpi, which is mostly untested.

Works like a charm
>ubuntu@box:~$ brctl show 
bridge name	bridge id	STP enabled	interfaces 
br0	8000.02bb94a48605	no	ap0 
bnep0

holta, 2020-05-08:
Related:

- [Netplan](http://netplan.io) is incomplete (it currently lacks support to specify country/frequencies and hidden SSID's) but these upstream docs largely summarize how to set SSID/password for the WiFi upstream of IIAB: https://github.com/iiab/iiab/wiki/IIAB-Platforms#operating-systems
- #2382 "merged 2380 [how to configure upstream & downstream WiFi?]"

holta, 2020-05-09:
Related:

PR iiab/iiab-admin-console#318 "Add wpa support of U20"

holta, 2020-05-15:
There's a lot of progress here.  And at the same time the current language at http://box/admin -> Control > Wifi Control is still very confusing (as discussed during today's IIAB call off of http://minutes.iiab.io "external WiFi router" is highly ambiguous).  What is meant and what is possible here seems to get extremely confusing, extremely fast.

Worse, words like "upstream" and "downstream" seem to confuse a LOT of people, so we need to try to avoid engineering terms and technical jargon, as much as possible anyway...

Call consensus is that we're hoping there might be a way to simplify/normalize the language into much more approachable leyman's terms, that's nearly universal across all IIAB docs/videos/UX etc.  Ideally!?  If so, IIAB 7.1 (and if necessary further release candidates) in coming days might start down that road, or at least do a lot better which would would be incredible, so I'll try to call @tim-moody if he has time to co-evolve us a bit further down the road (of this inevitably painful task) in coming days.

tim-moody, 2020-05-15:
>(as discussed during today's IIAB call off of http://minutes.iiab.io "external WiFi router" is highly ambiguous)

is not even mentioned in the minutes to today's call. @holta claim is that this term is 'ambiguous' even though 99% of installs do not use an external lan router and that router can not be configured using Admin Console.

The word Internet is not regarded as confusing even though there will be many circumstances in which the upstream router does not provide internet access.

holta, 2020-05-16:
1) @tim-moody mentions that filenames mentioned in /etc/netplan (e.g. 00-installer-config.yaml, /etc/netplan/01-iiab-config.yaml, 02-iiab-config.yaml, 50-cloud-init.yaml) may no longer be accurate here in the documentation 2 weeks later:

   https://github.com/iiab/iiab/wiki/IIAB-Platforms#operating-systems

   Are both sections ("ethernets:" and "wifis:") now consolidated within `/etc/netplan/02-iiab-config.yaml` ?  I'll try to confirm with @jvonau in the coming 24h if he's available.

2) See "Internet Access" language proposed within PR https://github.com/iiab/iiab-admin-console/pull/323

tim-moody, 2020-05-16:
an a new install today with latest master on rpi4 with ub20:

after os install but no iiab:

```
root@ubuntu:/opt/iiab/iiab# ll /etc/netplan/
total 12
drwxr-xr-x  2 root root 4096 Apr  1 17:23 .
drwxr-xr-x 95 root root 4096 May 16 18:17 ..
-rw-r--r--  1 root root  416 Apr  1 17:23 50-cloud-init.yaml

```
after iiab install

```
root@ubuntu:/opt/iiab/iiab# ll /etc/netplan/
total 20
drwxr-xr-x   2 root root  4096 May 16 18:59 .
drwxr-xr-x 123 root root 12288 May 16 19:01 ..
-rw-r--r--   1 root root   416 Apr  1 17:23 02-iiab-config.yaml

```
This is what Admin Console expects and modifies as agreed.

holta, 2020-05-19:
For the record, in case this helps anybody now or down the road, @jvonau notes:

> 01-iiab is used if wan_ip is using static ip address set when admin-console says to use static, it wipes out /etc/netplan/* except 02 to get rid of the stock dhcp info and drops 01 in place with the static info.  Now just not supplying 01 if the interface is using dhcp as the need to fix netplan's old behavior of needing to add 'optional' appears to of gone away. All the files in /etc/netplan are viewed as user files and only removed when needed as above. I would expect a user to edit any of the /etc/netplan/*.yaml file to suit the network that is being connected to(ie wifi ssid/pw or static address) to gain the gateway to the internet prior to running iiab-install or iiab-network.

Maybe we should add the above backgrounder to https://github.com/iiab/iiab/wiki/IIAB-Platforms#operating-systems and/or https://github.com/iiab/iiab/wiki/IIAB-Network ?

(Apologies I personally don't grasp all the ramifications.)

holta, 2020-06-20:
Good enough explanation for now @ https://github.com/iiab/iiab/wiki/IIAB-7.1-Release-Notes#whats-new

closed 2020-06-20 by holta