iiab/iiab #2382: merged 2380

tim-moody, 2020-05-02:
Is this expected? on latest raspbian with latest master.

```
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN group default qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
    inet 127.0.0.1/8 scope host lo
       valid_lft forever preferred_lft forever
2: eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc mq state UP group default qlen 1000
    link/ether dc:a6:32:02:b5:c5 brd ff:ff:ff:ff:ff:ff
    inet 192.168.3.98/24 brd 192.168.3.255 scope global dynamic noprefixroute eth0
       valid_lft 71924sec preferred_lft 61124sec
3: wlan0: <NO-CARRIER,BROADCAST,MULTICAST,UP> mtu 1500 qdisc pfifo_fast state DOWN group default qlen 1000
    link/ether dc:a6:32:02:b5:c8 brd ff:ff:ff:ff:ff:ff
4: br0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc noqueue state UNKNOWN group default qlen 1000
    link/ether 8a:d8:0f:21:c9:48 brd ff:ff:ff:ff:ff:ff
    inet 172.18.96.1/19 brd 172.18.127.255 scope global noprefixroute br0
       valid_lft forever preferred_lft forever

```

tim-moody, 2020-05-02:
after reboot

```
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN group default qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
    inet 127.0.0.1/8 scope host lo
       valid_lft forever preferred_lft forever
2: eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc mq state UP group default qlen 1000
    link/ether dc:a6:32:02:b5:c5 brd ff:ff:ff:ff:ff:ff
    inet 192.168.3.98/24 brd 192.168.3.255 scope global dynamic noprefixroute eth0
       valid_lft 86390sec preferred_lft 75590sec
3: wlan0: <NO-CARRIER,BROADCAST,MULTICAST,UP> mtu 1500 qdisc pfifo_fast state DOWN group default qlen 1000
    link/ether dc:a6:32:02:b5:c8 brd ff:ff:ff:ff:ff:ff
4: ap0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc pfifo_fast master br0 state UP group default qlen 1000
    link/ether 02:33:91:71:a6:ee brd ff:ff:ff:ff:ff:ff
5: br0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc noqueue state UP group default qlen 1000
    link/ether 02:33:91:71:a6:ee brd ff:ff:ff:ff:ff:ff
    inet 172.18.96.1/19 brd 172.18.127.255 scope global noprefixroute br0
       valid_lft forever preferred_lft forever

```

holta, 2020-05-03:
> Is this expected? on latest raspbian with latest master.

@jvonau do you know?

jvonau, 2020-05-03:
In short yes, the ap0 cloning process has to occur before wpa_supplicant starts, which in the case of raspbian is started by dhcpcd. Rather than risk disrupting the install in order to accommodate installs over wifi the network is not fully available until rebooted. I'm not looking to improve on that at this point unless networking is dead last in 'iiab', to avoid disrupting the installation of iiab-admin-console.

tim-moody, 2020-05-03:
FYI, in the above wpa_supplicant was not really need because it had no credentials anyway. I know I asked this before, but the answer is hidden somewhere in a closed issue. I subsequently set upstream wifi credentials in wpa_supplicant (with adm cons). wlan0 did not automatically associate, which I think is expected. After reboot it did associate. Is there any way to associate short of another reboot that adm cons could run? Would restarting dhcpcd do it or is it necessary to up wlan0 first?

holta, 2020-05-03:
Related:

#2379 "Explain how to connect IIAB's internal Wi-Fi to the Internet: how to enter SSID & password"

tim-moody, 2020-05-03:
Think this was the answer. Still true?

> Not really, but let me walk through what the sequence might be
> killall wpa_supplicant to kill off the process that dhcpcd spawns but never shuts down on restart, that is part of the reason for the reboot.
> systemctl restart dhcpcd to spawn wpa_supplicant connecting to the the upstream access point, the dhcpcd_hook will run adjusting the channel in hostapd.conf if needed.
> sleep 5 adjust to taste
> systemctl restart hostapd to recreate ap0 and pickup the new channel.
> killall wpa_supplicant because the cloning disconnected the upstream wifi
> systemctl restart dhcpcd to re-establish the upstream wifi connection, now there should be no channel mis-match between wlan0 and ap0, the dhcpcd_hook runs when the upstream
> wifi becomes active giving ap0 a kick in the pants.
> 
> The less painful way would be to setup the upstream wifi first and reboot without the wired interface connected. Run the fresh install from that point or run iiab-network, both would need systemctl enable hostapd and reboot. Think if you set the upstream credentials and reboot twice the transition is less troublesome, more or less the same as the 2 killall calls above.

tim-moody, 2020-05-03:
the above causes numerous problems. bridge is gone, hostapd does not restart, no association with upstream wifi. even ssh becomes very slow to respond.

on reboot everything back working.

jvonau, 2020-05-03:
> FYI, in the above wpa_supplicant was not really need because it had no credentials anyway. 

The wpa_supplicant.conf file would have the country= line set that is what sets the regulatory domain for the wifi chip set.

> I know I asked this before, but the answer is hidden somewhere in a closed issue. I subsequently set upstream wifi credentials in wpa_supplicant (with adm cons). wlan0 did not automatically associate, which I think is expected. 

Think what you hit might be due to a channel mismatch between hostapd's current operating channel and what channel the upstream wifi is on. 

> After reboot it did associate. Is there any way to associate short of another reboot that adm cons could run? Would restarting dhcpcd do it

You would need to kill off the running wpa_supplicant first before restarting dhcpcd which only spawns wpa_supplicant but never ever restart it 

>or is it necessary to up wlan0 first?

Think you have to wait for ap0+wifi_up_down2 with the cloning split, that opens up better options.

holta, 2020-05-03:
Related:

https://github.com/iiab/iiab-admin-console/pull/313 "Changes for AP0"

jvonau, 2020-05-03:
> > Is this expected? on latest raspbian with latest master.
> 
> @jvonau do you know?

8751b33 would be responsible for the change
iiab_wireless_lan_iface(wlan0) - wifi_up_down False needs hw present to be set
iiab_wireless_lan_iface(ap0) - wifi_up_down True needs hw present to be set
hostapd_enabled False - set in hostapd.yml to avoid the restart with wifi_up_down False

Can't recall why, that was in the middle of the NM bugging. Think it might be safe to revert this one commit, the fallout might be on ubuntu side or wifi gateway installs.

tim-moody, 2020-05-03:
I seem to recall wpa_supplicant.conf having been moved on ub20 on rpi. Any way wlan0 did not associate after I put credentials in /etc/wpa_supplicant/wpa_supplicant.conf and rebooted.

tim-moody, 2020-05-03:
oops. password was wrong. retrying. NOPE still doesn't associate.

jvonau, 2020-05-03:
> I seem to recall wpa_supplicant.conf having been moved on ub20 on rpi. Any way wlan0 did not associate after I put credentials in /etc/wpa_supplicant/wpa_supplicant.conf and rebooted.

don't do that... use the netplan directions

jvonau, 2020-05-03:
#2383 cleaner replacement for 8751b33 might allow wired gateway installs have fully configured network without rebooting, depends if wpa_supplicant plays nice with the newly created ap0 device.

tim-moody, 2020-05-03:
>don't do that... use the netplan directions

I'd rather not be changing /etc/netplan/02-iiab-config.yaml which has all kinds of stuff in it. But where is wlan0 under wifis: ? and where is country= 

The other thing I don't like is that from what I read the passwords are in plain text.

tim-moody, 2020-05-03:
well at least the following file when added as /etc/netplan/03-iiab-config.yaml causes association. not sure where that should be to persist, though in fact it does persist across reboot.

```
# This file is generated from information provided by the datasource.  Changes
# to it will not persist across an instance reboot.  To disable cloud-init's
# network configuration capabilities, write a file
# /etc/cloud/cloud.cfg.d/99-disable-network-config.cfg with the following:
# network: {config: disabled}
network:
    version: 2
    wifis:
        wlan0:
            dhcp4: true
            access-points:
                "XXXX":
                    password: "yyyy"

```

jvonau, 2020-05-03:
> > don't do that... use the netplan directions
> 
> I'd rather not be changing /etc/netplan/02-iiab-config.yaml which has all kinds of stuff in it. But where is wlan0 under wifis: ? and where is country=
> 

Upstream issue if you have not been following along.
> The other thing I don't like is that from what I read the passwords are in plain text.

The pw's are in plain text in wpa_supplicant.conf also... Now the alternative to netplan is to use systemd-networkd directly, it will read files named wpa_supplicant-'iface'.conf.

make a symlink wpa_supplicant-wlan0.conf pointing to wpa_supplicant.conf 

start the service to test `sudo systemctl start wpa_supplicant@wlan0`
enable the service for reboot `sudo systemctl enable wpa_supplicant@wlan0`

jvonau, 2020-05-03:
> well at least the following file when added as /etc/netplan/03-iiab-config.yaml causes association. not sure where that should be to persist, though in fact it does persist across reboot.

The contents of /etc/netplan/02-iiab-config.yaml is never edited by iiab-network, it is a verbatim copy of what was 50-cloud-init.yml which had its roots in /boot/firmware/network-config. This is the upstream provided sample copy.

tim-moody, 2020-05-03:
```
wpa_supplicant@wlan0.service - WPA supplicant daemon (interface-specific version)
     Loaded: loaded (/lib/systemd/system/wpa_supplicant@.service; enabled; vendor preset: enabled)
     Active: active (running) since Sun 2020-05-03 20:19:14 UTC; 10min ago
   Main PID: 1230 (wpa_supplicant)
      Tasks: 1 (limit: 2100)
     CGroup: /system.slice/system-wpa_supplicant.slice/wpa_supplicant@wlan0.service
             └─1230 /sbin/wpa_supplicant -c/etc/wpa_supplicant/wpa_supplicant-wlan0.conf -Dnl80211,wext -iwlan0

May 03 20:19:14 box-pi-test.lan systemd[1]: Started WPA supplicant daemon (interface-specific version).
May 03 20:19:14 box-pi-test.lan wpa_supplicant[1230]: Successfully initialized wpa_supplicant
May 03 20:19:19 box-pi-test.lan wpa_supplicant[1230]: wlan0: Trying to associate with SSID 'SunRoom'
May 03 20:19:23 box-pi-test.lan wpa_supplicant[1230]: wlan0: Associated with c0:4a:00:e5:ec:04
May 03 20:19:23 box-pi-test.lan wpa_supplicant[1230]: wlan0: CTRL-EVENT-CONNECTED - Connection to c0:4a:00:e5:ec:04 completed [id=0 id_str=]
May 03 20:19:23 box-pi-test.lan wpa_supplicant[1230]: wlan0: CTRL-EVENT-SUBNET-STATUS-UPDATE status=0

```

it's like it associated but got no ip (? because nothing is asking for it)

jvonau, 2020-05-03:
Try 'systemctl restart systemd-networkd', but with a reboot the setting should stick.

tim-moody, 2020-05-04:
>Try systemctl restart systemd-networkd, but with a reboot the setting should stick.

Neither works. There's still no ip addr for wlan0.

I prefer this solution because it is simpler, so will hold off on implementing /etc/netplan/99-iiab-admin-access-points.yaml which I think will work.

tim-moody, 2020-05-04:
Also, does it matter if the install is on a desktop os?

jvonau, 2020-05-04:
> > Try systemctl restart systemd-networkd, but with a reboot the setting should stick.
> 
> Neither works. There's still no ip addr for wlan0.
> 

Don't have time to debug this with out direct access, netplan might be fighting it.. 

> I prefer this solution because it is simpler, so will hold off on implementing /etc/netplan/99-iiab-admin-access-points.yaml which I think will work.

It does work, that is not the point, just overwrite /etc/netplan/02-iiab-config.yaml as that file is protected from mods during the network stage. Just trying to avoid having duplicate keys like 'wifis' or 'wlan0'  within files in /etc/netplan/ not sure if netplan plays nice with keys defined twice.

jvonau, 2020-05-04:
> Also, does it matter if the install is on a desktop os?

On which os? Would you like me to go into detail about the differences in networking between the various desktops? I'd rather spend my time trying to document that within the code, which I've been trying to do.

tim-moody, 2020-05-04:
I won't bother you any more.

jvonau, 2020-05-05:
> > Try systemctl restart systemd-networkd, but with a reboot the setting should stick.
> 
> Neither works. There's still no ip addr for wlan0.
> 
> I prefer this solution because it is simpler, so will hold off on implementing /etc/netplan/99-iiab-admin-access-points.yaml which I think will work.

Think the missing component would be a systemd-networkd interface file for wlan0 try starting with
 
/etc/systemd/network/wlan0.network
[Match]
Name=wlan0

[Link]
RequiredForOnline=no

[Network]
DHCP=yes

jvonau, 2020-05-05:
> > Also, does it matter if the install is on a desktop os?
> 
> On which os? Would you like me to go into detail about the differences in networking between the various desktops? I'd rather spend my time trying to document that within the code, which I've been trying to do.

Using the wlan0.network file above would shift the management of the wifi device away from NetworkManager if a desktop is installed and might interfere with operation from within the desktop. Going the netplan route allows a desktop to function correctly given the desktop has the backend set to NM and if not the default would be networkd then need to do anything else goes away. I'd just write the wpa*|netplan files based on os, if you use a different name I'll add that to the protected list. Now it's just a key clash between files that I'm concerned about but if the latter file prevails when netplan reads them we should be good but if netplan pukes with the same key defined in two files we can only have one file used.

tim-moody, 2020-05-05:
I can live with netplan and yaml read/write would allow merging into a single file. e.g.

```
if ‘wifis’ not in netplan['network']:
    netplan['network'][‘wifis’] = {}

```
and so forth. 

Also I expect netplan merges multiple files but the key conflict would have to be tested.

holta, 2020-05-07:
PS #2384 -> #2393 is now merged.

Are there open issues within this tkt (#2382) that still need addressing or can we close it now?

tim-moody, 2020-05-07:
Can you point me to the documentation that explains what files to modify and with what settings in order to get an upstream wifi connection for both raspbian and ubuntu 20 on an rpi.

holta, 2020-05-07:
> Can you point me to the documentation that explains what files to modify and with what settings in order to get an upstream wifi connection for both raspbian and ubuntu 20 on an rpi.

netplan is incomplete (lacks support to specify country/frequencies and hidden SSID's) but see the upstream OS docs mentioned herein for starters:

https://github.com/iiab/iiab/wiki/IIAB-Platforms#operating-systems

Ref: #2379 "Explain how to connect IIAB's internal Wi-Fi to the Internet: how to enter SSID & password"

holta, 2020-05-08:
Related:

PR https://github.com/iiab/iiab-admin-console/pull/318 "Add wpa support of U20"

jvonau, 2020-05-08:
> netplan is incomplete (lacks support to specify country/frequencies and hidden SSID's) but see the upstream OS docs mentioned herein for starters:

Netplan has been updated upstream with the package in testing but not release yet 
http://launchpadlibrarian.net/477673548/netplan.io_0.99-0ubuntu3_source.changes
Further up the netplan chain has improvements for bands(2.4|5) and hidden SSID's but currently unreleased and unsure of the testing status.

jvonau, 2020-05-09:
I'd like to point out that iiab-network makes use of pre-existing network connections that were setup with whatever tools the distro provides, and doesn't create new upstream connections. Adding new wifi connections via these distro provided tools works well for Raspbian and Ubuntu while Debian a wifi new connection needs some additional steps that would normally already be in place when iiab-network work runs. Think the easiest way to backfit Debian from the admin-console or commandline is the systemd-networkd method outlined above, create the .network file, restart systemd-networkd, use the symlink for the conf file, enable and restart the wpa*@iface.service. Although this is limited to non-desktops as dealing with NetworkManager is a whole other ball of wax. Thinking if a desktop is available that would be the preferred method of selecting the wifi network. Enabling the systemd-networkd method should override and hopefully not break NM but might need some tuning to pull off.

holta, 2020-05-12:
Should this ticket be closed as we try to release IIAB 7.1 rather imminently?

Or moved to the [IIAB 7.2 Milestone](https://github.com/iiab/iiab/milestone/7) ?

jvonau, 2020-09-28:
Think this objective is met in 7.2 given the wide testing status that 7.2 now has.

closed 2020-09-28 by holta