iiab/iiab #2387: How It Works: IIAB 7.0 -> 7.1 Maps Doc

holta, 2020-05-05:
@georgejhunt can you please call me after reviewing this First Draft here?

I can then work on a 2nd draft with you here, so we can release [IIAB 7.1](https://github.com/iiab/iiab/wiki/IIAB-7.1-Release-Notes)  very shortly in coming days if all goes well:

https://github.com/iiab/iiab/wiki/IIAB-Maps

Goals:
1) Understand how IIAB Maps are installed & operate with IIAB 7.1 — both how these mechanics have evolved over the past year — and how these mechanics might further evolve?
2) Share this understanding of OSM inner-workings with newcomers asking for vivid clarity, when joining and building dynamic IIAB community of different kinds:
   - Articulate how IIAB Maps work in less-technical language wherever possible — so IIAB strengthens it grassroots/educator transparency, deepening their understanding of all their options around IIAB Maps...
   - ...thereby welcoming aligned movements wanting to work with — and possibly help enhance these maps.
4) Close or Evolve closely-related OSM tickets like: https://github.com/iiab/iiab/issues?q=is%3Aopen+OSM

Ref:
- #2096 "NGINX & ./runrole & ./iiab-network [Dual WiFi] & IIAB Maps need approachable docs for newbie/legacy implementers"

holta, 2020-05-05:
Let's also add some forward-looking recommendations (prayers?!) for everyone trying to navigate archive.org's very painfully downloads.

e.g. downloading the 53GB "World" map pack can take 3 days (about 72h) given its currently typical download speeds of 1-to-2 megabits per second.

(Possibly these slower-than-2019 speeds are a result of the Internet Archive's newfound popularity during the COVID-19 pandemic?)

holta, 2020-05-06:
Thanks much to @georgejhunt who called me earlier today, to explain much more how IIAB Maps infra works overall, and where this org chart / layout is likely heading in future.

So another draft of the IIAB Maps doc is now posted here &mdash; getting us closer to [IIAB 7.1](https://github.com/iiab/iiab/wiki/IIAB-7.1-Release-Notes)'s final release:

https://github.com/iiab/iiab/wiki/IIAB-Maps

_LMK further improvement(s) required above!_

holta, 2020-05-06:
A 3rd draft was just posted:

https://github.com/iiab/iiab/wiki/IIAB-Maps

georgejhunt, 2020-05-06:
You do good work. Thank you

holta, 2020-05-07:
Good enough for [IIAB 7.1](https://github.com/iiab/iiab/wiki/IIAB-7.1-Release-Notes) (Release Candidate 1 is now published).

Feel free to revise https://github.com/iiab/iiab/wiki/IIAB-Maps if there are errors/omissions!

closed 2020-05-07 by holta