iiab/iiab #2389: Authenticate via secure user/password without basic auth (tracking tkt)

holta, 2020-05-05:
Great news if this is possible in coming months: https://github.com/iiab/iiab-admin-console/issues/314

Towards avoiding these login conflicts/bugs:
- #1532 "http://box/kolibri causes browser to infinitely loop, preventing login (works initially...server reboot does not help...happens with all client OS's & all server OS's)"
- #2366 "When logged into Admin Console Nextcloud tries to use admin credentials to login."

holta, 2020-05-21:
Related:

PR iiab/iiab-admin-console#324 "0.4.1.4 remove auth_pam and basic auth"

jvonau, 2020-09-26:
wouldn't this be 7.2 related?

holta, 2020-09-26:
> wouldn't this be 7.2 related?

Oops, this was solved as part of IIAB 7.1 I think.

closed 2020-09-26 by holta