iiab/iiab #2391: CMDSRV has started but is not ready (when boot from usbSSD)

brandhoff1234, 2020-05-07:
i have iiab working on an sd card already so i copied that to an SSD.
rasbian boots up and logs onto wifi. i can access the home page, all the links work fine.
however i get these error codes on the admin page
i am using the same PI3 for both tests.
i have edited the local_vars.yml but like i said i just copied the SD Image.

date installed 2020-05-01T20:41:40Z
installed using http://download.iiab.io/

Here are my error codes:

5/6/2020, 10:58:00 PM: procOneCatalog found empty data
5/6/2020, 10:58:00 PM: procOneCatalog found empty data
5/6/2020, 10:57:57 PM: GET-EXTDEV-INFO FAILED and returned CMDSRV has started but is not ready.
5/6/2020, 10:57:57 PM: GET-SPACE-AVAIL FAILED and returned CMDSRV has started but is not ready.
5/6/2020, 10:57:57 PM: GET-OER2GO-STAT FAILED and returned CMDSRV has started but is not ready.
5/6/2020, 10:57:57 PM: GET-ZIM-STAT FAILED and returned CMDSRV has started but is not ready.
5/6/2020, 10:57:57 PM: initConfigVars found empty data
5/6/2020, 10:57:57 PM: GET-IIAB-INI FAILED and returned CMDSRV has started but is not ready.
5/6/2020, 10:57:57 PM: GET-ANS FAILED and returned CMDSRV has started but is not ready.
5/6/2020, 10:57:57 PM: GET-VARS FAILED and returned CMDSRV has started but is not ready.
5/6/2020, 10:57:54 PM: GET-WHLIST FAILED and returned CMDSRV has started but is not ready.
5/6/2020, 10:57:54 PM: Successfully connected to Server
5/6/2020, 10:57:54 PM: Command GET-EXTDEV-INFO sent to server
5/6/2020, 10:57:54 PM: Command GET-SPACE-AVAIL sent to server
5/6/2020, 10:57:54 PM: Command GET-OER2GO-STAT sent to server
5/6/2020, 10:57:54 PM: Command GET-ZIM-STAT sent to server
5/6/2020, 10:57:54 PM: Command GET-IIAB-INI sent to server
5/6/2020, 10:57:54 PM: Command GET-ANS sent to server
5/6/2020, 10:57:54 PM: Command GET-VARS sent to server
5/6/2020, 10:57:54 PM: Command GET-WHLIST sent to server
5/6/2020, 10:57:54 PM: Checking Server Connection
5/6/2020, 10:57:54 PM: Starting init

tim-moody, 2020-05-07:
Can you do the following and tell me what happens:
ssh in
as root
cd /opt/admin/cmdsrv/
scripts/wr-sdcard-params.sh

probably this script is not compatible with an ssd

brandhoff1234, 2020-05-07:
[image]

Just sits there.

tim-moody, 2020-05-07:
that was my guess. it usually comes back in several minutes.
what does ls /sys/devices/platform/soc/ say?
I need to find a quick test to know we are not on an sd card.

brandhoff1234, 2020-05-08:
[image]

holta, 2020-05-08:
@brandhoff1234 I can't guarantee we'll be able to solve this for the imminent release of [IIAB 7.1](https://github.com/iiab/iiab/wiki/IIAB-7.1-Release-Notes), but can you share the URL that results from you running `iiab-diagnostics` if possible?

Thanks if possible!

holta, 2020-05-08:
@tim-moody please unassign yourself if you don't have time to work on this, Thanks all!

tim-moody, 2020-05-08:
I could tell more if I could log into the machine, but my best guess is that you can work around this problem by editing /opt/admin/cmdsrv/iiab-cmdsrv3.py

comment out the two lines (should be around 2633f)

```
if is_rpi:
        write_sdcard_params()
```

Please tell us if this solves the problem.

brandhoff1234, 2020-05-08:
http://dpaste.com//3E9R45P

tim-moody, 2020-05-08:
@brandhoff1234 did you try the filx?

brandhoff1234, 2020-05-08:
i think we are good. that's awesome. i love the project! thank you so much for the help!!

closed 2020-05-08 by brandhoff1234

tim-moody, 2020-05-08:
If you patched, it will get overwritten if you do another install.

brandhoff1234, 2020-05-09:
like if i download more content or install all of iiab again?

tim-moody, 2020-05-09:
if you reinstall admin console the patch will be gone. if you confirm that commenting out those two lines fixed the problem I will make it permanent.

brandhoff1234, 2020-05-09:
Yes it is fixed.

tim-moody, 2020-05-09:
>if you confirm that commenting out those two lines fixed the problem I will make it permanent.

done iiab/iiab-admin-console#317