iiab/iiab #2409: trying to get moodle to work

kmw, 2020-05-14:
### Expected behavior and actual behavior.
  Moodle on RACHEL: Moodle is a learning platform designed to provide educators, administrators and learners with a single robust, secure and integrated system to create personalized learning environments. requires system built with RACHEL Installer v2.1+.
Moodle on RACHEL
Moodle is a learning platform designed to provide educators, administrators and learners with a single robust, secure and integrated system to create personalized learning environments. requires system built with RACHEL Installer v2.1+.

Language: en
Category: educational,software
Age Range: adult
Rating: 3.0
Version: v1.0
Size: 350.13 MB
Number of Files: 41

Size: 350.13 MB - INSTALLED
[7.0_raspbian-10_2020-05-13_17:16:55_PDT_ctn-moodle-update.log](https://github.com/iiab/iiab/files/4625346/7.0_raspbian-10_2020-05-13_17.16.55_PDT_ctn-moodle-update.log)

http://box.lan/home/
http://box.lan/working/rachel
Not Found
The requested URL was not found on this server.

Apache/2.4.38 (Raspbian) Server at box.lan Port 80
### Steps to reproduce the problem.
the first attached file is with moodle un installed
the second attached file is with moodle install and rpi 3 updated

### Specifications like the version of the IIAB, operating system version, or hardware details.
1TB external ssd connected via usb
no other hardware has been added.
pi@box:~ $ sudo iiab-diagnostics

Accelerate troubleshooting by collecting your IIAB diagnostics into 1 file:
COMPLETE! To share this on the web (68K, 1333 lines) run:

pastebinit < /etc/iiab/diag/7.0_raspbian-10_2020-05-13_08:53:57_PDT_ctn

http://paste.debian.net/
[7.0_raspbian-10_2020-05-13_08:53:57_PDT_ctn.log](https://github.com/iiab/iiab/files/4624189/7.0_raspbian-10_2020-05-13_08.53.57_PDT_ctn.log)
[7.0_raspbian-10_2020-05-13_17:16:55_PDT_ctn-moodle-update.log](https://github.com/iiab/iiab/files/4625364/7.0_raspbian-10_2020-05-13_17.16.55_PDT_ctn-moodle-update.log)

kmw, 2020-05-14:
783 | scripts/oer2go_install_move.sh en-moodle
-- | --
main()File "/opt/admin/cmdsrv/scripts/get_oer2go_catalog", line 160, in mainproc_module(module) # process menu itemFile "/opt/admin/cmdsrv/scripts/get_oer2go_catalog", line 275, in proc_moduleoutp = subprocess.check_output(args)File "/usr/lib/python2.7/subprocess.py", line 223, in check_outputraise CalledProcessError(retcode, cmd, output=output)subprocess.CalledProcessError: Command '['wget', '-O', '/library/www/html/js-menu/menu-files/images/en-osm-omt-min.']' returned non-zero exit status 1Getting the OER2Go/RACHEL Catalog Failed.Please run iiab-get-oer2go-cat again later. | SUCCEEDED | 05-13 12:2800:10:12
-- | -- | --

782 | /usr/bin/rsync -Pavz --size-only rsync://dev.worldpossible.org/rachelmods/en-moodle /library/working/rachel/
-- | --

0 0% 0.00kB/s 0:00:002,884 100% 4.57kB/s 0:00:00 (xfr#41, to-chk=0/43)sent 823 bytes received 337,588,428 bytes 557,537.99 bytes/sectotal size is 367,056,762 speedup is 1.09 | SUCCEEDED | 05-13 12:2800:10:08
-- | -- | --

holta, 2020-05-14:
There's a better way to install Moodle (the latest 3.8.3 !) using /etc/iiab/local_vars.yml

Best to customize the above file prior to installing IIAB, if possible anyway!

See: https://github.com/iiab/iiab-admin-console/issues/321

tim-moody, 2020-05-14:
@kmw this is indeed a bug, which is that moodle is not supposed appear in the list of items that can be installed with Admin Console. It needs to be installed using ansible, not from the OER2Go catalog. There are some items in that catalog that require more than a simple download, like moodle which requires an entire setup. We filter catalog items of type 'html', but moodle is html when it should be 'php'.

So the short answer is that you need to edit /etc/iiab/local_vars.yml and change
moodle_install: False
moodle_enabled: False
to True
then
cd /opt/iiab/iiab
./runrole moodle
in order to install it

tim-moody, 2020-05-14:
@kmw you have shown another error that 

> Getting the OER2Go/RACHEL Catalog Failed.

I wasn't able to reproduce that when I clicked on Install Content -> Refresh OER2Go Catalog. Can you tell me how that error occurred. It seems to be on the Job Status screen.

tim-moody, 2020-05-14:
> I wasn't able to reproduce that when I clicked on Install Content -> Refresh OER2Go Catalog. Can you tell me how that error occurred. It seems to be on the Job Status screen.

I did now see this once and then it went away. I wonder if the catalog was republished by RACHEL and had an intermittent problem, as I don't see the problem now.

tim-moody, 2020-05-14:
>/usr/lib/python2.7/subprocess.py

these scripts are on python3 now, so not sure what is happening there either

kmw, 2020-05-14:
Thank you very much. Do I need to upgrade to the latest iiab version that has python 3? How do I do that and keep the /library?
Note I've had the moodle problem on my other installs. I'll create a new install on a different box and see if I can get it to work.

tim-moody, 2020-05-14:
Yes. start with a fresh install. Then you could copy your content files using Admin Console Install Content ->Manage Content. (This will do zims and modules, so you might have to rsync other content such as osm manually, depending on what you have.)

kmw, 2020-05-14:
> @kmw this is indeed a bug, which is that moodle is not supposed appear in the list of items that can be installed with Admin Console. It needs to be installed using ansible, not from the OER2Go [...]
pi@box:/opt/iiab/iiab $ sudo ./runrole moodle
the only error I see in the terminal:
TASK [moodle : Restart postgresql-iiab] ****************************************
fatal: [127.0.0.1]: FAILED! => {"changed": false, "msg": "Could not find the requested service postgresql-iiab: host"}

PLAY RECAP *********************************************************************
127.0.0.1                  : ok=31   changed=12   unreachable=0    failed=1    skipped=16   rescued=0    ignored=0   

pi@box:/opt/iiab/iiab $ 

is: /working/rachel where it's located

holta, 2020-05-14:
@kmw: As an aside, https://github.com/iiab/iiab/wiki/IIAB-7.1-Release-Notes should be released in coming days, so if you get a chance to test our Release Candidate (http://download.iiab.io) that'd be extremely helpful!

tim-moody, 2020-05-14:
was this a fresh install?
I guess I misled you on runrole as I thought it would pick up postgresql.
the next thing would be in /opt/iiab/iiab run
./iiab-install --reinstall
this will do a complete install
however if the last step got part way through, existing files may block a full install, so you are still better off with a fresh install
>is: /working/rachel where it's located

no that's just a staging directory

holta, 2020-05-15:
Can this ticket be closed now?

(Or deferred to [IIAB 7.2](https://github.com/iiab/iiab/milestone/7) if there are remaining issues?)

kmw, 2020-05-17:
i was not able to get moodle installed on my existing instance but a fresh install did work on raspian. thanks

closed 2020-05-17 by kmw

holta, 2020-05-17:
SEE ALSO: #2412