iiab/iiab #2410: Calibre-Web (http://box/books) no longer works with Python 3?

holta, 2020-05-15:
Thanks @shanti-bhardwa for reporting this!

Thanks @jvonau for confirming "502 bad gateway" and the details below on http://10.8.0.22/books machine 182-rpi4-2g-lite-BIG-RC2-0512:

```
iiab-admin@box:~ $ sudo /usr/local/calibre-web-py3/bin/python3 /usr/local/calibre-web-py3/cps.py -p /library/calibre-web/config/app.db
Traceback (most recent call last):
  File "/usr/local/calibre-web-py3/lib/python3.7/site-packages/sqlalchemy/engine/base.py", line 1248, in _execute_context
    cursor, statement, parameters, context
  File "/usr/local/calibre-web-py3/lib/python3.7/site-packages/sqlalchemy/engine/default.py", line 590, in do_execute
    cursor.execute(statement, parameters)
sqlite3.OperationalError: no such column: user.mature_content

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/usr/local/calibre-web-py3/cps.py", line 34, in <module>
    from cps import create_app
  File "/usr/local/calibre-web-py3/cps/__init__.py", line 64, in <module>
    ub.init_db(cli.settingspath)
  File "/usr/local/calibre-web-py3/cps/ub.py", line 658, in init_db
    migrate_Database(session)
  File "/usr/local/calibre-web-py3/cps/ub.py", line 555, in migrate_Database
    if session.query(User).filter(User.role.op('&')(constants.ROLE_ANONYMOUS) == constants.ROLE_ANONYMOUS).first() \
  File "/usr/local/calibre-web-py3/lib/python3.7/site-packages/sqlalchemy/orm/query.py", line 3300, in first
    ret = list(self[0:1])
  File "/usr/local/calibre-web-py3/lib/python3.7/site-packages/sqlalchemy/orm/query.py", line 3078, in getitem
    return list(res)
  File "/usr/local/calibre-web-py3/lib/python3.7/site-packages/sqlalchemy/orm/query.py", line 3405, in iter
    return self._execute_and_instances(context)
  File "/usr/local/calibre-web-py3/lib/python3.7/site-packages/sqlalchemy/orm/query.py", line 3430, in _execute_and_instances
    result = conn.execute(querycontext.statement, self._params)
  File "/usr/local/calibre-web-py3/lib/python3.7/site-packages/sqlalchemy/engine/base.py", line 984, in execute
    return meth(self, multiparams, params)
  File "/usr/local/calibre-web-py3/lib/python3.7/site-packages/sqlalchemy/sql/elements.py", line 293, in _execute_on_connection
    return connection._execute_clauseelement(self, multiparams, params)
  File "/usr/local/calibre-web-py3/lib/python3.7/site-packages/sqlalchemy/engine/base.py", line 1103, in _execute_clauseelement
    distilled_params,
  File "/usr/local/calibre-web-py3/lib/python3.7/site-packages/sqlalchemy/engine/base.py", line 1288, in _execute_context
    e, statement, parameters, cursor, context
  File "/usr/local/calibre-web-py3/lib/python3.7/site-packages/sqlalchemy/engine/base.py", line 1482, in _handle_dbapi_exception
    sqlalchemy_exception, with_traceback=exc_info[2], from_=e
  File "/usr/local/calibre-web-py3/lib/python3.7/site-packages/sqlalchemy/util/compat.py", line 178, in raise_
    raise exception
  File "/usr/local/calibre-web-py3/lib/python3.7/site-packages/sqlalchemy/engine/base.py", line 1248, in _execute_context
    cursor, statement, parameters, context
  File "/usr/local/calibre-web-py3/lib/python3.7/site-packages/sqlalchemy/engine/default.py", line 590, in do_execute
    cursor.execute(statement, parameters)
sqlalchemy.exc.OperationalError: (sqlite3.OperationalError) no such column: user.mature_content
[SQL: SELECT user.id AS user_id, user.nickname AS user_nickname, user.email AS user_email, user.role AS user_role, user.password AS user_password, user.kindle_mail AS user_kindle_mail, user.locale AS user_locale, user.sidebar_view AS user_sidebar_view, user.default_language AS user_default_language, user.mature_content AS user_mature_content, user.denied_tags AS user_denied_tags, user.allowed_tags AS user_allowed_tags, user.denied_column_value AS user_denied_column_value, user.allowed_column_value AS user_allowed_column_value 
FROM user 
WHERE (user.role & ?) = ?
 LIMIT ? OFFSET ?]
[parameters: (32, 32, 1, 0)]
(Background on this error at: http://sqlalche.me/e/e3q8)
```

holta, 2020-05-16:
@OzzieIsaacs are you familiar with this Calibre-Web 0.6.7 glitch related to Python 3 possibly?

(Or if not, how we should best go about diagnosing it?)

Thanks!  Traceback at: https://github.com/iiab/iiab/issues/2410#issue-619319574

OzzieIsaacs, 2020-05-16:
This is not a problem of python3, this error would also occur with python2. 
What happened (in my opinion): You are having a very old settings database (equals old version of calibre-web). **Is this possible?** In between I created a setting to hide content (mature_content), afterwards I extended this global content hiding system to a system per user base. This made the mature_content setting obsolete. 
If I generate new columns I check if the setting exists and if not, the codes adds the settings. In this case, I don't create the setting anymore (as it#s not needed anymore), but I try to migrate from an existing mature_content setting to the user based setting. the setting isn't available -> program crashes.
Workarounds -> 
A) downgrade to a version between 0.6.6 and 0.6.0 (strange, all versions since 0.6.0 had the migration code)
B) delete settings database app.db and start with a fresh one

holta, 2020-05-16:
@OzzieIsaacs that would make sense:

Two-to-three years ago @arky set in motion IIAB's install routine, to auto-copy his own https://github.com/iiab/iiab/blob/master/roles/calibre-web/files/app.db into position at `/library/calibre-web/config/app.db` i.e. anytime somebody installs IIAB with Calibre-Web.

In 2020 would you recommend a better way to seed/create a fresh `app.db` during each IIAB install?

Keeping in mind that @arky likely customized a few values within the above static/seed `app.db` &mdash; which we might also have to do in 2020 &mdash; so worst case live auto-seeding of `app.db` (creating it on the fly) might not be possible?!

closed 2020-05-16 by holta

OzzieIsaacs, 2020-05-17:
You changed some settings which make no sense to do the changes within Calibre-Web, so the way you did now (creating a new version) makes most sense. 
Your version was a bit to early all "real" releases would have been compatible. I try to be as compatible as possible, but I delete from time to time too old migration routines (otherwise I get very strange effects during startup, which I don't understand). With the new version you now generated, you should be future proof for the next few years.

holta, 2020-05-17:
> With the new version you now generated [app.db] you should be future proof for the next few years.

Many Thanks @OzzieIsaacs !

PR #2411 (2020-05-16): Calibre-Web: fresh new app.db (customized for Calibre-Web 0.6.7 with IIAB)
Fixes #2410

1) This new `app.db` was customized using [Calibre-Web](https://github.com/janeczku/calibre-web) 0.6.7, which created a fresh database after I deleted `/library/calibre-web/config/app.db` and then ran `systemctl start calibre-web`.  Which I then modified, e.g.:
   - Location of Calibre Database: `/library/calibre-web` (to store books/content)
   - Admin account (for http://box/books = http://box:8083) changed from admin/admin123 to Admin/changeme
     - Admin account further customized
   - Guest account cr [... cut]