iiab/iiab #2412: Moodle installation failed on fresh install

kmw, 2020-05-17:
### Expected behavior and actual behavior.
Created a fresh #3 install where moodle install was True by default
all went smooth during installation
moodle was not at http://box.lan/moodle

### Steps to reproduce the problem.
not sure. this has been my experience, over and over again. i'm don some 8+ installs where I most always choose to install moodle.

### Specifications like the version of the IIAB, operating system version, or hardware details.
ubuntu 20.0.4 server install using:
https://ubuntu.com/download/raspberry-pi
i used the .deb program to boh download and write the image to a 256g micro sdc
used terminal for install but when it rebooted it had a gui login. 
under activities the terminal has an icon but would not open.
i tried this over and over. no matter what i did i could not open a terminal.
also, the screen saver see to freeze leaving just a _ in the top left corner of the screen. i killed the power each time i left the machine for more than a while.
i changed to a different pi4 w/ 4g mem and the same thing just happened. 
this reminds me that from the very first release, i've never been able to use vanilla ubuntu  anything.
i will prepare a fresh raspian micro sdc and start over with another #3 install

holta, 2020-05-17:
Just fyi I installed Moodle 3.8.3 successfully[*] as part of a clean/fresh install IIAB 7.1 pre-release over the past hour, onto Ubuntu 20.04 LTS (graphical/desktop version) on a virtual machine (not a Raspberry Pi in this case).

[*] I logged in to http://box.lan/moodle worked, with username/password Admin/changeme

holta, 2020-05-17:
@kmw can you please share the URL that results after you the `iiab-diagnostics` command ?

kmw, 2020-05-17:
t i am not able to even get to the box anymore i'm going with raspian. i'll start it now.

kmw, 2020-05-17:
the raspian #3 level install created a working  moodle by default

kmw, 2020-05-17:
i will try the ubuntu again at some point. thanks

closed 2020-05-17 by kmw

holta, 2020-05-17:
> i will try the ubuntu again at some point. thanks

Thanks @kmw.

Please publish the URL resulting from the `iiab-diagnostics` command so more people can help you going forward!