iiab/iiab #2422: Test IIAB 7.1 RC4 on 64-bit Raspbian (renamed to Raspberry Pi OS)

holta, 2020-05-28:
@jvonau & All: the early beta of Raspberry Pi Foundation's new OS is here:
https://www.raspberrypi.org/forums/viewtopic.php?f=117&t=275370

Latest 32-bit versions are here:
https://www.raspberrypi.org/downloads/raspbian/
http://downloads.raspberrypi.org/raspios_armhf/release_notes.txt

holta, 2020-05-29:
Thanks @floydianslips who observes:
- He had to do `apt install vlc-data` before the IIAB install would begin
- He could not log into Admin Console (http://box.lan/admin)

SEE ALSO @georgejhunt report:

#2424 "raspbian-64bit-dnsmasq does not start because br0 is not up [RaspiOS]"

holta, 2020-05-29:
@floydianslips @georgejhunt did you also see this error below on 64-bit RaspiOS on RPi 4?

> TASK [2-common : Install 24 common packages: acpid, bridge-utils, bzip2, curl, gawk, hostapd, htop, i2c-tools, logrotate, make, mlocate, netmask, net-tools, ntfs-3g, pandoc, pastebinit, rsync, sqlite3, sudo, tar, unzip, usbmount, usbutils, wget] ***
fatal: [127.0.0.1]: FAILED! => {"changed": false, "msg": "No package matching 'usbmount' is available"}

holta, 2020-05-29:
Even when I comment out pkg usbmount from https://github.com/iiab/iiab/blob/master/roles/2-common/tasks/packages.yml#L63 it still fails later on in Stage 4, predictably:

> TASK [usb_lib : Enable exFAT and NTFS in /etc/usbmount/usbmount.conf] **********
fatal: [127.0.0.1]: FAILED! => {"changed": false, "msg": "Destination /etc/usbmount/usbmount.conf does not exist !", "rc": 257}

holta, 2020-05-29:
Until the above issue is solved, I've temporarily set the following in [/etc/iiab/local_vars.yml](http://wiki.laptop.org/go/IIAB/FAQ#What_is_local_vars.yml_and_how_do_I_customize_it.3F) :

```
usb_lib_install: False
usb_lib_enabled: False

holta, 2020-05-29:
FYI Sugarizer also fails to install on 64-bit RaspiOS, due to...

> TASK [mongodb : Enable & Restart 'mongodb' systemd service if mongodb_enabled, incl daemon-reload (in case mongodb.service changed?)] ***
fatal: [127.0.0.1]: FAILED! => {"changed": false, "msg": "Unable to start service mongodb: Job for mongodb.service failed because the control process exited with error code.\nSee \"systemctl status mongodb.service\" and \"journalctl -xe\" for details.\n"}

> root@raspberrypi:~# systemctl status mongodb
● mongodb.service - High-performance, schema-free document-oriented database
   Loaded: loaded (/etc/systemd/system/mongodb.service; enabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Fri 2020-05-29 14:26:38 EDT; 10min ago
  Process: 10628 ExecStartPre=/usr/bin/iiab-mongodb-repair-if-no-lock (code=exited, status=127)
>
> May 29 14:26:38 box.lan systemd[1]: Starting High-performance, schema-free document-oriented database...
May 29 14:26:38 box.lan iiab-mongodb-repair-if-no-lock[10628]: /usr/bin/mongod: error while loading shared libraries: libstdc++.so.6: cannot open shared object file: No
May 29 14:26:38 box.lan systemd[1]: mongodb.service: Control process exited, code=exited, status=127/n/a
May 29 14:26:38 box.lan systemd[1]: mongodb.service: Failed with result 'exit-code'.
May 29 14:26:38 box.lan systemd[1]: Failed to start High-performance, schema-free document-oriented database.

tim-moody, 2020-05-29:
> This is a beta test of the Raspberry Pi OS (64 bit)
> 
> We still recommend the 32 bit operating system for all Pis at this time, although have decided it is now time to begin the move toward a 64bit OS. For the moment this is a 'beta' program, the OS is in heavy flux and its functionality is **likely to change significantly over the next few months**.

holta, 2020-05-29:
Good News first...

1) fyi @floydianslips Admin Console works for me on 64-bit RaspiOS e.g. I can log in to http://10.8.0.38/admin (182-rpi4-2g-raspios-64bit-MEDIUM) with the usual username/password (iiab-admin/g0adm1n) and navigate around Admin Console after that.

2) fyi @jvonau after reboot, dnsmasq no longer seems to start...

> root@box:~# systemctl status dnsmasq
● dnsmasq.service - dnsmasq - A lightweight DHCP and caching DNS server
   Loaded: loaded (/lib/systemd/system/dnsmasq.service; disabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Fri 2020-05-29 15:01:16 EDT; 25min ago
  Process: 623 ExecStartPre=/usr/sbin/dnsmasq --test (code=exited, status=0/SUCCESS)
  Process: 649 ExecStart=/etc/init.d/dnsmasq systemd-exec (code=exited, status=2)
>
> May 29 15:01:15 box.lan systemd[1]: Starting dnsmasq - A lightweight DHCP and caching DNS server...
May 29 15:01:15 box.lan dnsmasq[623]: dnsmasq: syntax check OK.
May 29 15:01:16 box.lan dnsmasq[649]: dnsmasq: unknown interface br0
May 29 15:01:16 box.lan systemd[1]: dnsmasq.service: Control process exited, code=exited, status=2/INVALIDARGUMENT
May 29 15:01:16 box.lan systemd[1]: dnsmasq.service: Failed with result 'exit-code'.
May 29 15:01:16 box.lan systemd[1]: Failed to start dnsmasq - A lightweight DHCP and caching DNS server.

tim-moody, 2020-05-29:
@holta and @floydianslips can you report if captive portal is running
cat /var/log/uwsgi/app/admin-console.log if authentication doesn't happen. it will tell why

floydianslips, 2020-05-29:
@tim-moody this is the output of `/var/log/uwsgi/app/admin-console.log`. 

http://dpaste.com/05CTF8D

floydianslips, 2020-05-29:
>     1. fyi @floydianslips Admin Console works for me on 64-bit RaspiOS e.g. I can log in to http://10.8.0.38/admin (182-rpi4-2g-raspios-64bit-MEDIUM) with the usual username/password (iiab-admin/g0adm1n) and navigate around Admin Console after that.

What happens if you change the password?

tim-moody, 2020-05-29:
@floydianslips there are a lot of log entries that say the user is invalid, though there is a decryption failure as well.

holta, 2020-05-29:
> > ```
> > 1. fyi @floydianslips Admin Console works for me on 64-bit RaspiOS e.g. I can log in to http://10.8.0.38/admin (182-rpi4-2g-raspios-64bit-MEDIUM) with the usual username/password (iiab-admin/g0adm1n) and navigate around Admin Console after that.
> > ```
> 
> What happens if you change the password?

That works &mdash; fyi I can continue to log into http://10.8.0.38/admin after changing iiab-admin's password.

floydianslips, 2020-05-30:
The new 64-bit OS shows as Debian 10 if you `cat /etc/os-release`:
```
PRETTY_NAME="Debian GNU/Linux 10 (buster)"
NAME="Debian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"
VERSION_CODENAME=buster
ID=debian
HOME_URL="https://www.debian.org/"
SUPPORT_URL="https://www.debian.org/support"
BUG_REPORT_URL="https://bugs.debian.org/"
```
compared to old: ```

PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"
VERSION_CODENAME=buster
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"
```
Debian doesn't have `usbmount` in their packages...

holta, 2020-05-30:
Raspbian which has now been renamed RaspiOS (Raspberry Pi OS) is explained here...alongside its 64-bit Beta preview and 32-bit updates etc...as released 2 days ago:

https://www.raspberrypi.org/blog/latest-raspberry-pi-os-update-may-2020/

jvonau, 2020-06-04:
https://github.com/raspberrypi/Raspberry-Pi-OS-64bit/issues/40

holta, 2020-08-25:
Can someone try their new attempt at a 64-bit Lite?
https://downloads.raspberrypi.org/raspios_lite_arm64/images/

(I don't expect it's fully ready for prime time...given their late May 2020 64-bit RaspiOS "with desktop" still seemed to be quite buggy ??)

ASIDE: 32-bit "2020-08-20" Raspberry Pi OS images released:
https://www.raspberrypi.org/downloads/raspberry-pi-os/

holta, 2020-08-25:
> I've temporarily (also) set the following in /etc/iiab/local_vars.yml :
>
> usb_lib_install: False
> usb_lib_enabled: False

Still true 3 months later, when installing IIAB installed onto the brand new experimental 64-bit RaspiOS (https://github.com/iiab/iiab/issues/2422#issuecomment-680000675) on this Raspberry Pi 4:

183-rpi4-2g-lite-64bit-BIG-2487 = 10.8.0.58

Likewise I again commented out Line 63 of https://github.com/iiab/iiab/blob/master/roles/2-common/tasks/packages.yml#L63 here:

```
#      - usbmount
```

holta, 2020-08-25:
FWIW they've not changed `ID=debian` to `ID=raspbian` here:

holta, 2020-08-25:
IIAB's current [MongoDB 3.0.14](https://github.com/iiab/iiab/blob/master/roles/mongodb/tasks/install.yml#L27-L43) for 32-bit RaspiOS doesn't work on 64-bit RaspiOS.  As @jvonau & I reconfirmed today.

If there's a need to get Sugarizer working on 64-bit RaspiOS, this might possibly be the way forward:
https://andyfelong.com/2019/03/mongodb-4-0-6-64-bit-on-raspberry-pi-3/

holta, 2020-08-25:
Related to usbmount: ([roles/usb_lib](https://github.com/iiab/iiab/tree/master/roles/usb_lib) issue [above](https://github.com/iiab/iiab/issues/2422#issuecomment-680095394))

@jvonau might be replacing PR [#2491](https://github.com/iiab/iiab/pull/2491) "Usbmnt2 [usbmount_0.0.14.1_all.deb -> usbmount_0.0.22_all.deb on Debian]"

...with upcoming PR #2492 ?

holta, 2020-08-25:
> 64-bit RaspiOS ([#2422 (comment)](https://github.com/iiab/iiab/issues/2422#issuecomment-680000675)) on this Raspberry Pi 4:
> 
> 183-rpi4-2g-lite-64bit-BIG-2487 = 10.8.0.58

I also need to set Monit to False/False in /etc/iiab/local_vars.yml as follows: (package does not exist for 64-bit RaspiOS)

monit_install: False
monit_enabled: False

Perhaps this means it's now time for Monit to be removed from BIG-sized IIAB installations (at long last!?)

holta, 2020-08-25:
Related:

PR #2493 "RFC Mongo2" by @jvonau

holta, 2020-09-03:
FYI/FWIW some complaints about 64-bit RaspiOS...not quite being ready for market?
https://www.raspberrypi.org/forums/viewtopic.php?f=117&t=275370

(Not that 32-bit RaspiOS is ever fully ready for market, as we know from #2497 i.e. mainline RaspiOS kernel again breaking WiFi etc, wink ;)

holta, 2020-09-15:
Seems to work.  But further testing is needed during the IIAB 8.0 release cycle, if we (and the Raspberry Pi Foundation!) are to support these 64-bit OS's more officially!

holta, 2020-11-03:
I and @georgejhunt did several tests of IIAB 7.2 pre-releases onto 64-bit RaspiOS ("Lite" and "with desktop") this week and it looks solid!

The catch it of course that microSD cards will not be portable (insertable) into 32-bit RPi Zero W machines however :/

holta, 2020-11-06:
Any guess how much faster 64-bit Raspberry Pi OS is?

Is it roughly equivalent to Ubuntu Server 20.10 on RPi 4?

Are both measurably faster than 32-bit RaspiOS on RPi 4 with 2GB or 4GB RAM?

Any outstanding/major risks with 64-bit RaspiOS field communities need to watch for?

georgejhunt, 2020-11-06:
> Any guess how much faster 64-bit Raspberry Pi OS is?
Since I had a 64bit desktop already built to test the automatic browser
home page, I searched for a similar SD card (SanDisk extreme). Both were
loaded with IIAB min, and I timed the startup from power on until the Home
page was populated.
64bit == 39sec
32bit == 36sec
So I conclude at the first approximation that the 32bit OS is 10% faster.

holta, 2020-11-06:
Other Data Points &mdash; Kolibri + MEDIUM-sized IIAB on 64-bit RaspiOS Lite on RPi 4 w/ 4GB RAM [*] took...

- 53 seconds (avg) to reboot when kolibri.service was enabled (4 tries, 45-to-61 seconds)
- 38 seconds (avg) to reboot when kolibri-serice was disabled (4 tries, 37-to-41 seconds)

...measured up to the point where RaspiOS's login prompt appeared on the attached screen.

[*} and a large microSD card full of content.

holta, 2020-11-06:
WiFi firmware 7.45.0.18 (for 32 simultaneous WiFi devices) acts unstably on this OS.  It "works" (sort of) but pages take 1-2 to load even when the WiFi connection doesn't drop.

Android 11 clients seem to figure out that the WiFi connection is wholey inadequate, and intentionally drops the WiFi connection within about 2 minutes.  Windows and Ubuntu WiFi client devices (2 laptops attempted) seem to try to keep the WiFi connection alive, even when...maybe they shouldn't!

This means 64-bit RaspiOS likely cannot be used it most classroom.  The jury is out on other 64-bit OS's for RPi like Ubuntu 20.04.1 and 20.10

Schools will have to stick with 32-bit OS's in most cases it appears.  I've added a quick/cautionary note in the top-left of Table 1 here...and some idle gossip about Cypress firmware 7.45.221 just released 2020-09-25 for those running tiny kiosks...and so not needing many WiFi client devices: https://github.com/iiab/iiab/issues/823#issuecomment-662285202

holta, 2020-11-07:
I forgot to mention that wifi firmware 7.45.0.18 tricks you in the 1st-minute-or-few before it goes haywire (on 64-bit RaspiOS).

It's best to click on a lot of different content (actual, installed content) on IIAB's main page (http://box or http://box.lan) so that the WiFi failure is quickly induced usually within 1-or-2 minutes.

Video-intensive pages like http://box/kiwix/teded_en_all_2020-06/ would seem to help induce the failure mode even more quickly / maybe?

jvonau, 2020-11-09:
There are testing users who don't care if the WiFi hotspot is functioning, the successful install reports returned may not be complete in testing coverage as one may think. I suggest re-evaluation of the universal use with wifi_hotspot_capacity_rpi_fix: True in local_vars.yml in use since 09cdf28643542d56b4bc31ecf3dc679d7e484d97. The "fix" has now demonstrated not to be universal, had I thought you were going to go against my wishes to have the rollback to be voluntarily and opt-in, I would never of written the code to begin with. Let the end-user chose to be part of the experimental testing as this operation is currently one way and hard for the end-user to recover from the change. Lets see what 2624 brings to light but being able to perform back-2-back comparisons being an opt-in, the way I envisioned is a pile easier when evaluating while not intruding into other use cases that don't require 32-clients on the WiFi AP or prefer the newest firmware for other valid reasons.

jvonau, 2021-04-23:
#2739 or close this one in favor of 2748

holta, 2021-05-21:
This appears largely resolved since 2020.  The WiFi hotspot now appears to work well with WiFi firmware 7.45.18.0 on both 64-bit versions of RaspiOS on RPi 4.

Please move followup discussion to #2748.

closed 2021-05-21 by holta

holta, 2022-03-11:
FYI https://downloads.raspberrypi.org/raspios_lite_arm64/images/raspios_lite_arm64-2022-01-28/2022-01-28-raspios-bullseye-arm64-lite.zip would appear to work well with the latest pre-release of [IIAB 8.0](https://github.com/iiab/iiab/wiki/IIAB-8.0-Release-Notes) !

This could become especially important as it allows for the installation of [calibredb](https://manual.calibre-ebook.com/generated/en/calibredb.html) (5.12 for now) in order to bulk important books/docs/etc into Calibre-Web.

Reason: it's no longer possible to install the calibre apt package (including the calibredb command) onto the Bullseye version of 32-bit Raspberry Pi OS.  As discussed here: https://www.mobileread.com/forums/showthread.php?t=343422

Ref:

- #3131
- #3132