iiab/iiab #2424: raspbian-64bit-dnsmasq does not start because br0 is not up

georgejhunt, 2020-05-29:
http://dpaste.com/0XAN7CQ
I think I've seen this before, but I don't remember the cause
/etc/dnsmasq.d/iiab.conf L6 is set to br0. 
The systemctl status says "dnsmasq: unknown interface br0
But br0 is up, and has a fixed 172.18.96.1 ip.

jvonau, 2020-05-29:
Inaccurate error description, the install/networking both stopped before ever reaching dnsmasq
>dhcpcd_result = enabled
network_manager_active = False
systemd_networkd_active = False
wan_in_interfaces = False

dhcpcd_result show that dhcpcd is in use above. 
>2020-05-28 20:59:50,009 p=1706 u=root n=ansible | TASK [network : NetworkManager in use] *****************************************
2020-05-28 20:59:50,059 p=1706 u=root n=ansible | skipping: [127.0.0.1]
2020-05-28 20:59:50,144 p=1706 u=root n=ansible | TASK [network : systemd-networkd in use] ***************************************
2020-05-28 20:59:50,190 p=1706 u=root n=ansible | skipping: [127.0.0.1]
2020-05-28 20:59:50,277 p=1706 u=root n=ansible | TASK [network : Raspbian uses dhcpcd only with no N-M or SYS-NETD active] ******
2020-05-28 20:59:50,309 p=1706 u=root n=ansible | skipping: [127.0.0.1]
2020-05-28 20:59:50,397 p=1706 u=root n=ansible | TASK [network : Not RPi, Not NetworkManager, Not systemd-networkd in use] ******
2020-05-28 20:59:50,565 p=1706 u=root n=ansible | included: /opt/iiab/iiab/roles/network/tasks/debian.yml for 127.0.0.1
2020-05-28 20:59:50,799 p=1706 u=root n=ansible | TASK [network : For upgrades from earlier IIAB 6.2, remove br0 file] ***********
2020-05-28 20:59:50,847 p=1706 u=root n=ansible | skipping: [127.0.0.1]
2020-05-28 20:59:50,933 p=1706 u=root n=ansible | TASK [network : Supply resolvconf.conf] ****************************************
2020-05-28 20:59:52,225 p=1706 u=root n=ansible | ok: [127.0.0.1]
2020-05-28 20:59:52,315 p=1706 u=root n=ansible | TASK [network : Supply dhcpcd.conf] ********************************************
2020-05-28 20:59:53,647 p=1706 u=root n=ansible | ok: [127.0.0.1]
2020-05-28 20:59:53,734 p=1706 u=root n=ansible | TASK [network : Copy the bridge script] ****************************************
2020-05-28 20:59:53,893 p=1706 u=root n=ansible | fatal: [127.0.0.1]: FAILED! => {"changed": false, "msg": "AnsibleUndefinedVariable: 'gui_desired_network_role' is undefined"}
2020-05-28 20:59:53,897 p=1706 u=root n=ansible | PLAY RECAP *********************************************************************
2020-05-28 20:59:53,897 p=1706 u=root n=ansible | 127.0.0.1                  : ok=86   changed=17   unreachable=0    failed=1    skipped=90   rescued=0    ignored=1   

Should be using ''network : Raspbian uses dhcpcd only with no N-M or SYS-NETD active" ,
'when: is_raspbian' is set in OS.yml while iiab.env show:

>OS=debian
OS_VER=debian-10

Think with the Raspbian OS name change we may have to revisit the OS detection for that OS.

jvonau, 2020-05-29:
`cat /etc/os-release` would be useful information, ID is used in scripts/local_facts.fact which forms the base of OS_VER, from an older image below
>pi@box:~ $ cat /etc/os-release 
PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"
VERSION_CODENAME=buster
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"

holta, 2020-05-29:
Just fyi the current 64-bit Beta of RaspiOS shows:

```
root@raspberrypi:~# cat /etc/os-release
PRETTY_NAME="Debian GNU/Linux 10 (buster)"
NAME="Debian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"
VERSION_CODENAME=buster
ID=debian
HOME_URL="https://www.debian.org/"
SUPPORT_URL="https://www.debian.org/support"
BUG_REPORT_URL="https://bugs.debian.org/"

root@raspberrypi:~# cat /etc/issue
Debian GNU/Linux 10 \n \l

root@raspberrypi:~# cat /etc/issue.net
Debian GNU/Linux 10

root@raspberrypi:~# cat /etc/debian_version
10.4

root@raspberrypi:~# cat /etc/rpi-issue
Raspberry Pi reference 2020-05-27
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 30e2dd32ba47cc3bec15ab1413c16a17e5797775, stage4
```

holta, 2020-05-29:
There are other issues with 64-bit RaspiOS, but this one (#2424) appears solved by PR #2426 (now merged).

closed 2020-05-29 by holta