iiab/iiab #2428: Use nginx gzip_static for storage savings on static modules

tim-moody, 2020-05-31:
As an experiment I created a gzip directory under the web root and put a gzipped hello.html file there. With the following directive nginx unzips on the fly. Even better it tries the non-zipped file if it does not find the gzipped file, so no need to gzip everything. For example the image I included only saved 1%, so not really worth it. This is huge for things in modules, like cdc which has 5G of html that should compress quite a bit. These settings are copy/paste so improvement may be possible.

```
location /gzip {
gzip_static  always;
gzip_proxied expired no-cache no-store private auth;
gunzip       on;
}

```

This

jvonau, 2022-04-05:
This seem easy to implement why 8.1 on almost 2 year old issue?

holta, 2022-04-05:
> This seem easy to implement why 8.1 on almost 2 year old issue?

I've changed it to 8.0 so the question is at least addressed:

I don't know the answer.

@tim-moody may be able to clarify if this is worth it &mdash; whether a priority or not?