iiab/iiab #2429: Review search engine

traverseda, 2020-05-31:
I've been working on a search engine for my personal archive, and I think it could be a good fit for IIAB.

Unfortunately I'm not an IIAB user. I'd like some people to install the project on their IIAB instances, test it out, and figure out what needs to happen for this to be included as part of IIAB. Play around with it, open up some tickets, generally help me polish it up a bit.

https://github.com/traverseda/iiab-searchServices

I've done my best to make this easy to install and deploy.

traverseda, 2020-05-31:
I have simplified this a great deal since the search engine API proposal, my goal right now is to just get something that's good enough out the door sooner rather than later, and then to iterate.

tim-moody, 2020-05-31:
> This does not currently work!

according to the readme

holta, 2020-05-31:
fyi @mitra42 if you can consider in early June?

traverseda, 2020-05-31:
It's not in a complete state, I wouldn't want the general public to try using it. One of the big features that's missing is the ability to remove urls. That's on my toDo list, but I think it's important to get feedback early in the process in case there are any major issues.

It does currently work, but it is in the middle of some significant refactors. The schema may change before a 1.0 release, or other breaking changes.

Anyway, I've changed the readme to say that it's an early pre-release instead, although at this point I make no guarantee that a particular commit will work.

holta, 2020-06-03:
I will ask others to take a look & try during our IIAB weekly call (10AM NYC Time call this Thursday off of http://minutes.iiab.io).

Starting here: https://github.com/traverseda/iiab-searchServices/blob/master/README.md

traverseda, 2020-06-04:
Worth noting that this isn't a PR *yet*. Want I want to get out of this is some more concrete information on exactly what features IIAB is going to need me to add. I can imagine a lot of things that might be useful, but without any significant experience with IIAB's internals I will be just guessing.

Right now I'm writing it to fill my needs, and to generally make it flexible enough for other people to use, but I imagine there are specific requirements that IIAB is going to have for something like this, even if it's as simple as "let as change the URL the title text points to".

I'm happy to help integrate into IIAB, but I'm not really familiar with the internals, so stuff like writing cronjobs or jobs that integrate all the apps with this search panel are going to be outside of my purview, for example.

tim-moody, 2020-06-04:
I think the most important thing to keep in mind is that all content included in IIAB originates from somewhere else. The particular content items included are curated by implementers and vary from deployment to deployment.

traverseda, 2020-06-04:
Of course, yeah. I'm expecting to write some docs about how to integrate an app with LCARS, and any effort towards actually doing that to be pretty gradual.

holta, 2020-10-29:
FYI on tangentially related note Content choices should rise dramatically in coming years thanks to:

> New scraping strategy possible, with WARC to ZIM: (does “headless Chrome” help?!)
> https://lists.wikimedia.org/pipermail/offline-l/2020-October/001641.html

Background/Context to keep this search design challenge on track:
#2218 #2219

holta, 2025-01-01:
Progress here, using a fully offline[*] LLM.  Thanks to @Abdulrahman392011.  Please see:

- #3796

[*] Needs testing to confirm!!