iiab/iiab #2431: Admin Console login fails, 3 installs in a row

holta, 2020-06-04:
Thanks to Luc in Tokyo for the report:

> As I was preparing for classes this afternoon, I did a complete clean install of 32-bit RaspiOS from the imager and updated.
>
> I ran the one line install and selected the BIG install.  I did not modify the VAR file or change the iiab password.
>
> Same result of invalid login while being able to access the content of the server on my client machine running Ubuntu 20.04  on the Chrome Browser.   
>
> here is the new diagnostic   http://dpaste.com/1QN5DXC

Also here is his earlier report if nec:

> http://dpaste.com/1YXM468

CLARIF: @tim-moody asked for the results of `iiab-diagnostics /etc/group /var/log/uwsgi/app/admin-console.log` (hence the contents of those 2 files, appearing in both dpaste's above).

lucgougeon, 2020-06-04:
Please find a screenshot of the login.  

[image]

jvonau, 2020-06-04:
Observation: when testing iiab-admin's PR 327, post-install/pre-reboot I was not able to login, but once rebooted I was able to login.

tim-moody, 2020-06-04:
I looked at several pastebins. In each case authentication is running and the log says the user was invalid. I am running a new test on latest raspbios.

tim-moody, 2020-06-04:
One theory was that these python modules are not present, but it has proven falseL
import crypt
import spwd
import grp
import secrets

tim-moody, 2020-06-04:
Please verify that you can ssh in as iiab-admin. Please also test whether you can login as pi.

tim-moody, 2020-06-04:
I just did a retest:
wrote 2020-05-27-raspios-buster-lite-armhf.img to sdcard
copied medium vars to /etc/iiab
cd /opt/iiab/iiab
./iiab-install
cd /opt/iiab/iiab-admin-console
./install
on completion I successfully logged into Admin Console as iiab-admin without rebooting

holta, 2020-06-04:
@lucgougeon are you able to respond to Tim's 2 questions below...today if possible?  Thanks!

> Please verify that you can ssh in as iiab-admin. Please also test whether you can login [to Admin Console, and over ssh to your IIAB machine] as pi.

holta, 2020-06-04:
FYI this IIAB install was built on Raspberry Pi OS (32-bit) with desktop...but WITHOUT recommended software (middle of the 3 available tiers!)  Same as @lucgougeon is using.  As reconfirmed by `stage4` here:

root@raspberrypi:~# cat /etc/rpi-issue
Raspberry Pi reference 2020-05-27
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 30e2dd32ba47cc3bec15ab1413c16a17e5797775, stage4

holta, 2020-06-04:
Oops my mistake, the above 10.8.0.58 is 64-bit RaspiOS.

But otherwise almost identical to @lucgougeon's 32-bit.

As both are BIG-sized installs of IIAB stage4 = RaspiOS Desktop w/o recommendd software.

tim-moody, 2020-06-04:
/etc/passwd and /etc/shadow must be readable by www-data, but I don't know why the issue is not universal. are these raspbios or lite?

holta, 2020-06-04:
> /etc/passwd and /etc/shadow must be readable by www-data, but I don't know why the issue is not universal. are these raspbios or lite?

1) See above https://github.com/iiab/iiab/issues/2431#issuecomment-638914213 -- so far @lucgougeon and I are seeing this failure on RaspiOS with Desktop, and have not tested RaspiOS Lite.

2) BIG-sized IIAB seems to be affected every time...whereas MIN-sized IIAB is not?

tim-moody, 2020-06-04:
problem line is spwddb = spwd.getspnam(user)
on .58 /etc/passwd was readable by world, but /etc/shadow was readable by group
www-data needs to be in that group

tim-moody, 2020-06-04:
the group is shadow, so need to add www-data to shadow

holta, 2020-06-04:
@tim-moody @jvonau: is it possible that `www-data` IS being properly added to the 'shadow' group in /etc/group (thanks to https://github.com/iiab/iiab/blob/master/roles/nginx/tasks/install.yml#L19-L22 invoked in Stage 3) only to be removed later on during BIG-sized IIAB installs?

(This would seem to the case according `/opt/iiab/iiab/iiab-install.log` on 10.8.0.58)

tim-moody, 2020-06-04:
yes, I thought it was somewhere

tim-moody, 2020-06-04:
for whatever reason www-data is not in the shadow group on .58

tim-moody, 2020-06-04:
```
- name: Install NGINX if 'nginx_installed' not defined, e.g. in {{ iiab_state_file }}    # /etc/iiab/iiab_state.yml
  include_tasks: install.yml
  when: nginx_installed is undefined
```

tim-moody, 2020-06-04:
please do the following
cd /opt/iiab/iiab-admin-console
git pull
./install

closed 2020-06-04 by holta

lucgougeon, 2020-06-05:
It worked, thank you very much.

PR #2434 (2020-06-04): Fix for #2431: roles/httpd should preserve user www-data in group shadow
Fixes #2431.

Tested on BIG-sized RaspiOS with desktop.

Related: https://github.com/iiab/iiab-admin-console/pull/329