iiab/iiab #2433: Nextcloud 19.0.0's new warning: "secure connection...generating insecure URLs...behind a reverse proxy...overwrite config variables are not set correctly" ?

holta, 2020-06-04:
@floydianslips & All: have you seen this new orange warning within http://box/nextcloud > Settings (within dropdown in top-right) > Administration / Overview (on left) ?

> You are accessing your instance over a secure connection, however your instance is generating insecure URLs. This most likely means that you are behind a reverse proxy and the overwrite config variables are not set correctly. Please read [the documentation page about this](https://docs.nextcloud.com/server/19/go.php?to=admin-reverse-proxy).

Is there something we can / should do about this for IIAB 7.1's release later this week?

NOTE the above warning is above-and-beyond the usual Nextcloud warning &mdash; that we're all well aware of &mdash; strongly advising to move towards SSL/TLS (https) when possible:

> Accessing site insecurely via HTTP. You are strongly advised to set up your server to require HTTPS instead, as described in the [security tips ↗](https://docs.nextcloud.com/server/19/go.php?to=admin-security).

jvonau, 2020-06-07:
The question is how to set 'trusted_proxies', direct editing of config/config.php or can 'occ' handle that in the same way as 'trusted_domains'?

jvonau, 2022-04-05:
still an issue?

holta, 2022-04-05:
> still an issue?

This Nextcloud warning has not appeared in a long time now.  Good enough.

closed 2022-04-05 by holta