iiab/iiab #2445: Lokole (community email) opportunities for tighter integration

holta, 2020-06-08:
Nzola Swasisa & Clemens Wolff (@c-w) have outlined 2 opportunities for IIAB integration here, that we should investigate:

1) Enable Lokole to use a connected cellphone for internet access: https://github.com/ascoderu/lokole/issues/453

2) Configure Lokole on Internet-in-a-Box to run the email synchronization: https://github.com/ascoderu/lokole/issues/454

nzola, 2020-06-10:
Hello software and code developers, we need your help to complete Lokole email full integration on iiab box. See details here above.
There is a great need for this full email app for use in the Republic Democratic of the Congo (DRC). This country's population is more than 80 million, the second largest country of Africa in territory size, but less than 10% of the population use Internet. Because most the people cannot afford WhatsApp SIM data/credit/plan, business communications of the government, most of private local organizations and all educational institutions is chaotic.  Lokole in the iiab will be affordable and sustainable for them to use. For more info: www.ascoderu.ca and https://github.com/ascoderu. Thanks

nzola, 2020-07-03:
From Clemens;
Thanks for the reply. Here are some thoughts:

Good for off-line emails as with IIAB, for a school or a small "closed" community.
To be effective, it needs to have the "sync" capability with wider email community.
The original design goal for Lokole on IIAB was to be a closed local chat community since we assumed that IIAB deployments don't have internet access. On our main Lokole product outside of the IIAB integration, we do have internet sync capabilities. First we run WvDial to connect to internet via a USB modem (see StartInternetConnection) and then we connect to our email service on Azure to upload emails sent from the client and download emails sent to the client (see SyncEmails). We put a special eye to minimizing bandwidth usage to reduce costs as much as possible. The Lokole product that we deploy to users in the Congo DRC uses this connectivity functionality to periodically or on-demand sync emails.

So you really don't need a dongle to do the "sync".
Let me think about this. I am NOT a developer, but may be we can ask the wider IIAB Community to look at the feasibility as you have suggested.
Can you let me what skills are required and the amount effort needed.
It's an interesting idea to replace the USB modem with a tethered phone. We already have a modality that assumes that the Raspberry Pi has access to a normal internet connection so that we skip the WvDial step (e.g. see code for hard-wired Ethernet scenarios). This means that as long as the phone can provide a normal Linux connectivity device similar to eth0, Lokole should be able to reuse this internet connection without any code changes. Is this how you envision the tethered phone to work? If not: how do you expect the phone to provide the internet connection to apps running on IIAB? I created an issue in the Lokole backlog to track this investigation (see #453).

Note that for performance and reliability reasons, the synchronization runs in a background worker (implemented in Celery, see run-celery.sh and run-crontab.sh) which currently isn't being run on the Lokole IIAB setup. Thus, a piece required for enabling email sync on Lokole on IIAB would be to extend the IIAB Lokole Ansible template to not only run the Lokole webapp but also the background workers. The amount of effort required for this change should be fairly minimal: adding a few more scripts to the Ansible template and testing the integration. The skills required for this work are Ansible, Bash and Python. I created an issue in the Lokole backlog to track this work and provide more background (see #454). If there's someone in your community who would be interested in picking up this work, I can also provide more technical support.

Last but not least, to control the amount of traffic to our email synchronization service in Azure, we require an explicit registration step for each new Lokole client. In our existing setup, the registration step happens as part of the installation of the Lokole software on the Raspberry Pi (see ClientSetup). To enable this registration on an ad-hoc basis for Lokole running on IIAB, I suggest to add a form to the Lokole application that takes the user through the registration flow (see #454 for additional details). This work will require some more intimate knowledge of the Lokole codebase, so if we go forward with this project and others can help with the modification of the IIAB Lokole Ansible templates and cellphone tethering investigation as described earlier in this email, I can commit to implementing this.

Hope this helps clarify some of my thoughts. Let me know what additional detail I can provide.

nzola, 2020-07-03:
From Clemens;
Thanks for the reply. Here are some thoughts:

Good for off-line emails as with IIAB, for a school or a small "closed" community.
To be effective, it needs to have the "sync" capability with wider email community.
The original design goal for Lokole on IIAB was to be a closed local chat community since we assumed that IIAB deployments don't have internet access. On our main Lokole product outside of the IIAB integration, we do have internet sync capabilities. First we run WvDial to connect to internet via a USB modem (see StartInternetConnection) and then we connect to our email service on Azure to upload emails sent from the client and download emails sent to the client (see SyncEmails). We put a special eye to minimizing bandwidth usage to reduce costs as much as possible. The Lokole product that we deploy to users in the Congo DRC uses this connectivity functionality to periodically or on-demand sync emails.

So you really don't need a dongle to do the "sync".
Let me think about this. I am NOT a developer, but may be we can ask the wider IIAB Community to look at the feasibility as you have suggested.
Can you let me what skills are required and the amount effort needed.
It's an interesting idea to replace the USB modem with a tethered phone. We already have a modality that assumes that the Raspberry Pi has access to a normal internet connection so that we skip the WvDial step (e.g. see code for hard-wired Ethernet scenarios). This means that as long as the phone can provide a normal Linux connectivity device similar to eth0, Lokole should be able to reuse this internet connection without any code changes. Is this how you envision the tethered phone to work? If not: how do you expect the phone to provide the internet connection to apps running on IIAB? I created an issue in the Lokole backlog to track this investigation (see #453).

Note that for performance and reliability reasons, the synchronization runs in a background worker (implemented in Celery, see run-celery.sh and run-crontab.sh) which currently isn't being run on the Lokole IIAB setup. Thus, a piece required for enabling email sync on Lokole on IIAB would be to extend the IIAB Lokole Ansible template to not only run the Lokole webapp but also the background workers. The amount of effort required for this change should be fairly minimal: adding a few more scripts to the Ansible template and testing the integration. The skills required for this work are Ansible, Bash and Python. I created an issue in the Lokole backlog to track this work and provide more background (see #2445). If there's someone in your community who would be interested in picking up this work, I can also provide more technical support.

Last but not least, to control the amount of traffic to our email synchronization service in Azure, we require an explicit registration step for each new Lokole client. In our existing setup, the registration step happens as part of the installation of the Lokole software on the Raspberry Pi (see ClientSetup). To enable this registration on an ad-hoc basis for Lokole running on IIAB, I suggest to add a form to the Lokole application that takes the user through the registration flow (see #2445 for additional details). This work will require some more intimate knowledge of the Lokole codebase, so if we go forward with this project and others can help with the modification of the IIAB Lokole Ansible templates and cellphone tethering investigation as described earlier in this email, I can commit to implementing this.

Hope this helps clarify some of my thoughts. Let me know what additional detail I can provide.

jvonau, 2021-07-21:
close

closed 2021-07-21 by holta