iiab/iiab #2446: Let's get 3+3 IIAB images published for RPi

holta, 2020-06-12:
Thanks to @georgejhunt who's done hard work documenting this in recent months, and is working on a couple key Python scripts here to streamline the regular (quaterly or so) updates required here:

- New top level for rpi-imager  PR iiab/iiab-factory#114 (merged)
- Rework upload.py PR iiab/iiab-factory#116

@jvonau and @tim-moody have also done a ton of work here, so let's try to get aligned, ideally with no more than (about!) 3 + 3 = 6 RPi images published off the TOC (Table of Contents) on the right-side of https://github.com/iiab/iiab/wiki > Raspberry Pi Images

Here's that 3 + 3 = 6 Summary Suggestion from yesterday's weekly call (http://minutes.iiab.io) that I'm quite happy with for starters: (though this will surely evolve!)
https://github.com/iiab/iiab-factory/pull/113#issuecomment-642819993

In short, newcomers/newbies demand and really do need rather extreme simplicity ("paradox of choice").  So let's try our best to keep this from ballooning above and beyond 3 + 3 = 6 images if at all possible (each of which needs to be rebuild regularly, so everyone pays a large cost if we fail to keep this tight !!)

tim-moody, 2020-06-12:
>Here's that "3 + 3 = 6 RPi images" Summary Suggestion from yesterday's weekly call (http://minutes.iiab.io) that I'm quite happy with for starters:

To which item are you referring. I see no definitive list of 3+ 3 images

holta, 2020-06-12:
So far it's 3 + 1 RPi images as described here by the 2 current .json files herein:

https://github.com/iiab/iiab-factory/pull/113#issuecomment-642819993

(But let's get more details from @jvonau around his 2 images, that should make it 3 + 3 RPi images.)

tim-moody, 2020-06-12:
I'd like raspbian lite min, medical, and medium. People who want desktop can do their own install.

georgejhunt, 2020-06-12:
Tim,
Can you transfer the new top level menu at
https://github.com/iiab/iiab-factory/blob/master/box/rpi/iiab-imager/os_list_imagingutility.json
to
http://iiab.me, so that we have the may 27 2020 versions of raspbian
available from our menu system? Thanks

georgejhunt, 2020-06-12:
Tim,
Are you planning to create the images that you want?

holta, 2020-06-12:
> People who want desktop can do their own install.

Not exactly (most cannot, literally ;)

As the desktop (graphical) version of RaspiOS-formerly-Raspbian is the #1 request among IIAB newcomers by far (even when I ask them to seriously consider headless they refuse essentially every time, as these folks are getting to know Linux at the same time, and often want their teachers exposed to RaspiOS Desktop for similar reasons).

i.e. this newcomer/newbie crowd is precisely the demographic that most needs a prefab image, whereas folks who understand servers are far more Linux-fluent.

Not that it's one monolithic demographic, but something like 90% newcomers literally beg for the Graphical/Desktop version of the OS, so we need to offer that very prominently (even if they grow out of it later, etc!)

tim-moody, 2020-06-12:
top level has been transferred.

>Are you planning to create the images that you want?

yes, building as we speak, and add them to sub_items.

was thinking of a u20 64 medium for experimental

tim-moody, 2020-06-13:
To be clear, I don't object if someone wants to do a desktop image or if Jerry has functionally equivalent images. But '90% of newcomers literally beg' stretches my imagination unless you cherry pick the population, for example leaving out the 80 units that JH shipped in 2019.

georgejhunt, 2020-06-13:
great, thank you

tim-moody, 2020-06-14:
I have updated subitems. Don't know if it will be controversial.

holta, 2020-06-14:
RaspiOS Desktop (# 1 request by school implementers) & Ubuntu 20.04 (experimental) are also needed.

@tim-moody if you aren't doing those please let us know, so that we can arrange for someone else to do those.

tim-moody, 2020-06-14:
>To be clear, I don't object if someone wants to do a desktop image

Don't think I'm the right person as I've never built desktop images.

I'll do ub20 64 medium if no one else wants to.

georgejhunt, 2020-06-14:
I can do the desktop image. Which IIAB flavor?  On one of our calls we had
coined the phrase "kitchen sink"  for an image that is based upon raspbian
-full and IIAB big. But I think it is 13+GB. If I were a newbie, I might
prefer that. But ...
I'm inclined to raspbian desktop (not full), and IIAB medium.

tim-moody, 2020-06-14:
I was working on a one-size-fits-all, but got stuck knowing what to include. I was thinking more or less a medium installed, but with only a min enabled. So people could turn on other services without having to install, but not have the burden of all those services running if not desired.

tim-moody, 2020-06-14:
I confess I don't know the difference between desktop and full.

holta, 2020-06-14:
> I can do the desktop image. Which IIAB flavor?

MEDIUM-sized IIAB on RaspiOS Desktop would be great (the highest priority it appars) if you can @georgejhunt !

Personally I'd avoid the version of RaspiOS that contains additional "recommended software" for now, to avoid that final ~2GB.

holta, 2020-06-14:
> I was working on a one-size-fits-all, but got stuck knowing what to include. I was thinking more or less a medium installed, but with only a min enabled. So people could turn on other services without having to install, but not have the burden of all those services running if not desired.

@jvonau is working on something a lot like that.

tim-moody, 2020-06-14:
I'll do ub20.

tim-moody, 2020-06-14:
btw I did a test imager install on my laptop of medical and medium. both boot and bring up home page and adm cons.

tim-moody, 2020-06-14:
I agree desktop + medium. medium has cap port installed, but not enabled. is that right?

holta, 2020-06-14:
> cap port installed, but not enabled. is that right?

All MIN/MEDIUM/BIG installs of IIAB currently do that, yes &mdash; i.e. Captive Portal is installed but not enabled.

tim-moody, 2020-06-14:
how do you enable it?

holta, 2020-06-14:
> how do you enable it?

`./runrole captiveportal` after modifying /etc/iiab/local_vars.yml similar to most every other service/app.

This is detailed for newcomers/newbies here:

http://FAQ.IIAB.IO > "38 Captive Portal Administration: What tips & tricks exist?"

tim-moody, 2020-06-14:
medium on ub 20 added

holta, 2020-06-14:
Refs:

iiab/iiab-factory#113 "change ubuntu OS description"
iiab/iiab-factory#114 "new top level for rpi-imager"
iiab/iiab-factory#115 "More recent upload.py should be in master"
iiab/iiab-factory#116 "rework upload.py"
iiab/iiab-factory#117 "Working"
iiab/iiab-factory#118 "secure script needs updating"
iiab/iiab-factory#119 "add 7.1 on ub20 to experimental"

tim-moody, 2020-07-11:
rebuild awaiting testing of osm fixes

georgejhunt, 2020-07-11:
I verified that the new regions.json at
http://download.iiab.io/content/OSM/vector-tiles/maplist/hidden/assets/regions.json
works.
So the new images require php7.3-sqlite3 and the new regions.json for
"world to zoom 10" to work properly.

holta, 2020-09-03:
Good enough for now!

closed 2020-09-03 by holta