iiab/iiab #2451: wifi firmware symptoms -- cannot reassociate within 30 min of restart

georgejhunt, 2020-06-18:
This is just an early warning. See
http://dpaste.com/0K7YKW7 -- see the syslog snippet.
I was playing with moodle, and testing at box.lan, and reading internet on a local AP that is connected to internet. -- back an forth a few time. Then box.lan would not longer associate. Did a restart of box.lan. The symptom repeated.
The full syslog is at http://dpaste.com/27D81HG

jvonau, 2020-06-21:
>install_date = 2020-06-13T05:17:23Z
-rw-r--r-- 1 root root 142 Jun 14 20:52 /etc/wpa_supplicant/wpa_supplicant.conf

It appears that wlan0 was configured in the past but is now not being connected and eth0 is now providing the gateway. I suspect `iiab-install --reinstall` was employed to enable moodle while wlan0 was not connected to the upstream AP which would not allow the frequency|channel detection to alter the hostapd.conf to match upstream so the default channel 6 was selected to be used but I suspect that is not what the upstream AP is using. It should be noted that once hostapd is started 'scanning' for nearby BSSIDs is unavailable via the desktop or wpa_cli but if wpa_supplicant.conf is populated the connection should succeed anyway if the channels match. I would suggest to run `sudo iiab-hotspot-off` stopping hostapd to allow  'scanning' and then try running `wpa_cli -i wlan0 scan` then `wpa_cli -i wlan0 scan_results` and note if your upstream AP is present and what frequency is in use. Now if your AP is present try running `sudo test-wifi | tee -a /tmp/test.txt` and attach /tmp/test.txt to this issue please, then reboot and see if wlan0 connects for you.

georgejhunt, 2020-06-21:
This SD card started as an image that I created (desktop version). The
contents of the wpa-supplicant were rewritten during the last of the image
creation script.

root@box:/opt/iiab/iiab# cat /etc/wpa_supplicant/wpa_supplicant.conf

ctrl_interface=DIR=/var/run/wpa_supplicant GROUP=netdev

update_config=1

country=US

network={

ssid="<your ssid>"

psk="<your pre-shared key>"

}

root@box:/opt/iiab/iiab#
I suppose it's possible that what I chose as a network config confused
network setup.

So following your suggestion to discover the upstream yields the following:

root@box:~# wpa_cli -i wlan0 scan

OK

root@box:~# wpa_cli -i wlan0 scan_results

bssid / frequency / signal level / flags / ssid

e8:8d:28:61:9b:d4 5785 -32 [WPA2-PSK-CCMP][ESS] testnetwork

40:b0:76:4c:b4:6c 2437 -62 [WPA2-PSK-CCMP][WPS][ESS] barb

2c:4d:54:b6:3a:7c 2437 -33 [WPA2-PSK-CCMP][WPS][ESS] test

e8:8d:28:61:9b:d3 2412 -28 [WPA2-PSK-CCMP][ESS] testnetwork

fe:01:7c:13:ba:e7 2437 -48 [WPA2-PSK-CCMP][WPS][ESS][P2P] DIRECT-e7-HP M426
LaserJet

20:25:64:6a:6d:08 2462 -52 [WPA-PSK-CCMP+TKIP][WPA2-PSK-CCMP+TKIP][WPS][ESS]
GreenWave

20:25:64:6b:77:30 5240 -73 [WPA-PSK-CCMP+TKIP][WPA2-PSK-CCMP+TKIP][WPS][ESS]
GreenWave

1c:b7:2c:35:32:c8 2422 -75 [WPA2-PSK-CCMP][WPS][ESS] Harmony Room 2.4

2a:25:64:6b:77:30 5240 -73 [ESS] xfinitywifi

02:e9:80:15:5f:a4 2437 -9 [ESS] raspbian-64bit

02:f4:04:c8:2a:ff 2437 -10 [ESS] Internet in a Box

2a:25:64:6a:6d:08 2462 -54 [ESS] xfinitywifi

00:c2:c6:aa:06:bd 2437 -27 [ESS] Internet in a Box

30:10:b3:a1:8f:c1 2437 -39 [ESS] Ubuntu16.4

3a:25:64:6b:77:30 5240 -73 [WPA-PSK-CCMP+TKIP][WPA2-PSK-CCMP+TKIP][ESS]

1c:b7:2c:35:32:c9 2422 -71 [WPA2-PSK-CCMP][ESS] Health Connected Guest 2.4

36:25:64:6b:77:30 5240 -72 [WPA-EAP-CCMP+TKIP][WPA2-EAP-CCMP+TKIP][ESS]

root@box:~#

There's lots of competition on 2437 which I think is channel 6.
I'm going to see if the AP on the test machine continues to work if the
upstream is connected continuously.
I just read the https://github.com/raspberrypi/firmware/issues/1403 that
Jerry found. The log look very similar to our log.

Maybe we should be testing the firmware mentioned in the conversation,
which in their experience fixed the problem.

Thanks Jerry for your responses.

jvonau, 2020-06-21:
> This SD card started as an image that I created (desktop version). The contents of the wpa-supplicant were rewritten during the last of the image creation script. root@box:/opt/iiab/iiab# cat [...]
 
Think the 'network' section needs to be commented out if the ssid and psk values are not real and that is the source of 
>brcmf_run_escan: error (-110)
 
>So following your suggestion to discover the upstream yields the following: root@box: wpa_cli -i wlan0 scan OK root@box:~# wpa_cli -i wlan0 scan_results bssid / frequency / signal level / flags / 
snip
There's lots of competition on 2437 which I think is channel 6. I'm going to see if the AP on the test machine continues to work if the upstream is connected continuously. 

With hotspot-off selecting the available AP from the menu in the desktop should work or supplying a correct wpa_supplicant.conf file and `sudo test-wifi`. Once connected over wlan0 the dhcpcd hook should alter /etc/hostapd/hostapd.conf's channel line to match wlan0, I would double check that if you are seeing issues. One issue might be that the upstream AP connection is 5GHz and hostapd is using 2.4GHz only so check the channel in use on wlan0. Now just run `sudo iiab-hotspot-on` and reboot.

>I just read the [raspberrypi/firmware#1403](https://github.com/raspberrypi/firmware/issues/1403) that Jerry found. The log look very similar to our log. Maybe we should be testing the firmware mentioned in the conversation, which in their experience fixed the problem. 

Save that for the future, as the new firmware is with a reduced client limit it would appear.
>Thanks Jerry for your responses.

holta, 2020-09-12:
@georgejhunt should this ticket be closed?

Or deferred to [IIAB 8.0](https://github.com/iiab/iiab/milestone/8) ?

georgejhunt, 2020-09-12:
I think this was an early symptom of wifi firmware problems that have been understood, if not resolved.

closed 2020-09-12 by georgejhunt