iiab/iiab #2456: http://box/usb fails with Desktop version of RaspiOS when USB stick/driver is inserted AFTER boot

holta, 2020-06-25:
Thanks to @shanti-bhardwa for reporting this.

Thanks to @jvonau for helping him try to figure out why this is happening.
And if there might be a workaround (perhaps disabling the OS's automount of USB sticks/drives ?)

holta, 2020-06-25:
@shanti-bhardwa can you confirm @jvonau's suggested workaround below?

> in the file manager program one needs to "toggle 'Mount mountable volumes automatically on program startup' in Preferences"

shanti-bhardwa, 2020-06-25:
one workaround is to set the desktop to boot to CLI from Raspberry PI Configuration. Reboot to CLI, then Insert the USB drive. At this stage the USB device ( /dev/sda1) is mounted in /media/usb0
To go to desktop, type startx at command level, and the USB stick remains mounted on /media/usb0.
using 127.0.0.1/usb/USB0 now displays the folders and file on the device.

holta, 2020-06-25:
@shanti-bhardwa can you also clarify if @jvonau's suggestion works?  He wrote:

> in the file manager program one needs to "toggle 'Mount mountable volumes automatically on program startup' in Preferences"

shanti-bhardwa, 2020-06-25:
This workaround is done as follows:
1. Open up File Manger program, by clicking on the icon.
2. Click on Edit and then the last item Preferences
3. Click on Volume Management 
4, Under Auto-Mount, un-select the first two items i.e. Mount mountable Volumes automatically on program startup & Mount removable media automatically when they are inserted.
5. USB sticks will now be mounted to /media/usb0

holta, 2020-06-25:
Thanks @shanti-bhardwa & @jvonau !

To make your above recipe discoverable by others, I've documented...by linking to this ticket (#2456) from http://FAQ.IIAB.IO > "4 Can teachers display their own content?" > "Warnings"

Do you think we should try to automate your above workaround (or is the above documentation/workaround possibly quite sufficient?)

shanti-bhardwa, 2020-06-27:
Tested with IIAB 7.1 built with Ubuntu Server 20.04 + Mate Desktop 1.24.0. Again, Mate desktop behaves like Raspbian Desktop and auto mounts the USB drive on /media/user. The File Manager is Caja and it does not have the option to change the automount option from the Desktop.
I tried using this command  
gsetting set org.mate media-handling automount false 
and it works. 
Not tested with the default Ubuntu Desktop, but I guess all desktops  on Ubuntu will have some options to stop the auto mount at boot up. There is lot of info here: https://help.ubuntu.com/community/Mount/USB

shanti-bhardwa, 2020-06-27:
The final solution should be to fix the IIAB code so that it looks at /media to see if the USB sticks are mounted some where else besides the defaults of  /media/usb0.
on Admin Console, this is what happens when discovering USB media.

tim-moody, 2020-06-27:
>auto mounts the USB drive on /media/user

not sure what you mean here. all usbs mount on `'/media/user'`, all mount on `'/media/'<user>`, individual usbs mount as `usbN` under one of the two previous.

usb mount is relevant for iiab for 'pirate box' style ad hoc content and for several types of content copying in adm cons. for the former we started by only mounting if there was a share directory. then we allowed an entire usb if in /media/usbN. Now are we saying we should allow any directory of any usb mounted in /media?

>on Admin Console, this is what happens when discovering USB media. 

'this' means it finds all usb under `/media` or only usbN?

shanti-bhardwa, 2020-06-28:
1. I mean Desktops  auto mount to /media/<user>/<label of USB>  e.g. /media/shanti/4P .
2. Not sure what Admin console does, but it detects USB media on /media/usb  for build on Server OS like Raspbian Lite and also on /media/<user> on Desktop OS builds.
3. My point was that for Teacher Content, it should work the same for Server OS and Desktop OS.
4. On Desktop OS, it  the File Manager does the mount  first to /media/<user> and we have to put a work around to stop that for IIAB to detect usb media to /media/usb0 and so on.

jvonau, 2022-01-18:
#2739