iiab/iiab #2461: Maps for OSM for IIAB7.1 do not appear to be loading correctly

BagOfCoins, 2020-07-06:
### Expected behavior and actual behavior:
This is my first time posting and I don't know if this is the correct forum for raising this issue but here it goes.
It appears to me that with IIAB 7.1 that OpenStreetMap(OSM) is not working, unlike in IIAB 7.0. In IIAB 7.1, a map can be selected to download from the IIAB Admin section and the Job Status indicates that the three tasks associated with downloading and installing a map are marked "Succeeded".  But when OSM is opened, however, the OSM software appears to be working but there is no map image.  The Lat/Long moves with the mouse pointer, the zoom level changes when zoomed, but there is no map image.  There is also no response when a city is typed into the search box (i.e. no cities come up when searched).  In other words, all of the OSM functionality appears present, but there does not appear to be a loaded map.  This appears to be happening with both North America and Central America maps though other maps have not been tried.  Simple solutions like rebooting/reinstalling do not apear to correct the issue.  Other IIAB software appears to be functioning normally.  See the screenshot.
[image]
This was tried using the one-line-installer to load IIAB 7.1 on a fresh installation of Raspbian and it was tried using the image "iiab-7.1-200612-min-content-ready-g17253d4.img" and the results were the same.  
The maps loaded without any problem when using OSM in IIAB 7.0 in May 2020 (prior to the release of 7.1) when installed with the one-line-installer.  The maps also seem to load normally now when installed in IIAB 7.0 with the line "curl d.iiab.io/7.0/install.txt | sudo bash".
Maybe I'm doing something wrong but I've tried installing the OSM maps several times now using several different methods and this is my experience.  I am not super-knowledgeable about computer stuff but I can respond to simple questions about this issue.
Since you are still reading, I'd like to commend everyone working on this project as it is truly fantastic.  The improvements that you all have been making add alot of utility to the project.  IIAB really has the potential to make the world a better place and you all are making it happen.

### Steps to reproduce the problem:
Install the current (6 July 2020) version of Raspbian on a Raspberry Pi 3B+, then use the 1-line-installer to install IIAB 7.1.  Run IIAB and go to the Admin section and add the OSM map for North America.  Look at the map after installation.
OR
Install the image "iiab-7.1-200612-min-content-ready-g17253d4.img" on a Raspberry Pi3B+.  Run IIAB and go to the Admin section and add the OSM map for North America.  Look at the map after installation.

### IIAB version, OS version, hardware details:
IIAB 7.1
OS is from "iiab-7.1-200612-min-content-ready-g17253d4.img"
Raspberry Pi 3B+
Lexar 256GB MicroSD Card

### Please run the `iiab-diagnostics` command, to generate a URL summarizing your IIAB configuration, for volunteers seeking to help you: (Include that short URL here, Thanks!)
/etc/iiab/diag/7.2_raspbian-10_2020-07-06_14:00:54_BST_OSMprobIIAB71

tim-moody, 2020-07-06:
>/etc/iiab/diag/7.2_raspbian-10_2020-07-06_14:00:54_BST_OSMprobIIAB71

iiab-diagnostics should produce a pastebinit url. if you put the url here we will be able to access it remotely

tim-moody, 2020-07-06:
If I understand correctly you have downloaded maps for Central and North America. Can you try downloading World to Zoom 10 as well.

Also, can you open the javascript console and report any errors.

tim-moody, 2020-07-06:
I am able to reproduce the problem with world to zoom 10:
in javascript on http://192.168.3.96/osm-vector-maps/en-osm-omt_san_jose_2017-07-03_v0.2/

```
Uncaught Error: Unimplemented type: 3
    at Pbf.skip (index.js:166)
    at Pbf.readFields (index.js:16)
    at MVT.readFeatures (MVT.js:225)
    at VectorTile.<anonymous> (featureloader.js:61)
```

tim-moody, 2020-07-06:
http://dpaste.com//3PDQEH1

BagOfCoins, 2020-07-07:
I reloaded the IIAB 7.1 and loaded the OSM map for World to Zoom 10 and it did not work again.
It looks like tim-moody already did this but I reran the iiab-diagnostics and the results are at http://dpaste.com//1KGE384
It also looks like tim-moody dealt with the Javascript errors piece.

georgejhunt, 2020-07-07:
I am able to fix "world to zoom 10" by installing php7.3-sqlite3. 
The perplexing thing is that Tim's  lite image is ubuntu-18 from /etc/ansible/fact.d/local_facts.fact. 
Line 37 in roles/httpd/tasks/install.yml  has the (not ubuntu-16) qualifier -- which is true. So I think sqlite should have been installed.
I am sorry to be spending my time this way. I think it does not hurt to unconditionally install php-sqlite.

georgejhunt, 2020-07-07:
Maybe the problem was that ubuntu-18.yml calls out php-version 7.2, when it looks like 7.3 is really required.

georgejhunt, 2020-07-07:
I have too many rpi's. I was not debugging the machine I thought I was. My apologies

jvonau, 2020-07-07:
Apache2 wasn't installed (optional) so php7.3-sqlite3 was skipped.

georgejhunt, 2020-07-07:
I  think the root cause of this error was that httpd/tasks/install.yml at around L37 did not have "state: present". 
If correct, we will need to reissue the images.

jvonau, 2020-07-07:
> I think the root cause of this error was that httpd/tasks/install.yml at around L37 did not have "state: present".
> If correct, we will need to reissue the images.

Yup, that is an error, but the fact remains that the httpd role was not run based on the iiab_state.yml file.

holta, 2020-07-07:
Thanks @georgejhunt & @jvonau for uncovering both bugs!

We could make roles/httpd a prereq of roles/osm if truly nec?
Similar to {roles/cups, roles/elgg, roles/lokole, roles/moodle, roles/nodered}

(Or, if that's not nec, PR #2464 may be the better way to go.)

jvonau, 2020-07-07:
> Thanks @georgejhunt & @jvonau for uncovering both bugs!
> 
> We could make roles/httpd a prereq of roles/osm if truly nec?

Why would you want to do that when osm doesn't ship an apache config file?

holta, 2020-07-07:
> Why would you want to do that when osm doesn't ship an apache config file?

Agreed, let's avoid Apache if/wherever possible at this point &mdash; in favor of NGINX: https://github.com/iiab/iiab/blob/master/roles/nginx/README.md

georgejhunt, 2020-07-07:
> Agreed, let's avoid Apache if/wherever possible at this point — in favor

So I need to add php-sqlite3 to osm-vector-maps

holta, 2020-07-11:
Fixed by PR #2464, thanks all for your patience over the past 5+ days!

Please try a new/fresh install of IIAB 7.2 (pre-release) here:

http://download.iiab.io

closed 2020-07-11 by holta