iiab/iiab #2462: Editing .ZIM files w/ Docker - cannot run containers

Hypernovae0, 2020-07-06:
Please be advised - this is one of my first experiences with any linux distro, especially Raspbian. I found this issue while following another [tutorial](https://www.google.com/amp/s/codeablereason.wordpress.com/2016/04/09/how-to-build-zimdump-to-process-zim-files/amp/) for editing .zim files. If there is a better way to do this, please let me know. I'm posting here because, although the issue is not with the IIAB software itself, it's impeding the setup of my IIAB, and I know there's so much relevant support and guidance in this community. Thank you :)
### Expected behavior and actual behavior:
Test: `docker run hello-world` should pull the "hello-world" image, create a new container, and output demo text
Output:
```
Unable to find image ‘hello-world:latest‘ locally
latest: Pulling from library/hello-world
4ee5c797bcd7: Pull complete
Digest: sha256: [long sha256]
Status: Downloaded newer image for hello-world: latest
```
Running `docker run hello-world` more than once outputs nothing, but opens more containers.
### Steps to reproduce the problem:
Opened command line without any other programs
Installed Docker with `curl -fsSL https://get.docker.com -o get-docker.sh`  and `sh get-docker.sh`
`docker version` and `docker info` run correctly, showing a new "stopped" container for each `docker run hello-world`
### IIAB version, OS version, hardware details:
IIAB 7.0 rev.0 on Raspbian 10
Raspberry Pi Zero W + 128GB microSD
Additional hardware info: https://pastebin.com/9N8uS7hj
### Please run the `iiab-diagnostics` command, to generate a URL summarizing your IIAB configuration, for volunteers seeking to help you: (Include that short URL here, Thanks!)
https://pastebin.com/xyD9Xpin

tim-moody, 2020-07-06:
Not sure how to do this; it's more a Kiwix thing than an IIAB thing. I don't think you can edit a zim, but you can edit the underlying files and rebuild it. I'd start by reading

https://openzim.org/wiki/Zimwriterfs_instructions

https://github.com/openzim/zim-tools

there is also a new python wrapper that is supposed to read and write zims files

https://github.com/openzim/python-libzim

tim-moody, 2020-07-06:
BTW there are other ways to add content to IIAB besides creating a zim.

Hypernovae0, 2020-07-07:
@tim-moody I'm sorry, I could have been more clear. I'm trying to edit specific .zim files because I need to remove graphic wikipedia pages from the ones I already have installed. Most of the pages you linked seem to require Docker as an installation step, and that's where I'm stuck. As for Meson and Ninja, I would love to know their purpose as compared to Docker.
If I'm in the wrong place, I'll see if Kiwix has forums, too.

tim-moody, 2020-07-07:
>there is also a new python wrapper that is supposed to read and write zims files
https://github.com/openzim/python-libzim

This is new and I haven't tried it, but you could try writing a python program that uses this library to replace pages you find offensive with a notice. I'll leave the philosophical part to others. If the problem is age-appropriateness you could look at the Wikipedia for Schools.