iiab/iiab #2479: Validate Ansible 2.10 expected in 2020-08-13

holta, 2020-08-08:
1) ETA is ~5 days away according to: https://docs.ansible.com/ansible/devel/roadmap/ROADMAP_2_10.html

2) Should we ponder how long Ubuntu will be locked into Ansible 2.9.6, due to its incomplete transition to Python 3 ?  And if so what this means for IIAB's Ansible strategy, currently summarized at:

   https://github.com/iiab/iiab/blob/master/scripts/ansible.md

holta, 2020-08-20:
Working on it: see #2481 and PR #2482 and beyond...

(Hopefully Ansible 2.10.1 will be released in coming weeks and get us out of this mess!)

holta, 2020-08-24:
Ansible 2.10.x installation will be simplified around 2020-09-22:

'For users that prefer a batteries-included model, the "ansible" package in most distribution channels will be updated in the coming weeks to a curated snapshot of collections that depends on the ansible-base package. This will provide a one-stop installation experience, similar to all previous versions of Ansible'

https://groups.google.com/g/ansible-devel/c/T6sZnzAPDZA
https://docs.ansible.com/ansible/devel/roadmap/COLLECTIONS_2_10.html#release-schedule

holta, 2020-08-27:
Ansible ETA's Announced:

- 31 August 2020:
  ansible-base 2.10.1rc1
  ansible 2.9.13
  ansible 2.8.15

- 7 September 2020:
  ansible-base 2.10.1

- 28 September 2020:
  ansible-base 2.10.2rc1
  ansible 2.9.14rc1
  ansible 2.8.16rc1

- 5 October 2020:
  ansible-base 2.10.2
  ansible 2.9.14
  ansible 2.8.16

https://groups.google.com/forum/m/#!topic/ansible-announce/cFA8XBYSFSs

Yes we should clean up `scripts/ansible` in the coming 10 days...but things will likely change again October ~5th (-:

holta, 2020-09-01:
Ansible 2.9.13 is now available FWIW:
https://launchpad.net/~ansible/+archive/ubuntu/ansible
https://github.com/ansible/ansible/blob/stable-2.9/changelogs/CHANGELOG-v2.9.rst

> As per a previous email to the announce list, we plan to start doing release candidates for minor releases going forward.
This has already begun for ansible-base 2.10, and will begin for ansible 2.8.z and 2.9.z starting with the next release, three weeks from now, on 2020-09-24.
Future maintenance releases will occur approximately every 3 weeks.
So expect RCs for 2.8.16 and 2.9.14 around 2020-09-21, and releases the following week.

holta, 2020-09-14:
If all goes according to plan, ansible-base-2.10.1 should be released today and appear here:

http://ppa.launchpad.net/ansible/ansible/ubuntu/pool/main/a/ansible-base/

holta, 2020-09-14:
The new ansible-base-2.10.1 is now live here:
https://launchpad.net/~ansible/+archive/ubuntu/ansible-2.10

holta, 2020-09-14:
> Presumably later on it'll also appear here:
> https://launchpad.net/~ansible/+archive/ubuntu/ansible

That too...is now live.

FYI "bionic" is the only .deb published:
http://ppa.launchpad.net/ansible/ansible/ubuntu/pool/main/a/ansible-base/ansible-base_2.10.1-1ppa~bionic_all.deb

holta, 2020-09-15:
Done.  PR #2518 is now merged, so Ansible 2.10 is now live as part of IIAB 7.2 pre-releases, using ansible-base 2.10.1

closed 2020-09-15 by holta

holta, 2020-09-15:
Release Notes for ansible-base 2.10.1 are here:

https://github.com/ansible/ansible/blob/stable-2.10/changelogs/CHANGELOG-v2.10.rst#v2101

holta, 2020-09-18:
@nzola @jvonau these [73 Ansible Collections](https://github.com/ansible-community/ansible-build-data/blob/2.10.0rc1/2.10/ansible-2.10.0rc1.deps) can be installed automatically if you want a Really Big Hammer.  (These 73 are also listed here: https://github.com/ansible-community/ansible-build-data/blob/main/2.10/CHANGELOG-v2.10.rst#unchanged-collections)

i.e. if you have downloading/networking failures when trying to install Ansible, we should try to understand why, but in the meantime you can feel free to try this: `pip install ansible==2.10.0rc1` as mentioned in https://groups.google.com/forum/m/#!topic/ansible-devel/hgXx7CEugt0

Even if I personally suspect this approach is likely overkill and overweight (IIAB only needs [2 of these 73 Ansible Collections](https://github.com/iiab/iiab/blob/master/scripts/ansible#L94-L96)).

jvonau, 2020-09-18:
> @nzola @jvonau these [73 Ansible Collections](https://github.com/ansible-community/ansible-build-data/blob/2.10.0rc1/2.10/ansible-2.10.0rc1.deps) can be installed automatically if you want a Really Big Hammer. (Most of these ~73 Ansible Collections are also listed here: https://github.com/ansible-community/ansible-build-data/blob/main/2.10/CHANGELOG-v2.10.rst#unchanged-collections)
> 

Reference fodder, who do you think guided you through #2481?

> i.e. if you have downloading/networking failures when trying to install Ansible, we should try to understand why, but in the meantime you can feel free to try this: `pip install ansible==2.10.0rc1` as mentioned in https://groups.google.com/forum/m/#!topic/ansible-devel/hgXx7CEugt0
> 

To make it clear ansible-base 2.10.1 was installed, the issue occurred when installing the collections. Explain how using pip would magical fix a flaky network connection when pip uses the network? Think that falls under bad advice.

> Even if I personally suspect this approach is likely overkill and overweight (IIAB only needs [2 of these 73 Ansible Collections](https://github.com/iiab/iiab/blob/master/scripts/ansible#L94-L96)).
> 
> Ref: #2532 "Ansible Collections install seems flaky?"

That is verbatim what was reported, great impression introducing a new 'feature', as you say, this late in the release cycle.