iiab/iiab #2481: ansible has promised not to change things, but this seems like  big one

georgejhunt, 2020-08-11:
On ubuntu 18.04 VM, IIAB does not function because of the following sprinkled through the log:
[WARNING]: File '/tmp/heart-beat.txt' created with default permissions '600'. The previous default
was '666'. Specify 'mode' to avoid this warning.

All of the server tree is unreadable by www-data because permissions are set to 0600 (owner root). I have not found a config setting which reverts the new behavior.

I did not encounter this yesterday testing on a rpi4. umask is 0022, which should yield a 0766 default.

Ansible --version is 2.7.17 July 20, 2020
Ansible's promise for "copy" module:
This module is flagged as stableinterface which means that the maintainers for this module guarantee that no backward incompatible interface changes will be made.

Maybe I'm missing something

holta, 2020-08-11:
Got an error message handy, above & beyond the warning?

> Ansible --version is 2.7.17 July 20, 2020

That's an old version of Ansible not recommended — not sure if that matters!  But FYI IIAB's install instructions (https://github.com/iiab/iiab/wiki/IIAB-Installation#install-the-software) ask for /opt/iiab/iiab/scripts/ansible to install Ansible 2.9.12 — or if that's not possible, use Ansible 2.9.6 packaged by apt for Ubuntu 20.04.

More context if you know?  While IIAB does't fully support Ubuntu 18.04 anymore...is this a wider issue...affecting more than just Ubuntu 18.04?

tim-moody, 2020-08-11:
my min on ub20.04.1 vm ran to completion. 
osm viewer comes up.
admin comes up

holta, 2020-08-12:
@jvonau: I spoke with @georgejhunt and this problem appears to be exclusive to Ubuntu 18.04 (it's repeatable on George's 18.04 VM's) and he clarified that it's running Ansible 2.9.x not 2.7.17

George is installing IIAB by running `curl d.iiab.io/install.txt | bash` as root.

Any idea why file permissions might be getting screwed up on 18.04?

georgejhunt, 2020-08-13:
This seems related:
https://docs.ansible.com/ansible/devel/porting_guides/porting_guide_2.10.html#change-to-default-file-permissions
I conclude from the following that we have no choice but to modify our
playbook:

To address CVE-2020-1736, the default permissions for certain files created
by Ansible using atomic_move() were changed from 0o666 to 0o600. The
default permissions value was only used for the temporary file before it
was moved into its place or newly created files. If the file existed when
the new temporary file was moved into place, Ansible would use the
permissions of the existing file. If there was no existing file, Ansible
would retain the default file permissions, combined with the system umask,
of the temporary file.

georgejhunt, 2020-08-13:
But then we have the data point that U20.04 does not fail this way.
U20.04 ansible --version:

root@box:/opt/iiab/iiab# ansible --version

ansible 2.9.6

  config file = /opt/iiab/iiab/ansible.cfg

  configured module search path = ['/root/.ansible/plugins/modules',
'/usr/share/ansible/plugins/modules']

  ansible python module location = /usr/lib/python3/dist-packages/ansible

  executable location = /usr/bin/ansible

  python version = 3.8.2 (default, Jul 16 2020, 14:00:26) [GCC 9.3.0]

root@box:/opt/iiab/iiab#

But on U18.04:

root@box:~# ansible --version

ansible 2.9.9

  config file = /etc/ansible/ansible.cfg

  configured module search path = [u'/root/.ansible/plugins/modules',
u'/usr/share/ansible/plugins/modules']

  ansible python module location = /usr/lib/python2.7/dist-packages/ansible

  executable location = /usr/bin/ansible

  python version = 2.7.17 (default, Apr 15 2020, 17:20:14) [GCC 7.5.0]

root@box:~#

Maybe we just need to pin ansible at 2.9.6 until the dust settles. Or could
we entice U18 to run ansible under python3? Adam was mentioning the python3
issue which I did not understand.

holta, 2020-08-13:
Many Thanks @georgejhunt for digging into this and the progress you've made above, if indeed Ansible is changing permissions going forward?

(Even if IIAB no longer supports Ubuntu 18.04 since almost 4 months ago, according to https://github.com/iiab/iiab/wiki/IIAB-Platforms -- I still believe this is important to understand!)

jvonau, 2020-08-15:
The [background upstream issue](https://github.com/ansible/ansible/issues/71200) suggests that the 'default mode change to 0600' is only used when the mode is NOT declared in the stanza with the latest version 2.9.12. Thought that the use of mode was a given when creating files/directories with ansible but I see with 4b1b2780deb9cd7e2e7f3ac9f75c36e777238c08 this is no longer the norm. Good news is the change is [reverted](https://github.com/ansible/ansible/commits/stable-2.9) but needs to be released.

holta, 2020-08-15:
@jvonau do you happen to know if this ticket's problem is likely to occur with current IIAB installs (http://d.iiab.io) using the brand new Ansible 2.10 ?

jvonau, 2020-08-15:
Think you better re-confirm which variant of ansible you want installed, using the apt method in scripts/ansible one would be on the 2.9.X track, with 2.9.12 being the latest, as the apt package name changed between 2.9.X (from ansible) and 2.10.Y (to ansible-base and using python3). 2.10.X doesn't enter the picture unless one asks to install ansible-base from the beginning.

holta, 2020-08-19:
> the apt package name changed between 2.9.X (from ansible) and 2.10.Y (to ansible-base and using python3). 2.10.Y doesn't enter the picture unless one asks to install ansible-base from the beginning.

Does anybody know how to install "ansible-base | 2.10.0-1ppa~bionic | Ansible, Inc. (2020-08-13)" as listed at https://launchpad.net/~ansible/+archive/ubuntu/ansible ?

e.g. what `apt install` command should we try, to install this onto Ubuntu 20.04.1 on x86_64 ?

jvonau, 2020-08-19:
I'd try `apt install ansible-base`

holta, 2020-08-19:
> I'd try `apt install ansible-base`

No luck so far, on Ubuntu Server 20.04.1 here:

```
root@u20-srv:~# apt install ansible-base
Reading package lists... Done
Building dependency tree
Reading state information... Done
E: Unable to locate package ansible-base
```

```
root@u20-srv:~# apt -a list "*ansible*"
Listing... Done
ansible-doc/focal 2.9.6+dfsg-1 all
ansible-lint/focal 4.2.0-1 all
ansible-tower-cli-doc/focal 3.3.0-1.1 all
ansible-tower-cli/focal 3.3.0-1.1 all
ansible/focal 2.9.6+dfsg-1 all
```

jvonau, 2020-08-19:
You need to add the PPA to get to the ansible apt repo. Think the easiest would be to edit scripts/ansible, at Line 67 change "focal" to "XXX" and at Line 94 change "ansible" to "ansible-base", save and run the script.

holta, 2020-08-19:
This is a bare machine without any IIAB so far, where I'd already run the following:

```
add-apt-repository ppa:ansible/ansible
```

Which resulted in...

```
root@u20-srv:~# more /etc/apt/sources.list.d/ansible-ubuntu-ansible-focal.list
deb http://ppa.launchpad.net/ansible/ansible/ubuntu focal main
# deb-src http://ppa.launchpad.net/ansible/ansible/ubuntu focal main
```

It then failed as follows:

```
root@u20-srv:~# apt update
Hit:1 http://us.archive.ubuntu.com/ubuntu focal InRelease
Hit:2 http://us.archive.ubuntu.com/ubuntu focal-updates InRelease
Hit:3 http://us.archive.ubuntu.com/ubuntu focal-backports InRelease
Hit:4 http://us.archive.ubuntu.com/ubuntu focal-security InRelease
Ign:5 http://ppa.launchpad.net/ansible/ansible/ubuntu focal InRelease
Err:6 http://ppa.launchpad.net/ansible/ansible/ubuntu focal Release
  404  Not Found [IP: 91.189.95.83 80]
Reading package lists... Done
E: The repository 'http://ppa.launchpad.net/ansible/ansible/ubuntu focal Release' does not have a Release file.
N: Updating from such a repository can't be done securely, and is therefore disabled by default.
N: See apt-secure(8) manpage for repository creation and user configuration details.
```

Any ideas to get around this?

jvonau, 2020-08-19:
edit /etc/apt/sources.list.d/ansible-ubuntu-ansible-focal.list and change all 'focal' to be 'bionic', then `apt update`

holta, 2020-08-19:
> /etc/apt/sources.list.d/ansible-ubuntu-ansible-focal.list

That worked...allowing `apt install ansible-base` to proceed:

```
root@u20-srv:~# which ansible
/usr/bin/ansible

root@u20-srv:~# ansible --version
ansible 2.10.0
  config file = /etc/ansible/ansible.cfg
  configured module search path = ['/root/.ansible/plugins/modules', '/usr/share/ansible/plugins/modules']
  ansible python module location = /usr/lib/python3/dist-packages/ansible
  executable location = /usr/bin/ansible
  python version = 3.8.2 (default, Jul 16 2020, 14:00:26) [GCC 9.3.0]
```

jvonau, 2020-08-19:
The 2 edits in scripts/ansible mentioned above would of saved you the grief of editing ansible-ubuntu-ansible-focal.list as we create our own iiab-ansible.list file that contains 'bionic'.

holta, 2020-08-19:
> The 2 edits in scripts/ansible mentioned above would of saved you the grief of editing ansible-ubuntu-ansible-focal.list as we create our own iiab-ansible.list file that contains 'bionic'.

Indeed.  This smells like a bug in Ansible 2.10.0's (lack of) support for Ubuntu 20.04 (Focal Fossa) &mdash; given the instructions near the top of https://launchpad.net/~ansible/+archive/ubuntu/ansible say to do:

```
sudo add-apt-repository ppa:ansible/ansible
sudo apt-get update
```

holta, 2020-08-19:
When attempting `./iiab-install` using Ansible 2.10.0 it fails almost immediately &mdash; the behavior of Ansible 2.10.0 appears different than 2.9.x here &mdash; for reasons I don't yet understand:

```
TASK [0-init] ******************************************************************
ERROR! couldn't resolve module/action 'ini_file'. This often indicates a misspelling, missing collection, or incorrect module path.

The error appears to be in '/opt/iiab/iiab/roles/0-init/tasks/main.yml': line 153, column 3, but may
be elsewhere in the file depending on the exact syntax problem.

The offending line appears to be:

- name: "Add 'runtime' variable values to {{ iiab_ini_file }}"
  ^ here
We could be wrong, but this one looks like it might be an issue with
missing quotes. Always quote template expression brackets when they
start a value. For instance:

    with_items:
      - {{ foo }}

Should be written as:

    with_items:
      - "{{ foo }}"
```

jvonau, 2020-08-19:
>ERROR! couldn't resolve module/action 'ini_file'

I would suspect that the 'ini_file' module might of been renamed or is not part of ansible-base and may be part of 'collections' now.

holta, 2020-08-19:
> I would suspect that the 'ini_file' module might of been renamed or is not part of ansible-base and may be part of 'collections' now.

Unfortunate that https://docs.ansible.com/ansible/latest/modules/ini_file_module.html doesn't mention anything (yet!)

When I searched for `ini_file` at https://galaxy.ansible.com it suggested https://galaxy.ansible.com/community/general so I ran:

```
ansible-galaxy collection install community.general
```

Running `./iiab-install` now works better on 10.8.0.46 = ansible210-u20srv-MEDIUM where it crashed in Stage 3:

```
TASK [Install MySQL if 'mysql_installed' not defined, e.g. in /etc/iiab/iiab_state.yml] ***
fatal: [127.0.0.1]: FAILED! => {"reason": "couldn't resolve module/action 'mysql_user'. This often indicates a misspelling, missing collection, or incorrect module path.\n\nThe error appears to be in '/opt/iiab/iiab/roles/mysql/tasks/install.yml': line 119, column 3, but may\nbe elsewhere in the file depending on the exact syntax problem.\n\nThe offending line appears to be:\n\n# unfortunately it still doesn't work\n- name: Update MySQL root password for localhost root accounts\n  ^ here\n"}
```

Rinse & repeat, searching for each missing module at https://galaxy.ansible.com ?

Indeed, running the following seems to solve the above error:

```
ansible-galaxy collection install community.mysql
```

jvonau, 2020-08-19:
After that I suspect it might fail because the underlying debs are not installed, use the list from L94 of scripts/ansible to correct.

holta, 2020-08-19:
@georgejhunt `./iiab-install` appears stuck after about 20 minutes here:

```
TASK [osm-vector-maps : Install the World Map to zoom 10] **********************
```

Is this related to Ansible's new permissions problem?  (Or should we warn the implementer/operator in the text of the above task description if this is the new normal?)  So they know (roughly!) how long they might need to wait before giving up here:

https://github.com/iiab/iiab/blob/master/roles/osm-vector-maps/tasks/install.yml#L184-L186

holta, 2020-08-19:
> After that I suspect it might fail because the underlying debs are not installed, use the list from L94 of scripts/ansible to correct.

I ran `scripts/ansible` earlier, with its Line 94 changed "ansible" to "ansible-base" to be safe.

I don't know of any missing underlying debs (but shout if some such others need to be installed?)

jvonau, 2020-08-19:
> > After that I suspect it might fail because the underlying debs are not installed, use the list from L94 of scripts/ansible to correct.
> 
> I ran `scripts/ansible` earlier, with its Line 94 changed "ansible" to "ansible-base" to be safe.
> 
Then you should not have other issues, apt wise anyway. Should cover what we needed before 2.10, postgre might need the 'collections' treatment and other might crop up.

holta, 2020-08-19:
> Then you should not have other issues, apt wise anyway. Should cover what we needed before 2.10, postgre might need the 'collections' treatment and other might crop up.

Great!  Let's do a BIG-sized install later to confirm.

@georgejhunt this current MEDIUM-sized install is still stuck (must have hung) at `TASK [osm-vector-maps : Install the World Map to zoom 10]` after about 40 minutes.

jvonau, 2020-08-20:
> > Then you should not have other issues, apt wise anyway. Should cover what we needed before 2.10, postgre might need the 'collections' treatment and other might crop up.
> 
> Great! Let's do a BIG-sized install later to confirm.

Just move the big vars file into place and run ./iiab-configure, should run the installs for the newly changed *_install flags without having to start from scratch.

holta, 2020-08-20:
> @georgejhunt this current MEDIUM-sized install is still stuck (must have hung?) at TASK [osm-vector-maps : Install the World Map to zoom 10] after about 40 minutes.

CORRECTION: this step completed 38 minutes later, much more than tripling the total install time &mdash; when all other `./iiab-install` steps completed in 15 minutes total.

This seems very wrong ??

Certainly it makes for a very unfriendly install experience at the moment :/

holta, 2020-08-20:
> Just move the big vars file into place and run ./iiab-configure, should run the installs for the newly changed *_install flags without having to start from scratch.

Underway: `./iiab-configure` ran Stage 0, Stage 4 and is now proceeding thru Stage 6...

holta, 2020-08-20:
@jvonau the BIG-sized install failed on 10.8.0.46 here:

```
TASK [Enable & Start 'kolibri' systemd service, if kolibri_enabled] ************
fatal: [127.0.0.1]: FAILED! => {"changed": false, "msg": "Unable to start service kolibri: Job for kolibri.service failed because the control process exited with error code.\nSee \"systemctl status kolibri.service\" and \"journalctl -xe\" for details.\n"}
```

Is this is possibly a Kolibri 0.14.1 bug, that might go away tomorrow/Friday when Kolibri 0.14.3 is packaged up as a deb file for download from https://learningequality.org/download/ ?

jvonau, 2020-08-20:
no clue.

holta, 2020-08-20:
FYI changing the .deb from Kolibri 0.14.1 (https://learningequality.org/r/kolibri-deb-latest) to 0.14.3 (https://github.com/learningequality/kolibri/releases/download/v0.14.3/kolibri_0.14.3-0ubuntu1_all.deb) leads to the same failure:

```
TASK [Enable & Start 'kolibri' systemd service, if kolibri_enabled] ************
fatal: [127.0.0.1]: FAILED! => {"changed": false, "msg": "Unable to start service kolibri: Job for kolibri.service failed because the control process exited with error code.\nSee \"systemctl status kolibri.service\" and \"journalctl -xe\" for details.\n"}
```

```
root@u20-srv:/opt/iiab/iiab# systemctl status kolibri.service
● kolibri.service - Kolibri
     Loaded: loaded (/etc/systemd/system/kolibri.service; enabled; vendor preset: enabled)
     Active: failed (Result: exit-code) since Wed 2020-08-19 20:32:00 EDT; 1min 25s ago
    Process: 88861 ExecStart=/usr/bin/kolibri start (code=exited, status=1/FAILURE)

Aug 19 20:32:00 box.lan kolibri[88861]:   File "/usr/lib/python3/dist-packages/kolibri/utils/cli.py", line 272, in invoke
Aug 19 20:32:00 box.lan kolibri[88861]:     initialize()
Aug 19 20:32:00 box.lan kolibri[88861]:   File "/usr/lib/python3/dist-packages/kolibri/utils/cli.py", line 327, in initialize
Aug 19 20:32:00 box.lan kolibri[88861]:     check_debian_user(params.get("no_input"))
Aug 19 20:32:00 box.lan kolibri[88861]:   File "/usr/lib/python3/dist-packages/kolibri/utils/debian_check.py", line 29, in check_debian_user
Aug 19 20:32:00 box.lan kolibri[88861]:     with open("/etc/kolibri/username", "r") as f:
Aug 19 20:32:00 box.lan kolibri[88861]: PermissionError: [Errno 13] Permission denied: '/etc/kolibri/username'
Aug 19 20:32:00 box.lan systemd[1]: kolibri.service: Control process exited, code=exited, status=1/FAILURE
Aug 19 20:32:00 box.lan systemd[1]: kolibri.service: Failed with result 'exit-code'.
Aug 19 20:32:00 box.lan systemd[1]: Failed to start Kolibri.
```

...which would seem likely caused by Ansible's 666-to-600 file permissions changes?

```
root@u20-srv:/opt/iiab/iiab# ls -l /etc/kolibri/username
-rw------- 1 root root 7 Aug 19 20:13 /etc/kolibri/username
```

jvonau, 2020-08-20:
Yes, that would be the 600 issue, no mode in the stanza used in the role.

holta, 2020-08-20:
FYI the BIG-sized IIAB install completed on Ansible 2.10.0 (using `./iiab-configure`) after I set `kolibri_install: False` and `kolibri_enabled: False` in `/etc/iiab/local_vars.yml` -- and removed line `kolibri_installed: True` from `/etc/iiab/iiab_state.yml`

Conclusions:
- Only 2 Ansible "collections" are required (community.general and community.mysql) as mentioned [above](https://github.com/iiab/iiab/issues/2481#issuecomment-676799099).
- Somehow we need to solve Ansible's new 666-to-600 permissions problem, that @jvonau has [summarized here](https://github.com/iiab/iiab/issues/2481#issuecomment-674447247) &mdash; does this mean waiting for Ansible 2.10.1 likely being released in a few weeks?

jvonau, 2020-08-20:
2.9.12 is hooped in the same way, so until there is a new release consider all new installs broken.

jvonau, 2020-08-20:
One could audit all the uses of copy, template, file, etc and ensure there is a mode declared as @georgejhunt suggests above.

holta, 2020-08-20:
> 2.9.12 is hooped in the same way, so until there is a new release consider all new installs broken.

Presumably installs of IIAB onto Ubuntu 20.04 still work, as this uses Ansible 2.9.6

> One could audit all the uses of copy, template, file, etc and ensure there is a mode declared as @georgejhunt suggests above.

We could also:

- Revert to any recent version of Ansible like 2.9.6 or 2.9.11
- Try to set Ansible's default mode across the board?
- If nothing else, waiting for Ansible 2.9.13 and 2.10.1 in coming weeks should hopefully solve most all of this!

jvonau, 2020-08-20:
After `ansible-galaxy collection install community.general` where did the 'modules' directory get created? In the user pi's .ansible/plugins path or in /usr/share/ansible/plugins/?

holta, 2020-08-20:
> After `ansible-galaxy collection install community.general` where did the 'modules' directory get created? In the user ubuntu|pi's .ansible/plugins path or in /usr/share/ansible/plugins/?

On 10.8.0.46 (where I ran `ansible-galaxy` commands as root) the answer seems to be:

```
root@u20-srv:~# ls -l /root/.ansible/collections/ansible_collections/community/
total 12
drwxr-xr-x 8 root root 4096 Aug 19 18:53 general
drwxr-xr-x 8 root root 4096 Aug 19 18:54 kubernetes
drwxr-xr-x 7 root root 4096 Aug 19 19:02 mysql

root@u20-srv:~/.ansible/collections/ansible_collections/community# du -hsc *
23M     general
956K    kubernetes
856K    mysql
25M     total
```

Example 1:

```
root@u20-srv:~/.ansible/collections/ansible_collections/community/kubernetes# tree -d
.
├── changelogs
│   └── fragments
├── meta
├── molecule
│   └── default
│       ├── roles
│       │   └── helm
│       │       ├── defaults
│       │       └── tasks
│       │           └── tests_chart
│       ├── tasks
│       └── vars
├── plugins
│   ├── connection
│   ├── doc_fragments
│   ├── filter
│   ├── inventory
│   ├── lookup
│   ├── module_utils
│   └── modules
└── tests
    ├── integration
    │   └── targets
    │       └── kubernetes
    │           ├── defaults
    │           ├── files
    │           ├── handlers
    │           ├── library
    │           ├── meta
    │           └── tasks
    └── sanity

31 directories
```

Example 2:

```
root@u20-srv:~/.ansible/collections/ansible_collections/community/mysql/# tree -d
.
├── changelogs
│   └── fragments
├── meta
├── plugins
│   ├── doc_fragments
│   ├── module_utils
│   └── modules
└── tests
    ├── integration
    │   ├── old_mariadb_replication
    │   │   ├── defaults
    │   │   ├── meta
    │   │   └── tasks
    │   └── targets
    │       ├── setup_mysql
    │       │   ├── defaults
    │       │   ├── handlers
    │       │   ├── tasks
    │       │   ├── templates
    │       │   └── vars
    │       ├── setup_remote_tmp_dir
    │       │   ├── handlers
    │       │   └── tasks
    │       ├── test_mysql_db
    │       │   ├── defaults
    │       │   ├── meta
    │       │   └── tasks
    │       ├── test_mysql_info
    │       │   ├── defaults
    │       │   ├── meta
    │       │   ├── tasks
    │       │   └── templates
    │       ├── test_mysql_query
    │       │   ├── defaults
    │       │   ├── meta
    │       │   └── tasks
    │       ├── test_mysql_replication
    │       │   ├── defaults
    │       │   ├── meta
    │       │   └── tasks
    │       ├── test_mysql_user
    │       │   ├── defaults
    │       │   ├── files
    │       │   ├── meta
    │       │   └── tasks
    │       └── test_mysql_variables
    │           ├── defaults
    │           ├── meta
    │           └── tasks
    └── sanity

50 directories
```

Example 3...contains 839 directories...so this output shows only those up to depth 3:

```
root@u20-srv:~/.ansible/collections/ansible_collections/community/general# tree -dL 3
.
├── changelogs
│   └── fragments
├── meta
├── plugins
│   ├── action
│   │   └── system
│   ├── become
│   ├── cache
│   ├── callback
│   ├── connection
│   ├── doc_fragments
│   ├── filter
│   ├── inventory
│   ├── lookup
│   ├── module_utils
│   │   ├── compat
│   │   ├── docker
│   │   ├── identity
│   │   ├── net_tools
│   │   ├── oracle
│   │   ├── remote_management
│   │   ├── source_control
│   │   └── storage
│   └── modules
│       ├── cloud
│       ├── clustering
│       ├── database
│       ├── files
│       ├── identity
│       ├── monitoring
│       ├── net_tools
│       ├── notification
│       ├── packaging
│       ├── remote_management
│       ├── source_control
│       ├── storage
│       ├── system
│       └── web_infrastructure
├── scripts
│   ├── inventory
│   └── vault
└── tests
    ├── integration
    │   └── targets
    ├── sanity
    │   └── extra
    ├── unit
    │   ├── compat
    │   ├── mock
    │   └── plugins
    └── utils
        └── shippable

52 directories

holta, 2020-08-20:
Temporary Workaround merged (PR #2482) so all IIAB fresh installs should now be able to proceed...using Ansible 2.9.6 on all OS's for the moment...until Ansible cleans up this #2481 situation in coming weeks.

holta, 2020-09-08:
This is confirmed solved by Ansible 2.9.13 as implemented by PR #2501 &mdash; and the imminent Ansible 2.10.1+ as explained here: https://docs.ansible.com/ansible/devel/porting_guides/porting_guide_2.10.html#modules

closed 2020-09-08 by holta

jvonau, 2020-09-09:
The mass commenting out of 'mode' performed by #2147 & #2158 helped contribute to this "mode failure". Tell me why developers have taken great pains to ensure the the directory permissions were correct only to have that effort mass reverted?